arendst/Tasmota #4266: More than 4 IDX for Domoticz integration

mkrawcz1, 2018-11-03:
Hi :-)
According to https://github.com/arendst/Sonoff-Tasmota/issues/1545 - are there any plans to allow more than 4 IDXs? I'm trying to integrate Tasmota-based Wemos D1 irrigation controller (8 zones, relay controlled) with Domoticz and this feature would be much appreciated :-)
Cheers
MK

Jason2866, 2018-11-03:
Take a look here:
https://github.com/arendst/Sonoff-Tasmota/wiki/Rules#5-use-of-variables-and-tele-

mkrawcz1, 2018-11-03:
OK, I can do it also by NodeRED or with different approach, but I'm asking if simple way will be available soon...

Jason2866, 2018-11-04:
No, because changing will break backwards compability.

MrFusion25, 2021-01-07:
Is there any news on this thing? I do have a Power strip with 4 outlets and USB plugs. Missing the IDX5 to control it from domoticz. that's a pitty because all the rest works perfectly fine.

Barry40, 2024-11-16:
Also still waiting to use more then 4 relays for my heatpump projects. Thanks

sfromis, 2024-11-16:
Looks like you still can use more than 4 relays, by configuring via the `DzIdx` command. https://github.com/arendst/Tasmota/issues/21019#issuecomment-2028702899