arendst/Tasmota #4268: Inverted switching?

pickett2k, 2018-11-03:
Hi guys just a quick hopefully
Easy one, anyone know a way I can invert the switching so that on means off and off means on? 

I am using an Alexa group and 2
Sonoff basics, I want to just say “turn on cinema” and the main light goes off (sonoff 1) and the strip lights come on (sonoff 2) 

But when I do this currently if the main light is on it just stays on until I say off then they both turn off...

Hope this makes sense :) 

Thanks

reloxx13, 2018-11-03:
use alexa routine

Frogmore42, 2018-11-04:
Select generic device and configure the pin for the relay to be the inverse of what it is now.  ( It is probably Relay1, make it Relay1i and it will be on when the device thinks it is off.)

curzon01, 2018-11-04:
Reverse ON/OFF meaning for light Sonoff using 
`StateText1 ON`
`StateText2 OFF`

ascillato2, 2018-11-04:
Closing this issue as it has been answered.

### Support Information
See [Wiki](https://github.com/arendst/Sonoff-Tasmota/wiki) for more information.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.

closed 2018-11-04 by ascillato2

meingraham, 2018-11-05:
@curzon01 Norbert,

Fairly certain I know what the answer will be, but, nothing ventured, nothing gained...

For StateTextN, it appears that this is a device-wide setting. In other words, for devices with more than one channel (e.g., Sonoff 4Ch), one cannot set each channel to have its own StateText setting; correct?

Regards.

Mike

curzon01, 2018-11-06:
@meingraham Mike, you know the answer :wink: - yes `StateText<x>` is a global setting for all channels

apotapov, 2020-08-11:
I know I'm a little late to the game, but I came across a similar issue on a 3-way switch, where for some reason it had the on/off status flipped. Changing the StateText caused a bunch of issues for me with automations. One solution suggested was to rewire change the traveller wires, but I was too lazy for that. An alternate solution I found was go into the Tasmota Configuration->Configure Template and switch Relay1 to Relay1i for one of the GPIOs (GPIO12 in my case - treatlife 3-way switch) Hope it helps someone else.

northerntaco, 2020-08-15:
@apotapov, Your post helped me out. I ended up switching the travelers.

sillyfrog, 2021-01-18:
Also rather late to the game, but for others, I found that changing `SwitchMode` from `1` to `2` did exactly what I needed. Specifically, just for a single switch, I could do, for example, `SwitchMode2 2`. More info here: https://tasmota.github.io/docs/Buttons-and-Switches/

smizles, 2021-07-16:
> I know I'm a little late to the game, but I came across a similar issue on a 3-way switch, where for some reason it had the on/off status flipped. Changing the StateText caused a bunch of issues for [...]

I have this exact same problem. I opened an issue about it here (https://github.com/blakadder/templates/issues/1102#issue-945976853). In the meantime, I will experiment with your workaround. Thanks for the suggestion!

smizles, 2021-07-16:
> Also rather late to the game, but for others, I found that changing `SwitchMode` from `1` to `2` did exactly what I needed. Specifically, just for a single switch, I could do, for example, `SwitchMode2 2`. More info here: https://tasmota.github.io/docs/Buttons-and-Switches/

Hallelujah, setting it to `SwitchMode 2` in the console fixed it for me (and even got the LED on the switch to light when the switch is Off for "night light" illumination). Is there a way to implement this as the default in the template? Because I'm pretty sure this is how most people will want this switch set up (maybe not the LED night light, but definitely the correct switch state).