arendst/Tasmota #4270: Detect current flow direction with Sonoff POW and/or POW R2

dan-cristian, 2018-11-04:
Hi. I could not find any reference if Tasmona on Sonoff POW or POW R2 can detect current flow direction.

I have a solar system in the house (microinverters coupled to my power line) and want to detect when I am exporting power to the grid. Some power meters (Qubino for example: http://qubino.com/products/smart-meter/) display the energy as a negative value when exporting to the grid, but they are expensive.

Is it possible to detect the current flow, does the HW or SW support this?

- [x] Searched the problem in issues (https://github.com/arendst/Sonoff-Tasmota/issues)
- [x] Searched the problem in the wiki (https://github.com/arendst/Sonoff-Tasmota/wiki/Troubleshooting)
- [x] Searched the problem in the forum (https://groups.google.com/d/forum/sonoffusers)
- [x] Searched the problem in the chat (https://discord.gg/Ks2Kzd4)
- [X] Development/Compiler/Upload tools used : esptool
- [X] Hardware used : Sonoff POW R2
- [X] If a pre-compiled release or development binary was used, which one? : sonoff.bin
- [X] You have tried latest release or development binaries? :
- [ ] Provide the output of command ``status 0`` :
```
07:55:55 MQT: iot/sonoff/stat/sonoff-pow-3/STATUS = {"Status":{"Module":43,"FriendlyName":["sonoff-pow-3"],"Topic":"sonoff-pow-3","ButtonTopic":"0","Power":1,"PowerOnState":1,"LedState":1,"SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
07:55:55 MQT: iot/sonoff/stat/sonoff-pow-3/STATUS1 = {"StatusPRM":{"Baudrate":4800,"GroupTopic":"sonoffs","OtaUrl":"http://thehackbox.org/tasmota/release/sonoff.bin","RestartReason":"Power on","Uptime":"2T08:18:18","StartupUTC":"2018-11-01T22:37:37","Sleep":0,"BootCount":9,"SaveCount":32,"SaveAddress":"F4000"}}
07:55:55 MQT: iot/sonoff/stat/sonoff-pow-3/STATUS2 = {"StatusFWR":{"Version":"6.3.0","BuildDateTime":"2018-10-30T17:33:53","Boot":31,"Core":"2_3_0","SDK":"1.5.3(aec24ac9)"}}
07:55:55 MQT: iot/sonoff/stat/sonoff-pow-3/STATUS3 = {"StatusLOG":{"SerialLog":0,"WebLog":2,"SysLog":0,"LogHost":"192.168.0.9","LogPort":514,"SSId":["home2",""],"TelePeriod":300,"SetOption":["00008009","55A18000","00000000"]}}
07:55:55 MQT: iot/sonoff/stat/sonoff-pow-3/STATUS4 = {"StatusMEM":{"ProgramSize":489,"Free":512,"Heap":13,"ProgramFlashSize":1024,"FlashSize":4096,"FlashMode":3,"Features":["00000809","0FDAE794","000183A0","23B617CE","00003BC0"]}}
07:55:55 MQT: iot/sonoff/stat/sonoff-pow-3/STATUS5 = {"StatusNET":{"Hostname":"sonoff-pow-3","IPAddress":"192.168.0.123","Gateway":"192.168.0.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.0.1","Mac":"84:F3:EB:B1:D0:9F","Webserver":2,"WifiConfig":5}}
07:55:55 MQT: iot/sonoff/stat/sonoff-pow-3/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.0.12","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_B1D09F","MqttUser":"DVES_USER","MqttType":1,"MAX_PACKET_SIZE":1000,"KEEPALIVE":15}}
07:55:55 MQT: iot/sonoff/stat/sonoff-pow-3/STATUS7 = {"StatusTIM":{"UTC":"Sun Nov 04 06:55:55 2018","Local":"Sun Nov 04 07:55:55 2018","StartDST":"Sun Mar 25 02:00:00 2018","EndDST":"Sun Oct 28 03:00:00 2018","Timezone":1,"Sunrise":"07:42","Sunset":"17:25"}}
07:55:55 MQT: iot/sonoff/stat/sonoff-pow-3/STATUS9 = {"StatusPTH":{"PowerDelta":2,"PowerLow":0,"PowerHigh":0,"VoltageLow":0,"VoltageHigh":0,"CurrentLow":0,"CurrentHigh":0}}
07:55:55 MQT: iot/sonoff/stat/sonoff-pow-3/STATUS10 = {"StatusSNS":{"Time":"2018-11-04T07:55:55","ENERGY":{"TotalStartTime":"2018-10-31T20:05:27","Total":3.997,"Yesterday":2.073,"Today":0.160,"Power":205,"ApparentPower":216,"ReactivePower":68,"Factor":0.95,"Voltage":221,"Current":0.978}}}
07:55:55 MQT: iot/sonoff/stat/sonoff-pow-3/STATUS11 = {"StatusSTS":{"Time":"2018-11-04T07:55:55","Uptime":"2T08:18:18","Vcc":3.176,"POWER":"ON","Wifi":{"AP":1,"SSId":"home2","BSSId":"84:16:F9:66:3A:7C","Channel":7,"RSSI":42}}}
```

**(Please, remember to close the issue when the problem has been addressed)**

Jason2866, 2018-11-04:
No, not supported from hardware

dan-cristian, 2018-11-04:
Thanks. Any other hardware that runs tasmota you are aware that supports this?

Jason2866, 2018-11-04:
As far as i know, none.

closed 2018-11-04 by andrethomas2