arendst/Tasmota #4272: Teckin SP22

FredF63, 2018-11-04:
Using the latest release with configuration 52 Teckin for the Teckin SP22 doesn't work well.
You see the plug in the main menu and can switch but no reaction on the plug.

At the moment it works with 45 Blitzwolf only.

arendst, 2018-11-04:
Does it work well when blitzwolf is selected? Does energy monitoring work too?

If so, leave it that way. The Teckin module uses different io for relay so your hardware is compatible with blitzwolf.

FredF63, 2018-11-04:
with blitzwolf it works, also the energy monitoring works

arendst, 2018-11-04:
So let's decide your Teckin SP22 is compatible with Blitzwolf.

Perhaps I have to update the release note to mention Teckin SP22 as Blitzwolf compatible if others can confirm.

ascillato2, 2018-11-05:
Hi,

Anyone that have this hardware can also confirm this? Thanks

edelstahlratte, 2018-11-06:
My SP22 is working only in Blitzwolf mode too... There seem to be different versions under the same name and type... Ergo: some SP22 are Blitzwolf compatible...

ascillato2, 2018-11-06:
@edelstahlratte Thanks a lot for your feedback. :+1:

closed 2018-11-06 by ascillato2

wunderbaum, 2018-11-29:
I have the Teckin SP22, too.
Basic question, how can it be opened as it has no screws?

smarthomecoder, 2018-12-21:
And closed up again. Just took a saw and some sanding paper after that.

If anybody likes to comment, what's inside and what "quality" this is, you are very welcome.
not yet flashed.

[image]
[image]

PanicMan, 2023-10-11:
@smarthomecoder Did you manage to flash it? Or replace the ESP?

smarthomecoder, 2023-10-11:
> @smarthomecoder Did you manage to flash it? Or replace the ESP?

Man, that is a journey back in time :-) Yes I flashed many of them, without opening the rest and they still run with tasmota and home assistant. Don't ask me how I did it. I know it was just a script and the rest went smoothly.
Have fun!

sfromis, 2023-10-11:
https://templates.blakadder.com/teckin_SP22.html implies that Tuya-convert was used originally. By now, current devices will commonly be using another MCU, not compatible with Tasmota, or have a Tuya version no longer vulnerable to the security vulnerabilities exploited by Tuya-convert to hack devices.