arendst/Tasmota #4276: Is it possible to change the separator in Backlog command?

badabing2005, 2018-11-04:
Hi,

The default separator in the Backlog command is `;` semicolon
Is it possible to override the separator?
If not is it possible to escape `;` characters in any of the supplied values?
One of my values has semi-colon in it and Backlog command does not work, as it treats it as field separator.

Thanks
BB

arendst, 2018-11-04:
Nope.

what value would need a ``;``

andrethomas2, 2018-11-04:
Closing this issue as it has been answered.

### Support Information
See [Wiki](https://github.com/arendst/Sonoff-Tasmota/wiki) for more information.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.

closed 2018-11-04 by andrethomas2

badabing2005, 2018-11-04:
> 
> 
> Nope.
> 
> what value would need a `;`

A password
In any case it's not a big deal.
I can do it in multiple commands, a little inconvenience but works.

Thanks
BB