arendst/Tasmota #4278: HX711 Problem, no reading

nightcat91, 2018-11-04:
Make sure these boxes are checked [x] before submitting your issue - Thank you!

- [x] Searched the problem in issues (https://github.com/arendst/Sonoff-Tasmota/issues)
- [x] Searched the problem in the wiki (https://github.com/arendst/Sonoff-Tasmota/wiki/Troubleshooting)
- [x] Searched the problem in the forum (https://groups.google.com/d/forum/sonoffusers)
- [x] Searched the problem in the chat (https://discord.gg/Ks2Kzd4)
- [x] Development/Compiler/Upload tools used : Atom /PlatformIO
- [x] Hardware used : NodeMCU
- [ ] If a pre-compiled release or development binary was used, which one? :
- [x] You have tried latest release or development binaries? : dev (v. 6.3.0.2)
- [x] Provide the output of command ``status 0`` :
```
23:17:23 CMD: status 0
23:17:23 MQT: stat/esp10/STATUS = {"Status":{"Module":18,"FriendlyName":["ESP 10 Bed Scale"],"Topic":"esp10","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":7,"SaveData":1,"SaveState":1,"SwitchTopic":"esp10switch","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
23:17:23 MQT: stat/esp10/STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"esps","OtaUrl":"http://thehackbox.org/tasmota/release/sonoff.bin","RestartReason":"External System","Uptime":"0T00:01:35","StartupUTC":"2018-11-04T22:15:48","Sleep":0,"BootCount":7,"SaveCount":12,"SaveAddress":"F8000"}}
23:17:23 MQT: stat/esp10/STATUS2 = {"StatusFWR":{"Version":"6.3.0.2","BuildDateTime":"2018-11-04T21:36:11","Boot":31,"Core":"2_3_0","SDK":"1.5.3(aec24ac9)"}}
23:17:23 MQT: stat/esp10/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["Naitsabes","sebsmarthome"],"TelePeriod":10,"SetOption":["00008009","55818000","00000000"]}}
23:17:23 MQT: stat/esp10/STATUS4 = {"StatusMEM":{"ProgramSize":464,"Free":536,"Heap":19,"ProgramFlashSize":1024,"FlashSize":4096,"FlashMode":3,"Features":["00000809","0FCAE584","240183A0","23B6179E","00003BC0"]}}
23:17:23 MQT: stat/esp10/STATUS5 = {"StatusNET":{"Hostname":"esp10-5249","IPAddress":"192.168.10.240","Gateway":"192.168.10.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.10.1","Mac":"2C:3A:E8:43:14:81","Webserver":2,"WifiConfig":2}}
23:17:23 MQT: stat/esp10/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.10.150","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_431481","MqttUser":"esp10","MqttType":1,"MAX_PACKET_SIZE":1000,"KEEPALIVE":15}}
23:17:23 MQT: stat/esp10/STATUS7 = {"StatusTIM":{"UTC":"Sun Nov 04 22:17:23 2018","Local":"Sun Nov 04 23:17:23 2018","StartDST":"Sun Mar 25 02:00:00 2018","EndDST":"Sun Oct 28 03:00:00 2018","Timezone":99,"Sunrise":"07:04","Sunset":"16:51"}}
23:17:23 MQT: stat/esp10/STATUS10 = {"StatusSNS":{"Time":"2018-11-04T23:17:23","HX711":{"Weight":0}}}
23:17:23 MQT: stat/esp10/STATUS11 = {"StatusSTS":{"Time":"2018-11-04T23:17:23","Uptime":"0T00:01:35","Vcc":2.761,"Wifi":{"AP":1,"SSId":"Naitsabes","BSSId":"C8:0E:14:02:3D:E1","Channel":8,"RSSI":64}}}
```

I was excited when I saw that the HX711 got added to the project.
Because I have 4 weight scales around here that I wanted to use to build a bed scale to determine when someone is in the bed to use in HomeAssistant (for automations).

These are the load cells I use:
https://www.aliexpress.com/item/DIY-50Kg-Body-Load-Cell-Weighing-Sensor-Resistance-strain-Half-bridge-Original/32671637025.html
And this is the HX711:
https://www.aliexpress.com/item/HX711-Weighing-Sensor-Dual-Channel-24-Bit-Precision-A-D-Module-Pressure-Sensor/1859015792.html

My first try a few month back was using this project:
https://github.com/Skaronator/ESP8266-Load-Cell
But that never worked... I never got any reading other than "0" from that.... so I gave up.

Now I set everything back up and flashed a NodeMCU with the lastest version of Tasmota (v. 6.3.0.2) and configured everything.

I made sure the connections from the NodeMCU to the HX711 were correct and I used this diagram to wire up the load cells:
https://selfhostedhome.com/wp-content/uploads/2018/09/schematic-1024x501.png

But like before I am only getting a 0 as the weight displayed on the website of tasmota (and also in the sensor mqtt message).

I checked all my load cells to make sure the resistors inside where all right.
According to this (https://electronics.stackexchange.com/a/199907) all my load cells are correct, 1kOhm across the middle to sides (red to white/black), and 2kOhm across the sides (white to black)

The only thing I was able to find is that when I unplug one of the red wires from the HX711 is shows 6kg as weight... once I plug it back in it goes back to 0kg

Any help is welcome, thanks

nightcat91, 2018-11-05:
> I used this diagram to wire up the load cells:
> https://selfhostedhome.com/wp-content/uploads/2018/09/schematic-1024x501.png

mike2nl, 2018-11-05:
@nightcat91 
when you are online in discord in the evening, please talk to me. Same avatar name as here in discord.

ascillato2, 2018-11-05:
Hi, have you managed to solve your issue in the chat?

nightcat91, 2018-11-05:
No, not yet, @mike2nl was not online in discord the last few hours. Now its past midnight for me, so time for bed. Maybe tomorrow

mike2nl, 2018-11-06:
@ni
sorry had some health issues after i tested an updated library. 
I was not able to think straight anymore.
But today evening i will be online. Sorry again.

nightcat91, 2018-11-06:
@mike2nl no problem. I hope you are alright and well
I will be online in Discord around 7pm german time (10 hours from now)

ascillato2, 2018-11-14:
Hi

Have you managed to solve your issue?

nightcat91, 2018-11-14:
I did not get a solution yet, @mike2nl hasnt been online in discord last week.
After getting zero as a reading I did try to add weight but that did not change anything...
Like I wrote in the first post, the only way to see something else as zero is to unplug one of the red (center) cables from one of the load cells...

Jason2866, 2018-11-15:
I just ordered (from Ali...) a Load Cell and a HX711. Need a little time, but i am sure we get things sorted.
Mike2NL isnt much online since a time...

nightcat91, 2018-11-15:
@Jason2866 
Did you get the the set of 4 load cells (with 3 wires each), because the problem might be connected to the fact that I have to "bridge" them to connect them all to the HX711. 
This is different for other load cells (bigger ones) where they have 4 wires on one load cell

Jason2866, 2018-11-15:
Left you Infos on Discord

mimquadrat, 2018-11-17:
@nightcat91 
can you tell me how you did it in the end (3 or 4 wires, how wired,...)? 
I am a German beekeeper and need just a little help with the load cells...

Jason2866, 2018-11-17:
https://www.instructables.com/id/Arduino-Bathroom-Scale-With-50-Kg-Load-Cells-and-H/
Wiring for 4 half cells

nightcat91, 2018-11-17:
@mimquadrat 
what @Jason2866 posted or this:
https://selfhostedhome.com/wp-content/uploads/2018/09/schematic-1024x501.png
both show the same, just a differet picture!!

mimquadrat, 2018-11-17:
@nightcat91 @Jason2866 
Many thanks gentlemen, but as far is i know the load cells with 4 wires are much "better" than the ones with 3 wires.

Jason2866, 2018-11-17:
Not better, different. It will work with both.

nightcat91, 2018-11-17:
4 wire pictures can be found here:
https://github.com/arendst/Sonoff-Tasmota/issues/3502

I just connected my load cells (each 3 wire) according to the wire schematic from above and after a lot of trial and error it seems to be working now...
I will post a hopefully detailed explanaition of what I found out here later today

nightcat91, 2018-11-17:
Ok, so for anyone thats trying to get the HX711 to work using 3 wire sensors as found in my first post here is what I found out together with the help of @Jason2866 

First of all, here is a list of commands that is very helpful when working with the scale, these can be used in the console or over mqtt:
https://github.com/arendst/Sonoff-Tasmota/commit/fcc6d5437b1af84f39756a2056bb978cf17a64f2#diff-28f788983411fa5bb7d503298c3802c3R132

I used this picture to wire my load cells:
https://selfhostedhome.com/wp-content/uploads/2018/09/schematic-1024x501.png

It seems that you may need to change the two red wires going to A+ and A- on the HX711 if you only get a reading of 0kg. Apparently it can happen that the scale is measuring negative values but the GUI will not display negativ kg values, so you only get 0kg,
So try to switch A+ and A- on the HX711 and see if that helps.

You need to do the calibration and (at least for me) the calibration always failed with the two red wires from A+ and A- beeing one way, so I had to switch them to be able to do the calibation and then switch them back to have the scale do positive readings and not negative.
After that I did a "Sensor34 1" to reset the scale to 0kg and then I got correct readings...

Regarding calibration:
The calibration is somewhat strange as you cannot remove the reference weight when it tells you to do so, wait 1 second longer and remove it then, otherwise my calibration would always fail. 
See this issue I opened about more details and maybe an answer to when to exactly remove the reference weight.
https://github.com/arendst/Sonoff-Tasmota/issues/4373

I hope this is helpful to someone and if you dont undestand it just let me know and I might be able to explain it in discord...

closed 2018-11-17 by nightcat91

pejotigrek, 2019-05-02:
@nightcat91 do you happen to know if the cable lenght could be a problem with readings? I did soldered a longer ones to fit my bed.

I've got everything connected the same way you have. I tried switching A+ with A- lots of times. I've already lost my count of connection checks that I did to be sure everything is ok...

and still 0 kg readout & calibration failed every single time...
after few hours of trying to figure it out, I'm losing my mind.

    14:17:33 CMD: status 0
    14:17:33 MQT: stat/sypialnia/wyro/STATUS = {"Status":{"Module":18,"FriendlyName":["WYro"],"Topic":"sypialnia/wyro","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
    14:17:33 MQT: stat/sypialnia/wyro/STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"sonoffs","OtaUrl":"http://sonoff.maddox.co.uk/tasmota/sonoff.bin","RestartReason":"Software/System restart","Uptime":"0T00:05:24","StartupUTC":"2019-05-02T13:12:09","Sleep":50,"CfgHolder":4617,"BootCount":34,"SaveCount":80,"SaveAddress":"F5000"}}
    14:17:33 MQT: stat/sypialnia/wyro/STATUS2 = {"StatusFWR":{"Version":"6.5.0(release-sonoff)","BuildDateTime":"2019-03-19T12:24:10","Boot":31,"Core":"2_3_0","SDK":"1.5.3(aec24ac9)"}}
    14:17:33 MQT: stat/sypialnia/wyro/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["REDACTED",""],"TelePeriod":10,"Resolution":"558180C0","SetOption":["00008009","280500000100000000000000000000000000","00000000"]}}
    14:17:33 MQT: stat/sypialnia/wyro/STATUS4 = {"StatusMEM":{"ProgramSize":507,"Free":496,"Heap":14,"ProgramFlashSize":1024,"FlashSize":4096,"FlashChipId":"16405E","FlashMode":255,"Features":["00000809","0FDAE394","000783A0","23B617CE","00003BC0"]}}
    14:17:33 MQT: stat/sypialnia/wyro/STATUS5 = {"StatusNET":{"Hostname":"WYro","IPAddress":"REDACTED","Gateway":"REDACTED","Subnetmask":"255.255.255.0","DNSServer":"REDACTED","Mac":"REDACTED","Webserver":2,"WifiConfig":5}}
    14:17:33 MQT: stat/sypialnia/wyro/STATUS6 = {"StatusMQT":{"MqttHost":"REDACTED","MqttPort":1883,"MqttClientMask":"WYro","MqttClient":"WYro","MqttUser":"REDACTED","MqttCount":1,"MAX_PACKET_SIZE":1000,"KEEPALIVE":15}}
    14:17:33 MQT: stat/sypialnia/wyro/STATUS7 = {"StatusTIM":{"UTC":"Thu May 02 13:17:33 2019","Local":"Thu May 02 14:17:33 2019","StartDST":"Sun Mar 31 02:00:00 2019","EndDST":"Sun Oct 27 03:00:00 2019","Timezone":"+01:00","Sunrise":"05:28","Sunset":"20:06"}}
    14:17:33 MQT: stat/sypialnia/wyro/STATUS10 = {"StatusSNS":{"Time":"2019-05-02T14:17:33","HX711":{"Weight":0}}}
    14:17:33 MQT: stat/sypialnia/wyro/STATUS11 = {"StatusSTS":{"Time":"2019-05-02T14:17:33","Uptime":"0T00:05:24","Vcc":2.762,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":20,"Wifi":{REDACTED}}}

nightcat91, 2019-05-02:
all I can help you with is what I already wrote above.
But regarding long cables, yes I did have problems as well with my working setup, once I switched to long cables, so yes, the cable length can be a factor.
But for me the long cables only made strange readings but not a constant 0kg. just strange readings all over the place with long cables.

luisaga215, 2019-05-03:
Hi!
I'm having exactly the same issue and I've checked over it a thousand times, I thought it was the cells that were failing so I tried to "fake" a load cell by changing the resistors in the bridge but the lecture stays in 0.

Also when I switch the cables the reading goes to 1 but still doesn't change with pressure.

//PosibleSolution
Yesterday I noticed that if I use the B+ and B- terminals I get a reading!! So I tested it by changing voltages in B terminals and the readings changed. The sad part is that when I tested it with the cells it didn't work (so the cells might be the problem). I'm gonna keep trying this today, but I'm also beginning to look for another amplifier configuration (if you have any ideas about this please let me know!)

I'm gonna try (again) to change cables and calibrate... keep in touch

KICSAutomation, 2019-05-08:
Hi can you send us the Sensor34 4 Message response ? ( in the Console )
for example when sending this command Sensor34 3

KICSAutomation, 2019-05-08:
Hi,
Can you send the response message of this command ?       Sensor34 3

pejotigrek, 2019-05-08:
@KICSAutomation not sure whom did you asked, but here's mine:

    22:56:54 CMD: sensor34 3
    22:56:54 MQT: stat/sypialnia/wyro/RESULT = {"Sensor34":{"WeightRef":4100,"WeightCal":4092,"WeightMax":200000,"WeightItem":0.0}}

for me it looks like it's calibrated good [I set the calibration weight to 4100g, because I had this measured; looks like sensors "felt" 4092g, so very close; max is set to 200kg - from 4x 50kg sensors] but it still can't read current items placed on scale.

KICSAutomation, 2019-05-08:
{ but it still can't read current items placed on scale.}
you meant it reads but with wrong weight ? or doesn't read at all...

KICSAutomation, 2019-05-08:
can you send the 4 load cells connection diagram to the HX711 module ?

pejotigrek, 2019-05-08:
> you meant it reads but with wrong weight ? or doesn't read at all...

> can you send the 4 load cells connection diagram to the HX711 module ?

please look [here](https://github.com/arendst/Sonoff-Tasmota/issues/4278#issuecomment-488672143) - it's my first comment on this issue, 6 days ago. this is still valid [connection diagram is the same as OP's, no reason to paste it again; calibration still fails; still 0kg readout].

BUT output of the `sensor34 3` command **LOOKS like** it's calibrated ok [this is what I understand from te "WeightCal" variable - I guess it shows what was read during calibration, right?], which makes no sense because all I got during calibration was "calibration failed" and sensors show 0kg every time I try to weight something.

KICSAutomation, 2019-05-08:
in my opinion, to calibrate any load cell controller you have to use at least >30% of the max weight to have a good resolution scale...
here the WeightMax=200kg and the WeightRef=4.1kg
if you want to use the load cell to weigh up to 200kg, i suggest you calibrate it with at least 30-40kg WeightRef...

or change manually  the "WeightRef","WeightCal","WeightMax" to check if sensor's value will change.

pejotigrek, 2019-05-08:
> in my opinion

out of curiousity - you're talking from experience, or just theoretically? because I had WeightMax set up earlier at 10kg, 20kg, 50kg, 100kg - with the same results. also I played with all of `sensor34` options [from the wiki](https://github.com/arendst/Sonoff-Tasmota/wiki/Commands#sensor) and nothing changed the behaviour of my sensors [got 12 of them, and got 4 HX711 modules. tried them with waaaay too many combinations, to cross out failure of this or that element].

I don't want to have good resolution or reliable scale. I just want to have ANY readout, to have the possibility of sensing presence in the bed.

KICSAutomation, 2019-05-09:
sure, I'm talking from experience. I had installed a lot of load cells and pressure sensors which have the same HX711 chip in it, all are working as expected...

did you try to setup only one load cell to check if the HX711 is working or no ?

pejotigrek, 2019-05-09:
> did you try to setup only one load cell to check if the HX711 is working or no ?

yep. even tried with different wemos board.. that's why it's frustrating everytime I try to work it out.
good thing it's not a highly needed project with top priority, but still..

KICSAutomation, 2019-05-09:
Just for testing I configured my HX711 module now as shown below, with a NodeMCU 8266.
(without connecting the load cell) if I touch the module A+ A- E+ E- signals with my finger, the value is changing for arround 200kg...
[image]

{"Sensor34":{"WeightRef":4100,"WeightCal":10,"WeightMax":1000,"WeightItem":0.0}}

pejotigrek, 2019-05-09:
I just ordered another couple of sensors & modules. it would be total bad luck if all mine hardware was failed, but right now it's my only idea.. this, or wires problem [broken somewhere?] - in the end testing new sensors will be less frustrating than playing with wires inch by inch.. ;)

frodo1990, 2019-12-03:
Hello, yesterday I tried to get the HX711 up and running. The calibrate works and a weight of 3Kg are also displayed. That works great. But as soon as I take the ESP from the power or reboot, then the display shows 0Kg. Even if the weight of 3Kg is still on the sensors.

I would like to build the sensors under a water softener, so that I can read the weight of the salt tablets and send it via MQTT.

How can I save the value after a reboot? I have the Tasmota version 7.1.1 on the ESP.

KICSAutomation, 2019-12-04:
Don't know if you have a problem in the system but you can save any value in the eeprom, Check in the command rules Mem<x> for saving the value before power off.

lalo-uy, 2019-12-05:
> in the eeprom, Check in the command rules Mem for saving the value before

You can not save a value before power off, if you don't know when the mains
is going to fail.
If that very same eeprom place holds the calibration scale value , there is
no need to write the eeprom on each power cycle.

RugbyAl, 2020-01-06:
Did anyone manage to get a definitive solution for this?  With the wiring as recommended, I get weight 0 all the time. with A+ / A- reversed, I still get weight zero, but if i stand on the scales get weight 1.  Can't get calibration to work at all.
Tried 2 D1 minis, 2 HX711s, many permutations re GPIO assignments.

lalo-uy, 2020-01-09:
Why don’t you try an arduino script to be sure the hardware is ok?

RugbyAl, 2020-01-10:
Thanks for your reply, it was actually e being stupid, mounted the hardware incorrectly, couldn get deflection!

hummeruli, 2020-03-18:
I think the biggest problem on this is the wrong wiring. I also had the same failure. So I used a Arduino Nano and see that the vallues are negative. Then I changed the E+ wire to E- and now it works. After that I connected the Wemos D1 Mini and tasmota works. 
These steps I've done:
I put a smaller weight on it
sensor34 1  // reset
sensor34 3 8000  // my reference weight
sensor34 2  // calibration
Now I removed the small one and put the reference on
Finished.
My only problem is that on the website came the weights I put on very exactly, but only 0 over MQTT. I received messages but every time only after I removed all from it.

andreasloe, 2020-07-11:
This guy says that the missing connection between E- and GND is the problem
[https://www.eevblog.com/forum/projects/how-can-a-chip-(hx711)-work-without-a-ground-pin-connected/](url)
I do not believe that changing E+ and E- solves the issue: These are simple resistors in the end and if you look at the wiring already changing „upper left“ with „lower right“ or switching the load cells over would result in the same wiring as a change between E+ and E-. And simply switching the load cells does not change any current! So it must be some kind of a hardware failure and this is where the above webpage points to.

andreasloe, 2020-07-12:
Unfortunately, that didn't fo the trick. I connected E- do GND and got indeed different values. Before that, weight raw was around -800; no it is 934 with 25kg on the scale (four 25kg-weight cells) and 1001 with nothing on the scale. Also, LoadAvg decreased from 200 to 9 (sleep was 100). But still, this is far from being accurate. 

So, E-has to be connected to GND, but there are further problems and I am now sure that it has to do with the hardware. I am using these green HX711 with no description (besides J1 and J2) on it that are sold by the thousands. I will check a different *.ino file on my wemos and then change to another HX711. In the above link at least three deviations from the data sheet are indicated.

andreasloe, 2020-07-12:
I just installed [this library](https://github.com/olkal/HX711_ADC) tried to connect to HX711 using two Leolin Wemos. To no avail. So I am giving up on those green HX711, they are simply bad. We already know at least one wiring error, so why should this hardware work at all?

Barabba11, 2021-05-05:
If may can be helpful to someone, I've considered this library the best for arduino https://github.com/olkal/HX711_ADC
I've solved my zero read by wiring differently the sensor, red and green where crossed/wrongly colored (meause with tested and be sure the load is balanced)