arendst/Tasmota #4285: Counter

marrobHD, 2018-11-05:
How can I get the counter value via mqtt? Exists any setting to reset the counter after e.g. 2 seconds?
Thanks for help

ascillato, 2018-11-05:
Yes, please see commands in the wiki

ascillato2, 2018-11-05:
Closing this issue as it has been answered.

### Support Information
See [Wiki](https://github.com/arendst/Sonoff-Tasmota/wiki) for more information.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.

closed 2018-11-05 by ascillato2

marrobHD, 2018-11-05:
I cant find this in the wiki

ascillato2, 2018-11-05:
https://github.com/arendst/Sonoff-Tasmota/wiki/Commands#sensor

ascillato2, 2018-11-05:
For general questions, please use the support chat. Github issues are for bug reporting and using the provided template. Thanks.

marrobHD, 2018-11-05:
there is not declared mqtt and autoreset