arendst/Tasmota #4286: Fatal exception while using counter and rule

kajmaj, 2018-11-05:
I am using Wemos D1 UNO with switch connected and used as counter.
Tamota version: 6.3.0.1, Core/SDK Version | 2_3_0/1.5.3
Counter causes repeatedly fatal error: 
`17:35:12 MQT: tele/wemos_3/INFO3 = {"RestartReason":"Fatal exception:0 flag:2 (EXCEPTION) epc1:0x402180d8 epc2:0x00000000 epc3:0x00000000 excvaddr:0x00000000 depc:0x00000000"}`
Currently connected GPIO 14, 10K pull up resistor connected to 3.3V.
Rule used: 
`on tele-counter#c1>0 do publish stat/wind/counter1 %value% endon on tele-counter#c1>0 do counter1 0 endon`

I´ve tried differend power plugs to eliminate power failure, different GPIOs (with or without pull down resistor 4K7 or 10K) and 2 GPIOs with already installed 10K pull downs

Is it possible to define a cause?

Jason2866, 2018-11-05:
What have you connected and how?
Precompiled firmware?

ascillato, 2018-11-05:
Please, complete the troubleshooting template in order to have the required infornation so as to properly help you

ascillato, 2018-11-05:
Sometimes an exception is out of memory, so we need the template completed to gather this type of information (features chosen, self compiled or pre compiled, etc)

andrethomas, 2018-11-05:
You also need to provide the map file or the firmware used so that we can see where epc1:0x402180d8 is originating from.

kajmaj, 2018-11-05:
Sorry for my negligence:

- [X ] Development/Compiler/Upload tools used : PlatformIO (latest version)
- [X ] Hardware used : Wemos D1 UNO R2
- [X ] If a pre-compiled release or development binary was used, which one? :

This error is with sonoff.bin (6.3.0) from http://thehackbox.org/tasmota/release/ not modified, flashed OTA:
`18:34:07 MQT: tele/wemos_3/INFO3 = {"RestartReason":"Fatal exception:0 flag:2 (EXCEPTION) epc1:0x4020c678 epc2:0x00000000 epc3:0x00000000 excvaddr:0x00000000 depc:0x00000000"}` 

This one is with development version 6.3.0.1 self compilled using PlatformIO and flashed OTA: 
`17:35:12 MQT: tele/wemos_3/INFO3 = {"RestartReason":"Fatal exception:0 flag:2 (EXCEPTION) epc1:0x402180d8 epc2:0x00000000 epc3:0x00000000 excvaddr:0x00000000 depc:0x00000000"}` 

- [ X] You have tried latest release or development binaries? : latest release and development as well

- [ X] Provide the output of command ``status 0`` :
`18:44:12 CMD: status 0
18:44:12 MQT: stat/wemos_3/STATUS = {"Status":{"Module":18,"FriendlyName":["wemos_3"],"Topic":"wemos_3","ButtonTopic":"0","Power":0,"PowerOnState":0,"LedState":1,"SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
18:44:12 MQT: stat/wemos_3/STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"sonoffs","OtaUrl":"http://192.168.1.107:9541/data/firmwares/sonoff.bin","RestartReason":"Exception","Uptime":"0T00:02:24","StartupUTC":"2018-11-05T16:41:48","Sleep":50,"BootCount":1261,"SaveCount":1546,"SaveAddress":"F9000"}}
18:44:12 MQT: stat/wemos_3/STATUS2 = {"StatusFWR":{"Version":"6.3.0","BuildDateTime":"2018-10-30T17:33:53","Boot":31,"Core":"2_3_0","SDK":"1.5.3(aec24ac9)"}}
18:44:12 MQT: stat/wemos_3/STATUS3 = {"StatusLOG":{"SerialLog":0,"WebLog":2,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["HugoNET",""],"TelePeriod":20,"SetOption":["00008009","55818000","00000001"]}}
18:44:12 MQT: stat/wemos_3/STATUS4 = {"StatusMEM":{"ProgramSize":489,"Free":512,"Heap":15,"ProgramFlashSize":1024,"FlashSize":4096,"FlashMode":3,"Features":["00000809","0FDAE794","000183A0","23B617CE","00003BC0"]}}
18:44:12 MQT: stat/wemos_3/STATUS5 = {"StatusNET":{"Hostname":"wemos_3-1229","IPAddress":"192.168.1.73","Gateway":"192.168.1.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.1.1","Mac":"84:F3:EB:16:C4:CD","Webserver":2,"WifiConfig":5}}
18:44:12 MQT: stat/wemos_3/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.1.107","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_16C4CD","MqttUser":"admin","MqttType":1,"MAX_PACKET_SIZE":1000,"KEEPALIVE":15}}
18:44:12 MQT: stat/wemos_3/STATUS7 = {"StatusTIM":{"UTC":"Mon Nov 05 16:44:12 2018","Local":"Mon Nov 05 18:44:12 2018","StartDST":"Sun Mar 25 02:00:00 2018","EndDST":"Sun Oct 28 03:00:00 2018","Timezone":2,"Sunrise":"07:45","Sunset":"17:23"}}
18:44:12 MQT: stat/wemos_3/STATUS10 = {"StatusSNS":{"Time":"2018-11-05T18:44:12","COUNTER":{"C1":86,"C2":1}}}
18:44:12 MQT: stat/wemos_3/STATUS11 = {"StatusSTS":{"Time":"2018-11-05T18:44:12","Uptime":"0T00:02:24","Vcc":2.769,"Wifi":{"AP":1,"SSId":"XXXNET","BSSId":"08:96:D7:2A:01:D8","Channel":10,"RSSI":50}}}`

Connected is a wind sensor (a reed switch) by an approx 3 m long shielded wire, connected to wemos by DUPOND connectors

andrethomas, 2018-11-05:
According to map file, exception eminates from
`4020c678 g     F .irom0.text	00000014 _Z14CounterUpdate1v`

@ascillato bug?

andrethomas, 2018-11-05:
Please try to modify the rule to

`on tele-counter#c1>0 do backlog publish stat/wind/counter1 %value%;counter1 0 endon`

and report back the outcome

kajmaj, 2018-11-05:
Modified, will report back ....

kajmaj, 2018-11-05:
Issue persists:

`00:00:06 MQT: tele/wemos_3/INFO3 = {"RestartReason":"Fatal exception:0 flag:2 (EXCEPTION) epc1:0x4020c678 epc2:0x00000000 epc3:0x00000000 excvaddr:0x00000000 depc:0x00000000"}`

andrethomas, 2018-11-05:
please provide output of command

rule

kajmaj, 2018-11-05:
`18:59:51 RUL: TELE-COUNTER#C1>0 performs "backlog publish stat/wind/counter1 86;counter1 0"
18:59:51 MQT: stat/wind/counter1 = 86
18:59:51 MQT: stat/wemos_3/RESULT = {"Counter1":0}`

andrethomas, 2018-11-05:
That looks like it worked

Works for me also
```
19:00:51 RUL: TELE-COUNTER#C1>0 performs "backlog publish stat/wind/counter1 15;counter1 0"
19:00:51 MQT: stat/wind/counter1 = 15
19:00:51 MQT: stat/sonoff/RESULT = {"Counter1":0}
```

What I meant is the output produced when you send command "rule" on the console - i want to see all the rules

andrethomas, 2018-11-05:
And what is the other side of the reed switch connected to, GND ?

kajmaj, 2018-11-05:
`19:07:19 CMD: Rule1
19:07:19 MQT: stat/wemos_3/RESULT = {"Rule1":"ON","Once":"OFF","StopOnError":"OFF","Free":428,"Rules":"on tele-counter#c1>0 do backlog publish stat/wind/counter1 %value%;counter1 0 endon"}`

Your rule works in the same way as mine.

Reed switch is connected between GND and GPIO 14 on Wemos. Pull up resistor is connected between 3.3V PIN and GPIO 14.
Error appears in both cases - with pullup/ without pullup

andrethomas, 2018-11-05:
I cannot reproduce the error with a reed switch setup in the same way as you - I do not have a pull-up resistor because I use the internal one. Try disconnect and disable any other inputs you have and see if it can reproduce an error, mine is like this without a pull-up resistor and reed switch between GPIO14 and gnd

[image]

andrethomas, 2018-11-05:
I cannot reproduce your exception.

Let's assume there is something else weird in the config.

Do command:
reset 5
This will erase the data portion of the flash, restore to defaults from user_config.h and retain your wifi settings and reboot.

Then configure the input pin again and see if you can reproduce.

To make sure we're on the same page please use the binary from http://thehackbox.org/tasmota/release/sonoff.bin (6.3.0 release) so that we know our firmware is the same. If you are currently using a different binary please upgrade OTA first, then reset 5 and reconfigure as above.

If the problem persists, try it on another pin and revert back to us.

kajmaj, 2018-11-05:
I will try to reset 5 and report
Nonetheless:
I am on 6.3.0 => http://thehackbox.org/tasmota/release/sonoff.bin
Error appears only if reed switch produces pulses. 
I've already tested more pins with no difference.

andrethomas, 2018-11-05:
Sorry, can't reproduce - you will wait to see if someone else like @ascillato can reproduce the problem.

kajmaj, 2018-11-05:
Thanks anyhow :)

ascillato, 2018-11-06:
Sorry, I can't reproduce this either. Tasmota 6.3.0 and 6.3.0.3 with NodeMCU.

Have tested in several devices your rule and works fine.

Besides, I'm using a very similar rule for my watering system to measure the water flow and never had issues with that.

Have you done an erase before flash? Seems like a corrupted flashed firmware.

ascillato2, 2018-11-06:
When you have time, please report back if erasing and flashing the firmware again has solved your issue. Please, try the last precompiled bin from development branch. Thanks

### Support Information
See [Wiki](https://github.com/arendst/Sonoff-Tasmota/wiki) for more information.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.

closed 2018-11-06 by ascillato2

kajmaj, 2018-11-13:
Solved !! 
It seems to be caused by an active `Sleep`
I´ve disabled it (set to 0)  and fatal exception with related restart reboots is gone, OK it seems to be gone. At the moment it is working more than 3 hrs w/o problems/reboots.
So as result I did not need to reset flash an do full reflash.

andrethomas, 2018-11-13:
@kajmaj Very nice find! Thanks for providing the feedback so we will remember it for the future.

When I tested my devices are always pre-compiled for sleep 5 so I did not have issues on sleep 5 - can you confirm that it is only if you go as high as 50 or does it work for you on sleep 5 also?

kajmaj, 2018-11-13:
@andrethomas Just retested it. 5, 10 and 50 causes fatal exception, well at leat for my Wemos.

Furthermore while testing device (Wemos D1 UNO) I found that some GPIOs numbers on device WebUI configuration page are connected to another GPIOs in reality.  (tested on 2 devices)
If anybody is interested ( Theo, Adrian ?) I can share it.

andrethomas, 2018-11-13:
Thanks

The GPIO to Dx mapping is for the ESP8266 Generic Module itself - The Chinese manufacturers sometimes choose different D numbers for reasons we will never comprehend.

I have some boards that have two D2 pins, but in reality one of them is D1 ;)

kajmaj, 2018-11-14:
OK. Lets forget strange GPIOs mapping :)

kajmaj, 2019-04-27:
Small update:
I needed to connect INA219 to the related wemos -> I´ve upgraded (clean flash) to  sonoff-sensors (6.5.0 2.3.0) , did the same settings as on previous version and Fatal Exception is back. 
I noticed that (at least in my case) a combination [http://thehackbox.org/tasmota/release/sonoff-sensors.bin ](url) with counter causes repeated Fatal Exception.
If I flashed (OTA) back to sonoff-basic, everything is OK again.
Might be there any bug in FW?

Yuriyvp, 2019-10-13:
Good day all.
I would like to report same problem. Reproduced successfully on sonoff basic r1 and r2.
GPIO1 and GPIO3 connected to hall water flow sensors.
With counters enable - restart by Fatal Exception. Fix by command <sleep 0>