arendst/Tasmota #4290: Using a Display with Tasmota (question)

helmar74, 2018-11-06:
Hi,

I want to play around with Displays connected to Tasmota.
I'm not sure how to start. 
I want to try I2C Displays and E-Paper Displays.
For E-Paper Display, I found that one:
[https://www.ebay.de/itm/Waveshare-2-9-E-Ink-E-Paper-Display-Modul-296x128-SPI-Arduino-Raspberry-Pi-LCD/162968194447?_trkparms=aid%3D555018%26algo%3DPL.SIM%26ao%3D1%26asc%3D20140117130753%26meid%3Da477581f8d0f47a6a6e6ea7e1b21cfa8%26pid%3D100005%26rk%3D6%26rkt%3D10%26sd%3D273075802824%26itm%3D162968194447&_trksid=p2047675.c100005.m1851]

As a I2C Display  I found that one:
[https://www.ebay.de/itm/0-96-inch-128X64-OLED-LCD-Display-I2C-IIC-SPI-Gelb-Blau-SSD1306-Arduino-STM32/283225014720?hash=item41f186c1c0:g:T3QAAOSwlxRaZ471:rk:9:pf:0]

My questions (for using I2C Display or E-Paper) are:
1. Can I use the Displays above with Wemos Mini D and Tasmota?
2. Which Options Do I have to turn on in my_user_config.h?
3. Which GPIOs do I have to use?
4. How must I connect the Display Ports to the GPIOs (Layout)?
5. Which Settings for the GPIOs do I have to use in Tasmota?

Thank you very much for you help!

Make sure these boxes are checked [x] before submitting your issue - Thank you!

- [X] Searched the problem in issues (https://github.com/arendst/Sonoff-Tasmota/issues)
- [X] Searched the problem in the wiki (https://github.com/arendst/Sonoff-Tasmota/wiki/Troubleshooting)
- [X] Searched the problem in the forum (https://groups.google.com/d/forum/sonoffusers)
- [X] Searched the problem in the chat (https://discord.gg/Ks2Kzd4)
- [X] Development/Compiler/Upload tools used : Arduino 1.8.7 ESP Libary 2.3.0
- [X] Hardware used : Wemos Mini
- [X] If a pre-compiled release or development binary was used, which one? :
- [X] You have tried latest release or development binaries? :
- [X] Provide the output of command ``status 0`` :
```
19:08:42 CMD: status 0
19:08:43 MQT: stat/sonoffWemosMini_2/STATUS = {"Status":{"Module":18,"FriendlyName":["WemosMini-2"],"Topic":"sonoffWemosMini_2","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
19:08:43 MQT: stat/sonoffWemosMini_2/STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"sonoffWemosMiniStandardGruppe","OtaUrl":"http://192.168.x.x:5555/data/firmwares/sonoff.bin","RestartReason":"Software/System restart","Uptime":"0T00:07:45","StartupUTC":"2018-11-06T09:00:58","Sleep":0,"BootCount":50,"SaveCount":87,"SaveAddress":"FA000"}}
19:08:43 MQT: stat/sonoffWemosMini_2/STATUS2 = {"StatusFWR":{"Version":"6.3.0","BuildDateTime":"2018.11.06 09:30:47","Boot":31,"Core":"2_3_0","SDK":"1.5.3(aec24ac9)"}}
19:08:43 MQT: stat/sonoffWemosMini_2/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"SysLog":0,"LogHost":"192.168.x.x54","LogPort":514,"SSId":["SSID","sonoffap"],"TelePeriod":300,"SetOption":["00008009","55818000","00000001"]}}
19:08:43 MQT: stat/sonoffWemosMini_2/STATUS4 = {"StatusMEM":{"ProgramSize":520,"Free":480,"Heap":11,"ProgramFlashSize":1024,"FlashSize":4096,"FlashMode":3,"Features":["00000407","0FEAE79C","000187A0","23BE1FCE","00003BC0"]}}
19:08:43 MQT: stat/sonoffWemosMini_2/STATUS5 = {"StatusNET":{"Hostname":"sonoffWemosMini-2","IPAddress":"192.168.1.xxx","Gateway":"192.168.x.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.x.x","Mac":"DC:4F:22:1D:8F:49","Webserver":2,"WifiConfig":3}}
19:08:43 MQT: stat/sonoffWemosMini_2/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.1.50","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_1D8F49","MqttUser":"DVES_USER","MqttType":1,"MAX_PACKET_SIZE":1000,"KEEPALIVE":15}}
19:08:43 MQT: stat/sonoffWemosMini_2/STATUS7 = {"StatusTIM":{"UTC":"Tue Nov 06 09:08:43 2018","Local":"Tue Nov 06 19:08:43 2018","StartDST":"Sun Mar 25 02:00:00 2018","EndDST":"Sun Oct 28 03:00:00 2018","Timezone":99,"Sunrise":"07:03","Sunset":"16:42"}}
19:08:43 MQT: stat/sonoffWemosMini_2/STATUS10 = {"StatusSNS":{"Time":"2018-11-06T19:08:43","AM2301":{"Temperature":19.4,"Humidity":47.4},"TempUnit":"C"}}
19:08:43 MQT: stat/sonoffWemosMini_2/STATUS11 = {"StatusSTS":{"Time":"2018-11-06T19:08:43","Uptime":"0T00:07:45","Vcc":2.833,"Wifi":{"AP":1,"SSId":"SSID","BSSId":"E8:DF:70:67:B9:00","Channel":1,"RSSI":54}}}
```

**(Please, remember to close the issue when the problem has been addressed)**

helmar74, 2018-11-06:
Thanks for your help, of course I checked these infos before.
I ordered the displays, let's see, what will happen...

Which settings do I have to choose for a E-Paper (SPI) Display?
I get the following options:
[image]

curzon01, 2018-11-06:
First for E-Paper you need to compile it using USE_SPI, USE_DISPLAY and USE_DISPLAY_EPAPER
Three wires are connected as described CS, SCLK, MOSI

Then take a look into xdsp_05_epaper.ino:
    epd.cs_pin = pin[GPIO_SPI_CS];
    epd.mosi_pin = pin[GPIO_SPI_MOSI];  // 13
    epd.sclk_pin = pin[GPIO_SPI_CLK];   // 14

MOSI/SCLK are fixed to GPIO13 (MOSI) and GPIO14 (SCLK). SPI CS has to be user selected as `66 SPI CS` from module configuration (use any GPIO except GPIO13 and GIPO14).

helmar74, 2018-11-06:
Thanks for your help.
USE_SPI and  USE_DISPLAY I can find in my_user_config.h, but there's no USE_DISPLAY_EPAPER
(Tasmota Version 6.3.0)
-> Should I enter it  manually in my_user_config.h?

Jason2866, 2018-11-06:
ePaper Display support is only in development branch

helmar74, 2018-11-06:
ahh. Thanks a lot. Will try this

ascillato2, 2018-11-06:
Closing this issue as it has been answered.

### Support Information
See [Wiki](https://github.com/arendst/Sonoff-Tasmota/wiki) for more information.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.

closed 2018-11-06 by ascillato2

Makk1983, 2019-01-29:
Good evening, I'm trying to connect an i2c 1602 display to an esp12 with tasmota, from the first tests seems to work all, the only thing I do not understand is why with the command displaymode 1 instead of seeing date and time HH: HH: HH DD: DD: DDDD I see the last letters truncated, missing the last digit of the seconds and the last digit of the year. I tried to write with displaytext and the characters all work very well. Has anyone happened?

Condor-XYZ, 2020-11-29:
how can i interact this esp32 with display in tasmota?
https://github.com/Xinyuan-LilyGO/TTGO-T-Display

Jason2866, 2020-11-29:
@Condor-XYZ  it depends which display is used. If it is a ili9341 it is supported.
See https://tasmota.github.io/docs/ESP32/#odroid-go

Condor-XYZ, 2020-11-29:
According to the description it is a ST7789 display and unfortunately I can get it to work in tasmota. I'll try whether the odroid-go version works.

Condor-XYZ, 2020-11-30:
unfortunately it does not work .... my settings are see pictures
[image]
[image]

austrisv, 2021-03-19:
With TTGO T-Display I was having same trouble as @Condor-XYZ ... untill realized I should use display driver specific CS and DC (ST7789 CS, ST7789 DC). Plus the Oled reset and Backlight.
Then Check `DisplayModel` is 12 in Tasmota's console. Set `DisplayWidth 135`, then `DisplayMode 0` and `DisplayText HelloWorld` was working like charm!
[image]

zackfuchtel, 2022-10-05:
[austrisv](https://github.com/austrisv) commented [on 19 Mar 2021](https://github.com/arendst/Tasmota/issues/4290#issuecomment-802751661)

Wow, that's it. Thank you.