arendst/Tasmota #4292: wemos-d1-mini & user_config_override

zibous, 2018-11-06:
I use version 6.3.0.3 20181105 and try to flash it on the device wemos-d1-mini. The first try is that the user_config_override.h are not used. If I then enter STA_SSID1 and STA_PASS1 and then save, then the login fails WIFI.

I can not fix this this.

Can I set the WIFI settings via the console?

Why are the settings of user_config_override.h not used?
I never had any problems with the earlier versions.

Are the "platformio.ini" settings correct?

Platformio Settings
```
[env:wemos-d1-mini]
platform = ${common.platform}
framework = arduino
board = esp01_1m
board_build.flash_mode = ${common.board_build.flash_mode}
build_flags = ${common.build_flags} -DUSE_BASIC
monitor_speed = ${common.monitor_speed}
upload_port = ${common.upload_port}
upload_resetmethod = ${common.upload_resetmethod}
upload_speed = ${common.upload_speed}
extra_scripts = ${common.extra_scripts}
```

Error Messages Console
```
0:00:00 WIF: Connecting to AP1 Microwelle in mode 11N as ESP-GASCOUNTER...
00:00:09 WIF: Connect failed as AP cannot be reached
00:00:09 WIF: Connecting to AP1 Microwelle in mode 11N as ESP-GASCOUNTER...
00:00:18 WIF: Connect failed as AP cannot be reached
00:00:18 WIF: Connecting to AP1 Microwelle in mode 11N as ESP-GASCOUNTER...
....
```
[image]

Same problem I have when I flash new Sonoff devices, it always looks like the settings of user_config_override.h is not used.

Jason2866, 2018-11-06:
You have to uncomment in platformio.ini
  -DUSE_CONFIG_OVERRIDE  
```

; *** Fix espressif8266@1.7.0 induced undesired all warnings
build_unflags = -Wall

build_flags =
; if using esp8266 core 2.5.0 (stage) or up
;  -Wl,-Tesp8266.flash.1m.ld -Wl,-Map,firmware.map
; if using esp8266 core < 2.5.0
  -Wl,-Tesp8266.flash.1m0.ld -Wl,-Map,firmware.map
  -mtarget-align
;  -DUSE_CONFIG_OVERRIDE  
; lwIP 1.4 (Default)
;  -DPIO_FRAMEWORK_ARDUINO_LWIP_HIGHER_BANDWIDTH
; lwIP 2 - Low Memory
;  -DPIO_FRAMEWORK_ARDUINO_LWIP2_LOW_MEMORY
; lwIP 2 - Higher Bandwitdh
  -DPIO_FRAMEWORK_ARDUINO_LWIP2_HIGHER_BANDWIDTH
; VTABLES in Flash (default)
  -DVTABLES_IN_FLASH
; VTABLES in Heap
;  -DVTABLES_IN_DRAM
; VTABLES in IRAM
;  -DVTABLES_IN_IRAM

```

ascillato, 2018-11-06:
To use the user_config_override.h you need to enable it in my_user_config.h

Jason2866, 2018-11-06:
@ascillato  This i correct for Arduino IDE. 
For Platformio you just need to uncomment what i have explained

zibous, 2018-11-06:
... all WLAN Sonoff devices work perfectly with the Ubiquiti Networks UAP-AC-PRO only wemos-d1-mini do not work.

Am I doing something wrong with the settings?
platform = espressif8266@1.8.0

I have chosen espressif8266@1.8.0 because it allows the SONOFF Basic devices to run without any problems.

All other setting will NOW set by the user_config_override.h.

Jason2866, 2018-11-06:
If every other device works, the wemos mini is faulty.
I have several wemos minis and no problem with core 2.3.0, 2.4.2 and upcoming 2.5.0

ascillato2, 2018-11-06:
Hi, sorry can't reproduce.

If it is working in your other devices means that your module is faulty or the flashing was not ok.

closed 2018-11-06 by ascillato2

zibous, 2018-11-06:
I use several devices: 
```
Sonoff Pow Module         👍  Sonoff-Tasmota 6.2.1.15 by Theo Arends
Sonoff S2X Module	  👍  Sonoff-Tasmota 6.2.1.15 by Theo Arends
Sonoff SC Module	  👍  Sonoff-Tasmota 6.2.1.15 by Theo Arends
Sonoff Basic Module	  👎  Sonoff-Tasmota 5.12.0 by Theo Arends, > 5.12.0 is faulty
Sonoff Bridge Module      👍  Sonoff-Tasmota 6.2.1.15 by Theo Arends
Sonoff Dual Module	  👍  Sonoff-Tasmota 6.2.1.18 by Theo Arends
Sonoff 4CH Module	  👍  Sonoff-Tasmota 6.2.1.18 by Theo Arends
wemos mini                👎  Tasmota is faulty, but works with ESPurna, EspEasy, OpenMQTTGateway ...
```

I suspect it is due to the settings or the implementation how the devices use WIFI. I flash all devices with the same computer and always with PlatformIO.

Would be glad if there is a solution, because I would like to use the same firmware for all devices.

ascillato, 2018-11-06:
Please, try the last version from the development branch

Jason2866, 2018-11-06:
Try http://thehackbox.org/tasmota/sonoff-basic.bin
If this works your devices has a bad power supply. Latest Tasmota have many sensors and functions
enabled so it draws more power than older versions. The sonoff-basic.bin has disabled many things
so it uses less power. If this version works it is sure that the issue you have is hardware related.
Because there are many cloned wemos mini with faked 3.3V voltage regulators, only 150mA instead of 500mA! This regulators fail....

Jason2866, 2018-11-07:
Yes, power supply of new Sonoff basic is bad...