arendst/Tasmota #4293: MPR121 Touch sensor

satekp, 2018-11-06:
**IMPORTANT NOTICE**
If you do not complete the template below it is likely that your issue will not be addressed. When providing information about your issue please be as extensive as possible so that it can be solved by as little as possible responses.

**FAILURE TO COMPLETE THE REQUESTED INFORMATION WILL RESULT IN YOUR ISSUE BEING CLOSED**

Make sure these boxes are checked [x] before submitting your issue - Thank you!

- [ x] Searched the problem in issues (https://github.com/arendst/Sonoff-Tasmota/issues)
- [x ] Searched the problem in the wiki (https://github.com/arendst/Sonoff-Tasmota/wiki/Troubleshooting)
- [ x] Searched the problem in the forum (https://groups.google.com/d/forum/sonoffusers)
- [ ] Searched the problem in the chat (https://discord.gg/Ks2Kzd4)
- [ x] Development/Compiler/Upload tools used :
- [ x] Hardware used : 
- [ x] If a pre-compiled release or development binary was used, which one? :6.2.1
- [ x] You have tried latest release or development binaries? :
- [ x] Provide the output of command ``status 0`` :
```
15:24:25 CMD: status 0
15:24:25 MQT: stat/sonofftest/STATUS = {"Status":{"Module":18,"FriendlyName":["Sonoff","Sonoff2","Sonoff3","Sonoff4"],"Topic":"sonofftest","ButtonTopic":"","Power":0,"PowerOnState":3,"LedState":1,"SaveData":1,"SaveState":1,"ButtonRetain":0,"PowerRetain":0}}
15:24:25 MQT: stat/sonofftest/STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"sonoffs","OtaUrl":"http://sonoff.maddox.co.uk/tasmota/sonoff.bin","RestartReason":"External System","Uptime":"0T01:26:31","StartupUTC":"2018-11-06T12:57:54","Sleep":0,"BootCount":112,"SaveCount":1992,"SaveAddress":"3F9000"}}
15:24:25 MQT: stat/sonofftest/STATUS2 = {"StatusFWR":{"Version":"6.2.1","BuildDateTime":"2018-09-25T20:53:13","Boot":31,"Core":"2_4_2","SDK":"2.2.1(cfd48f3)"}}
15:24:25 MQT: stat/sonofftest/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["sutrdo",""],"TelePeriod":300,"SetOption":["00028009","55818000","00000000"]}}
15:24:25 MQT: stat/sonofftest/STATUS4 = {"StatusMEM":{"ProgramSize":499,"Free":2572,"Heap":18,"ProgramFlashSize":4096,"FlashSize":4096,"FlashMode":2,"Features":["00000809","0FDAE794","00000000","23B617CE","00000011"]}}
15:24:25 MQT: stat/sonofftest/STATUS5 = {"StatusNET":{"Hostname":"sonofftest-2313","IPAddress":"192.168.1.18","Gateway":"192.168.1.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.1.1","Mac":"60:01:94:29:69:09","Webserver":2,"WifiConfig":5}}
15:24:25 MQT: stat/sonofftest/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.1.107","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_296909","MqttUser":"openhabian","MqttType":1,"MAX_PACKET_SIZE":1000,"KEEPALIVE":15}}
15:24:25 MQT: stat/sonofftest/STATUS7 = {"StatusTIM":{"UTC":"Tue Nov 06 14:24:25 2018","Local":"Tue Nov 06 15:24:25 2018","StartDST":"Sun Mar 25 02:00:00 2018","EndDST":"Sun Oct 28 03:00:00 2018","Timezone":1,"Sunrise":"07:45","Sunset":"17:22"}}
15:24:25 MQT: stat/sonofftest/STATUS10 = {"StatusSNS":{"Time":"2018-11-06T15:24:25","Switch1":"ON"}}
15:24:25 MQT: stat/sonofftest/STATUS11 = {"StatusSTS":{"Time":"2018-11-06T15:24:25","Uptime":"0T01:26:31","Vcc":2.920,"POWER1":"OFF","POWER2":"OFF","POWER3":"OFF","POWER4":"OFF","POWER5":"OFF","POWER6":"OFF","Dimmer":69,"Fade":"OFF","Speed":1,"LedTable":"OFF","Wifi":{"AP":1,"SSId":"sutrdo","RSSI":100,"APMac":"F8:E9:03:92:65:AB"}}}
```
I have uncommented "#define USE_MPR121" in "user_config.h", Tasmota genericm
connected MPR121 module I2C ports to esp I2C port 

[image]

 I2Cscan
15:27:58 MQT: stat/sonofftest/RESULT = {"I2CScan":"Device(s) found at 0x5a"}
  Now need help on how to configure touch sensor input  in Rule to control esp gpios (power/pwm..)

      
**(Please, remember to close the issue when the problem has been addressed)**

ascillato2, 2018-11-06:
This driver only publish the buttons' touch by mqtt

satekp, 2018-11-07:
When I do I2C scan  console shows the I2C address of the module 
But on touching input No Mqtt message appear on tasmota console !