arendst/Tasmota #4294: Missing module parameters "5 (I2C SCL)", "6 (I2C SDA)" in Sonoff-Tasmota 6.3.0 on WEMOS D1 Mini Lite

empenoso, 2018-11-06:
**IMPORTANT NOTICE**
If you do not complete the template below it is likely that your issue will not be addressed. When providing information about your issue please be as extensive as possible so that it can be solved by as little as possible responses.

**FAILURE TO COMPLETE THE REQUESTED INFORMATION WILL RESULT IN YOUR ISSUE BEING CLOSED**

**Describe the bug**
_Missing module parameters "5 (I2C SCL)", "6 (I2C SDA)" in Sonoff-Tasmota 6.3.0 by Theo Arends on WEMOS D1 Mini Lite_

_Also, make sure these boxes are checked [x] before submitting your issue - Thank you!_
- [x] _Searched the problem in issues and in the wiki_
- [x] _Hardware used_ : WEMOS D1 Mini Lite
- [x] _Development/Compiler/Upload tools used_ : Visual Studio Code/ESP8266Flasher
- [ ] _If a pre-compiled release or development binary was used, which one?_ :
- [ ] _You have tried latest release or development binaries?_ :
- [x] _Provide the output of command_``status 0`` :
```
{
Status: {
Module: 18,
FriendlyName: [
"3dPrinterRoom"
],
Topic: "3dPrinterRoom",
ButtonTopic: "0",
Power: 0,
PowerOnState: 3,
LedState: 1,
SaveData: 1,
SaveState: 1,
SwitchTopic: "0",
SwitchMode: [
0,
0,
0,
0,
0,
0,
0,
0
],
ButtonRetain: 0,
SwitchRetain: 0,
SensorRetain: 0,
PowerRetain: 0
},
StatusPRM: {
Baudrate: 115200,
GroupTopic: "sonoffs",
OtaUrl: "http://thehackbox.org/tasmota/release/sonoff.bin",
RestartReason: "Software/System restart",
Uptime: "0T00:11:57",
StartupUTC: "2018-11-06T15:05:58",
Sleep: 0,
BootCount: 8,
SaveCount: 17,
SaveAddress: "FB000"
},
StatusFWR: {
Version: "6.3.0",
BuildDateTime: "2018-11-06T11:58:17",
Boot: 31,
Core: "2_3_0",
SDK: "1.5.3(aec24ac9)"
},
StatusLOG: {
SerialLog: 2,
WebLog: 2,
SysLog: 0,
LogHost: "",
LogPort: 514,
SSId: [
"ofnet",
"rinet"
],
TelePeriod: 300,
SetOption: [
"00008009",
"55818000",
"00000000"
]
},
StatusMEM: {
ProgramSize: 484,
Free: 516,
Heap: 19,
ProgramFlashSize: 1024,
FlashSize: 1024,
FlashMode: 255,
Features: [
"00000809",
"6050A790",
"248003A8",
"210217C6",
"000001C0"
]
},
StatusNET: {
Hostname: "3dPrinterRoom-2410",
IPAddress: "192.168.15.68",
Gateway: "192.168.15.1",
Subnetmask: "255.255.255.0",
DNSServer: "212.33.225.212",
Mac: "60:01:94:XX:89:6A",
Webserver: 2,
WifiConfig: 5
},
StatusMQT: {
MqttHost: "192.168.15.15",
MqttPort: 1883,
MqttClientMask: "DVES_%06X",
MqttClient: "DVES_82896A",
MqttUser: "DVES_USER",
MqttType: 1,
MAX_PACKET_SIZE: 1000,
KEEPALIVE: 15
},
StatusTIM: {
UTC: "Tue Nov 06 15:17:55 2018",
Local: "Tue Nov 06 20:17:55 2018",
StartDST: "Sun Mar 25 02:00:00 2018",
EndDST: "Sun Oct 28 03:00:00 2018",
Timezone: 5
},
StatusSNS: {
Time: "2018-11-06T20:17:55"
},
StatusSTS: {
Time: "2018-11-06T20:17:55",
Uptime: "0T00:11:57",
Vcc: 2.74,
Wifi: {
AP: 1,
SSId: "ofnet",
BSSId: "64:D1:XX:F5:XX:B4",
Channel: 3,
RSSI: 92
}
}
}
```

**To Reproduce**
_Steps to reproduce the behavior:_
I can not find GPIO commands "5 (I2C SCL)", "6 (I2C SDA)" in Sonoff-Tasmota 6.3.0.
http://192.168.15.68/cm?cmnd=gpios :
GPIOs1: [
"0 (None)",
"1 (DHT11)",
"2 (AM2301)",
"3 (SI7021)",
"4 (DS18x20)",
"7 (WS2812)",
"9 (Switch1)",
...

**Expected behavior**
_A clear and concise description of what you expected to happen._
I need to use I2C SCL and SDA.

**Screenshots**
_If applicable, add screenshots to help explain your problem._

**Additional context**
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

Jason2866, 2018-11-06:
Your self compiled firmware is faulty. Erase flash and try precompiled

ascillato2, 2018-11-06:
Hi, I can confirm that is missing I2C pins in the last version

Jason2866, 2018-11-06:
Andre has just the idea to add commit number to build version.
So we knew exact which version is used.
I tried with not the latest build.

arendst, 2018-11-06:
In what version is it missing? It should be missing in sonoff_basic.bin as i2c is disabled...

empenoso, 2018-11-10:
if I use release https://github.com/arendst/Sonoff-Tasmota/releases, there is no problem. I2C is there.
If I use Visual Studio Code/ESP8266Flasher - there is no I2C there. There are no errors when compiling.

What can go wrong?

ascillato, 2018-11-10:
Seems that your config does not have USE_I2C, or has other key that disables I2C like USE_BASIC. Just use the standard my_user_config.h from the Tasmota Repository

empenoso, 2018-11-11:
@ascillato Thank! Everything worked out.