arendst/Tasmota #4295: unable to flash Sonoff T1 Uk Version 1.1 2CH (believe it wont go into programming mode)

GreyMarvel, 2018-11-06:
```
- [x] Searched the problem in issues (https://github.com/arendst/Sonoff-Tasmota/issues)
- [x] Searched the problem in the wiki (https://github.com/arendst/Sonoff-Tasmota/wiki/Troubleshooting)
- [x] Searched the problem in the forum (https://groups.google.com/d/forum/sonoffusers)
- [ ] Searched the problem in the chat (https://discord.gg/Ks2Kzd4)
- [x] Development/Compiler/Upload tools used :arduino IDE with FDTI and SonOTA
- [ ] Hardware used : Sonoff T1 uk
- [ ] If a pre-compiled release or development binary was used, which one? :
- [ ] You have tried latest release or development binaries? :
- [ ] Provide the output of command ``status 0`` :

```
Ok so I am really struggling now I've flashed about 12 sonoff T1's today all are version 1.0 boards and are 1 channel, all have been successful with the Arduino IDE and the FDTI connector,
however I've got 2 x 2 channel T1's that will not flash no matter what I do I believe that it's not going into programming mode as the main LED D3 doesn't stop flashing , no button LED dim and no serial output on the monitor at baud 74880.
I've tried different button combinations and also I tried SonOTA unsuccessfully, I think this is because the stock Firmware is 2.6.0 on the boards

every time I've tried to flash, just in case it wasn't obvious it was in flash mode I get this error 

```
trying to connect
	espcomm_send_command: sending command header
	espcomm_send_command: sending command payload
warning: espcomm_sync failed
error: espcomm_open failed
error: espcomm_upload_mem failed

```

any help would be great , thanks in advance

ascillato, 2018-11-06:
Search for gpio0 of the esp8266 chip and put it to gnd before powering up

ascillato2, 2018-11-06:
Closing as this is not a Tasmota bug. For support please use the Tasmota Channel. Github issues are for bug reporting. Thanks

### Support Information
See [Wiki](https://github.com/arendst/Sonoff-Tasmota/wiki) for more information.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.

closed 2018-11-06 by ascillato2