arendst/Tasmota #4297: Support for  Gosund Sp1-C_V2.3

AugustH, 2018-11-06:
**Have you look for this feature in other issues and in the wiki?**
yes

**Is your feature request related to a problem? Please describe.**
Gosund SP1 has a new version - new layout and new pins, differs from the version shown at https://www.bastelbunker.de/gosund-sp1-mit-tasmota

**Describe the solution you'd like**
device should work out of the box ;-)

**Describe alternatives you've considered**
the measurement chip can only configured in the source :-(

Just got a Gosund SP1 (https://www.amazon.de/gp/product/B0777BWS1P) -- there is a new chip layout resulting in new GPIO connections. The board is labeled "SP1-C_V2.3 20180926".

I got it to work with the following config (not sure about the leds, but the powermeter and the relay is working):

    { "Goland SP1-C_v23",  // https://www.amazon.de/gp/product/B0777BWS1P
     GPIO_LED2_INV,    // GPIO00 Red Led (1 = On, 0 = Off)
     GPIO_USER,        // GPIO01 Serial RXD and Optional sensor
     GPIO_LED1_INV,    // GPIO02 Blue Led (1 = On, 0 = Off)
     GPIO_USER,        // GPIO03 Serial TXD and Optional sensor
     GPIO_HJL_CF,      // GPIO04 BL0937 or HJL-01 CF power
     GPIO_NRG_CF1,     // GPIO05 BL0937 or HJL-01 CF1 current / voltage
     0, 0, 0, 0, 0, 0, // Flash connection
     GPIO_NRG_SEL_INV, // GPIO12 BL0937 or HJL-01 Sel output (0 = Voltage)
     GPIO_KEY1,        // GPIO13 Button
     GPIO_REL1,        // GPIO14 Relay (0 = Off, 1 = On)
     0, 0, 0
     },

AugustH, 2018-11-07:
fixed Config (button and leds)

     { "Gosund SP1_v23",  // https://www.amazon.de/gp/product/B0777BWS1P
     0,    
     GPIO_LED1_INV,    // GPIO01 Serial RXD and LED1 (blue) inv
     0,        
     GPIO_KEY1,        // GPIO03 Serial TXD and Button
     GPIO_HJL_CF,      // GPIO04 BL0937 or HJL-01 CF power
     GPIO_NRG_CF1,     // GPIO05 BL0937 or HJL-01 CF1 current / voltage
     0, 0, 0, 0, 0, 0, // Flash connection
     GPIO_NRG_SEL_INV, // GPIO12 BL0937 or HJL-01 Sel output (0 = Voltage)
     GPIO_LED2_INV,    // GPIO13 LED2 (red) inv
     GPIO_REL1,        // GPIO14 Relay (0 = Off, 1 = On)
     0, 0, 0
     }

ascillato2, 2018-11-07:
Great, thanks a lot. Can you add this information to the wiki under other devices?

Thanks again :+1:

AugustH, 2018-11-08:
done, see https://github.com/arendst/Sonoff-Tasmota/wiki/Gosund-SP1

aritmeester, 2018-11-09:
Choosing the Teckin module is also [approved working](https://www.bastelbunker.de/gosund-sp1-mit-tasmota/) for this board revision. Same applies for #4303 

So now we have three possible configuration options for all those clones together:
1. Gosund SP1_v23
2. Teckin
3. BlitzWolf SHP2

R-Guhr, 2018-11-27:
Hardware button is not working! Tasmota v6.3.0, BlitzWolf BW-SHP2 v2.3 (Teckin)

prankousky, 2018-11-28:
Can you please post a photo and label the neccessary pins (GND, VCC, TX, RX, and the one to flash)?

I just bought those on the German amazon Black Friday sale, was looking forward to flash, and now cannot figure out how to. I tried finding the pins myself with a multimeter, but could only get RX and TX, and am not even sure about those (never actually had to try this before because this wiki -and the link you provided- had already done it).

Thanks in advance / vielen Dank 👍

AugustH, 2018-11-28:
see the pictures at https://github.com/arendst/Sonoff-Tasmota/wiki/Gosund-SP1

prankousky, 2018-11-28:
@AugustH awesome, thank you!

xnfctn, 2018-11-28:
> see the pictures at https://github.com/arendst/Sonoff-Tasmota/wiki/Gosund-SP1

that worked, thanks!

gunmetaldjs, 2018-12-09:
Anyone got lucky using the hardware button in the BlitzWolf BW-SHP2 v2.3 ?

HanSolo72, 2018-12-10:
**New Hardware revision Gosund SP1-C_V2.4 20181101**
But looks like V2.3.
Here are the pictures: https://drive.google.com/drive/folders/1xmTNGbQpoYkIsagzl5wdrWzfieF8snJ5

agsochi, 2019-02-22:
Hi. I have a BlitzWolf BW-SHP2 V2.4. Button works with "Gosund SP1 v23 (55)" /6.4.1, blue LED is flashing 1sec period.
When i push button - relay is on, but no RED led is on. How to fix RED led?