arendst/Tasmota #4298: Sonoff Basic WiFi cannot connect and resets ad infinitum with sonoff.bin flash

justoke, 2018-11-06:
Make sure these boxes are checked [x] before submitting your issue - Thank you!

- [x] Searched the problem in issues (https://github.com/arendst/Sonoff-Tasmota/issues)
- [x] Searched the problem in the wiki (https://github.com/arendst/Sonoff-Tasmota/wiki/Troubleshooting)
- [x] Searched the problem in the forum (https://groups.google.com/d/forum/sonoffusers)
- [x] Searched the problem in the chat (https://discord.gg/Ks2Kzd4)
- [x] Development/Compiler/Upload tools used : esptool-2.5.1
- [x] Hardware used : Sonoff Basic R2
- [x] If a pre-compiled release or development binary was used, which one? : sonoff.bin
- [x] You have tried latest release or development binaries? : sonoff.bin
- [x ] Provide the output of command ``status 0`` :
00:00:00 CMD: Status 0
00:00:00 RSL: stat/sonoff/STATUS = {"Status":{"Module":1,"FriendlyName":["Sonoff"],"Topic":"sonoff","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"SaveData":1,"SaveState":1,"ButtonRetain":0,"PowerRetain":0}}
00:00:00 RSL: stat/sonoff/STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"sonoffs","OtaUrl":"http://sonoff.maddox.co.uk/tasmota/sonoff.bin","RestartReason":"Hardware Watchdog","Uptime":"0T00:00:00","StartupUTC":"","Sleep":0,"BootCount":119,"SaveCount":121,"SaveAddress":"FB000"}}
00:00:00 RSL: stat/sonoff/STATUS2 = {"StatusFWR":{"Version":"6.2.1","BuildDateTime":"2018-09-09T16:50:26","Boot":31,"Core":"2_3_0","SDK":"1.5.3(aec24ac9)"}}
00:00:00 RSL: stat/sonoff/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["",""],"TelePeriod":300,"SetOption":["00008009","55818000","00000000"]}}
00:00:00 RSL: stat/sonoff/STATUS4 = {"StatusMEM":{"ProgramSize":471,"Free":532,"Heap":16,"ProgramFlashSize":4096,"FlashSize":4096,"FlashMode":3,"Features":["00000809","0FDAE794","000003A0","23B617CE","00000000"]}}
00:00:00 RSL: stat/sonoff/STATUS5 = {"StatusNET":{"Hostname":"sonoff-0570","IPAddress":"0.0.0.0","Gateway":"192.168.2.254","Subnetmask":"255.255.255.0","DNSServer":"192.168.2.27","Mac":"80:7D:3A:32:42:3A","Webserver":2,"WifiConfig":5}}
00:00:00 RSL: stat/sonoff/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.1.109","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_32423A","MqttUser":"DVES_USER","MqttType":1,"MAX_PACKET_SIZE":1000,"KEEPALIVE":15}}
00:00:00 RSL: stat/sonoff/STATUS7 = {"StatusTIM":{"UTC":"Thu Jan 01 00:00:00 1970","Local":"Thu Jan 01 00:00:00 1970","StartDST":"Thu Jan 01 00:00:00 1970","EndDST":"Thu Jan 01 00:00:00 1970","Timezone":1,"Sunrise":"07:49","Sunset":"16:07"}}
00:00:00 RSL: stat/sonoff/STATUS10 = {"StatusSNS":{"Time":"0000-01-01T00:00:00"}}
00:00:00 RSL: stat/sonoff/STATUS11 = {"StatusSTS":{"Time":"0000-01-01T00:00:00","Uptime":"0T00:00:00","Vcc":3.170,"POWER":"OFF","Wifi":{"AP":1,"SSId":"","RSSI":100,"APMac":"FF:FF:FF:FF:FF:FF"}}}
```

Background
I bought a batch of 7 Sonoff basic modules - they all have R2 on the labels and using the esptool.py utility I was able to flash them with the latest pre-compiled binary sonoff.bin. I used the Termite terminal application to connect to them and set the WIFI SSID and password as this is quicker than the WiFiManager. Six of the devices are now up and running. I should mention that I tested all devices by adding them to the EweLink application to ensure that they worked as originally intended under

Issue
One device will just not connect to the WiFi even after I have set the WiFi SSID and password. The behaviour is as shown by the serial monitor log ad infinitum:

00:00:00 Project sonoff Sonoff (Topic sonoff, Fallback DVES_32423A, GroupTopic sonoffs) Version 6.2.1-2_3_0
00:00:00 WIF: WifiManager active for 3 minutes
00:00:00 HTP: Web server active on sonoff-0570 with IP address 192.168.4.1
ets Jan  8 2013,rst cause:4, boot mode:(3,7)
wdt reset
load 0x4010f000, len 1384, room 16 
tail 8
chksum 0x2d
csum 0x2d
v60000608
~ld

00:00:00 Project sonoff Sonoff (Topic sonoff, Fallback DVES_32423A, GroupTopic sonoffs) Version 6.2.1-2_3_0
00:00:00 WIF: WifiManager active for 3 minutes
00:00:00 HTP: Web server active on sonoff-0570 with IP address 192.168.4.1

Note: The watchdog timer resets within 30 seconds and the WiFiManager does not appear active for anywhere near 3 minutes. Also there is no sonoff-0570 network coming up on my Wifi network list on my mobile device.

All the other devices would connect to the WiFi once the SSID and password had been set via the serial commands:
SSId myWifi
Password1 mypwd

Thinking that it might be the flash chip I have tried the many suggestions on the posts listed in the links shown at the header of this issue. I then replaced the 1MB with a 4MB earlier today and flashed back the Sonoff stock binary. Added the device to EweLink and it works again as original Sonoff device.

This might be a one-off issue and I would accept that the device may have an issue were it not for the fact that it works normally as a Sonoff device, but once flashed - even with a new flash ROM chip - the behaviour is repeatable.

Hopefully someone has experienced this issue and may be able to help resolve. Thank you.

justoke, 2018-11-07:
Thank you. I used the recommended binary and it worked as described.

andrethomas, 2018-11-11:
@justoke Do you know if your sonoff basic is an R2 or the older version? If its an R2 we would like you to perform a test.

justoke, 2018-11-12:
> @justoke Do you know if your sonoff basic is an R2 or the older version? If its an R2 we would like you to perform a test.

They are all R2. I'd be happy to test something.

andrethomas, 2018-11-12:
@justoke Great!

We've made some adjustment to how the wifi is reset in the current development codebase.

Could you be so kind as to test the latest development binary for sonoff.bin found here http://thehackbox.org/tasmota/sonoff.bin and report back if the wifi problems experienced with 6.3.0 persist or if it works fine after this update?

Thanks!

justoke, 2018-11-12:
I'm not sure what the expected behaviour is. Here are my observations:

After flashing the sonoff.bin I connected Termite serial monitor and captured the out messages.

 I can report the same behaviour that I reported when I raised this issue originally - that is, it is not possible to connect to the WiFi as it does not stay alive for 3 minutes. By the time I grab my phone browse the WiFi connections and try to connect it has reset. 
This sequence repeated over and over.
```
00:00:00 Project sonoff Sonoff (Topic sonoff, Fallback DVES_674507, GroupTopic sonoffs) Version 6.3.0.7(a0aff44-sonoff)-2_3_0
00:00:00 WIF: WifiManager active for 3 minutes
00:00:00 HTP: Web server active on sonoff-1287 with IP address 192.168.4.1

 ets Jan  8 2013,rst cause:4, boot mode:(3,7)

wdt reset
load 0x4010f000, len 1384, room 16 
tail 8
chksum 0x2d
csum 0x2d
v60000608
~ld

00:00:00 Project sonoff Sonoff (Topic sonoff, Fallback DVES_674507, GroupTopic sonoffs) Version 6.3.0.7(a0aff44-sonoff)-2_3_0
00:00:00 WIF: WifiManager active for 3 minutes
00:00:00 HTP: Web server active on sonoff-1287 with IP address 192.168.4.1

```
So I entered the WiFi credentials via the Serial terminal using these commands:
SSId myWifi
Password1 mypwd
MqttHost ipaddress

The settings persisted and the device restarted and displayed these messages:
```
00:00:00 Project sonoff Sonoff (Topic sonoff, Fallback DVES_674507, GroupTopic sonoffs) Version 6.3.0.7(a0aff44-sonoff)-2_3_0
00:00:00 WIF: Connecting to AP1 miHome in mode 11N as sonoff-1287...
00:00:07 WIF: Connected
00:00:07 DNS: Initialized
00:00:07 HTP: Web server active on sonoff-1287.local with IP address 192.168.1.121
22:02:45 RSL: tele/sonoff/UPTIME = {"Time":"2018-11-12T22:02:45","Uptime":"0T00:00:08"}
22:02:45 MQT: Attempting connection...
22:02:45 MQT: Connected
22:02:45 MQT: tele/sonoff/LWT = Online (retained)
22:02:45 MQT: cmnd/sonoff/POWER = 
22:02:46 MQT: tele/sonoff/INFO1 = {"Module":"Sonoff Basic","Version":"6.3.0.7(a0aff44-sonoff)","FallbackTopic":"DVES_674507","GroupTopic":"sonoffs"}
22:02:46 MQT: tele/sonoff/INFO2 = {"WebServerMode":"Admin","Hostname":"sonoff-1287","IPAddress":"192.168.1.121"}
22:02:46 MQT: tele/sonoff/INFO3 = {"RestartReason":"Software/System restart"}
22:02:46 MQT: stat/sonoff/RESULT = {"POWER":"OFF"}
22:02:46 MQT: stat/sonoff/POWER = OFF
22:02:46 MQT: stat/sonoff/STATUS5 = {"StatusNET":{"Hostname":"sonoff-1287","IPAddress":"192.168.1.121","Gateway":"192.168.1.254","Subnetmask":"255.255.255.0","DNSServer":"192.168.1.254","Mac":"80:7D:3A:67:45:07","Webserver":2,"WifiConfig":5}}
```
Status 0
```
22:10:37 CMD: status 0
22:10:37 MQT: stat/sonoff/STATUS = {"Status":{"Module":1,"FriendlyName":["Sonoff"],"Topic":"sonoff","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
22:10:37 MQT: stat/sonoff/STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"sonoffs","OtaUrl":"http://thehackbox.org/tasmota/release/sonoff.bin","RestartReason":"Software/System restart","Uptime":"0T00:08:00","StartupUTC":"2018-11-12T21:02:37","Sleep":0,"BootCount":23,"SaveCount":32,"SaveAddress":"F4000"}}
22:10:37 MQT: stat/sonoff/STATUS2 = {"StatusFWR":{"Version":"6.3.0.7(a0aff44-sonoff)","BuildDateTime":"2018-11-12T18:00:51","Boot":31,"Core":"2_3_0","SDK":"1.5.3(aec24ac9)"}}
22:10:37 MQT: stat/sonoff/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["miHome",""],"TelePeriod":300,"SetOption":["00008009","55818000","00000000"]}}
22:10:37 MQT: stat/sonoff/STATUS4 = {"StatusMEM":{"ProgramSize":492,"Free":508,"Heap":15,"ProgramFlashSize":1024,"FlashSize":1024,"FlashChipId":"146085","FlashMode":3,"Features":["00000809","0FDAE794","000183A0","23B617CE","00003BC0"]}}
22:10:37 MQT: stat/sonoff/STATUS5 = {"StatusNET":{"Hostname":"sonoff-1287","IPAddress":"192.168.1.121","Gateway":"192.168.1.254","Subnetmask":"255.255.255.0","DNSServer":"192.168.1.254","Mac":"80:7D:3A:67:45:07","Webserver":2,"WifiConfig":5}}
22:10:37 MQT: stat/sonoff/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.1.109","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_674507","MqttUser":"DVES_USER","MqttType":1,"MAX_PACKET_SIZE":1000,"KEEPALIVE":15}}
22:10:37 MQT: stat/sonoff/STATUS7 = {"StatusTIM":{"UTC":"Mon Nov 12 21:10:37 2018","Local":"Mon Nov 12 22:10:37 2018","StartDST":"Sun Mar 25 02:00:00 2018","EndDST":"Sun Oct 28 03:00:00 2018","Timezone":"+01:00","Sunrise":"07:54","Sunset":"17:14"}}
22:10:37 MQT: stat/sonoff/STATUS10 = {"StatusSNS":{"Time":"2018-11-12T22:10:37"}}
22:10:37 MQT: stat/sonoff/STATUS11 = {"StatusSTS":{"Time":"2018-11-12T22:10:37","Uptime":"0T00:08:00","Vcc":3.138,"POWER":"OFF","Wifi":{"AP":1,"SSId":"miHome","BSSId":"E8:65:D4:CD:CA:89","Channel":6,"RSSI":78}}}
```
Happy to assist with anything else.

justoke, 2018-11-13:
Yes, the WiFi connects after setting the SSId and password fields via the serial terminal commands. I would probably question why the WiFiManager logs its intention to be up for 3 minutes when this is not actually available. Why not remove that from the logging if it is not intended to be used?

justoke, 2018-11-14:
I tried running reset 1 and it went back to AP mode. As before the WiFiManager cannot actually be used and setting the SSId/Password using the serial commands results in it reconnecting to the WiFi and from there it can be configured.

andrethomas, 2018-11-15:
@justoke Looks like there may be something else lingering in the flash that could be causing this because it should not do what you are explaining.

Just to confirm, you are running the latest development binary from http://thehackbox.org/tasmota/sonoff.bin as at the time of your posting?

Please bear with me and try the following
`reset 2`
This will erase the flash area not used by the firmware - it will take a minute or so to complete and it will reboot after.

Please report back behavior experienced.

Thanks again for all the feedback.

justoke, 2018-11-15:
@andrethomas -  I downloaded and flashed the binary as linked in your previous post.

Result of running reset 2:
I have tried this and attached a video of the terminal output to explain what happens. Initially I thought that there was no change in the behaviour described, as it soon started the watchdog reset cycle. However, I was eventually able to connect to the WiFiManager and set the SSId/Password via the http://192.168.4.1 connection and the device restarted and connected. I was not able to do this before on any of my R2 devices - I had flashed a batch of 6 of them and used the serial terminal to configure the Wifi as the WiFiManager just did not work long enough to connect.

Attached mp4 video file:
[SonoffReset2.zip](https://github.com/arendst/Sonoff-Tasmota/files/2586645/SonoffReset2.zip)

andrethomas, 2018-11-15:
That is indeed very strange behaviour you captured on video! - never seen anything like it before.

I know this is becoming a little extensive and annoying but I really appreciate what you are doing.

Can you clarify how you are powering the device at this time? I'm asking because I am wondering if the continuous reboots could be related to the power limitation of your usb-serial connector.

If possible can you apply external 3.3V + GND from a good 3.3V source and then only connect GND, TX and RX from the usb-serial transceiver to see if the result is reproducible in that way?

justoke, 2018-11-15:
I'm using a basic [D-SUN USB to TTL](https://www.amazon.co.uk/gp/product/B00LODGRV8/ref=oh_aui_search_detailpage?ie=UTF8&psc=1) module for all the 3.3V, TX, RX and GND connections. I've tried a separate 3.3V supply and ran the reset 2 command again. The behaviour is the same as shown in the video.

To back track slightly, when I started with the whole flash upgrade process a few weeks ago, I first flashed the device using the USB connection, then put it back in the case and connected the mains and tried to configure using the WiFi Manager - but that is when I realised something was not working as I could just never connect to the WiFiManager. 

Only by connecting the serial terminal could I start to get insight into what was happening. It was then apparent that the best option for my needs was to flash and set up my devices using the serial terminal commands as this saved a lot of time and frustration.

I am new to Sonoff-Tasmota so I can't compare experience with previous versions. I have only ever had the Sonoff Basic R2 modules and used the latest release bin files.

I'm happy to help, just not sure if this is just a Sonoff R2 thing or my setup has some limitations. But from my experience the WiFiManager is not usable with the behaviour shown. It can take countless attempts to connect and configure via the web interface.

I've set aside one of the R2s for experimenting, so whatever I can try out just let me know.

andrethomas, 2018-11-16:
The untimely reboots suggests that the ESP8266 chip is not running as it should. I'll continue to think of reasons other than the one's we've discovered so far.

For now, from console try to enable sleep to see if it makes a difference.

From console enter command:
`sleep 50`

Thanks for your patience!

justoke, 2018-11-27:
Sorry for the delayed response. There have been other things taking up my time. I tried the sleep command, but didn't notice any difference in the behaviour. On a separate note, I have noticed another thing which is worrying. After flashing my set of 6 Sonoff Basics and reconnecting them to the mains voltage, I observed that if I powered them on and off a couple of times from the mains socket, a few of them would appear to lose their configuration and are back at the default state with the WDT reset loop as described in this thread. I then have to remove them and re-configure via the Serial terminal.