arendst/Tasmota #4299: SwitchTopic

kbaras, 2018-11-06:
**IMPORTANT NOTICE**
If you do not complete the template below it is likely that your issue will not be addressed. When providing information about your issue please be as extensive as possible so that it can be solved by as little as possible responses.

**FAILURE TO COMPLETE THE REQUESTED INFORMATION WILL RESULT IN YOUR ISSUE BEING CLOSED**

**Describe the bug**
With TASMOTA 6.3.0 I have issue with SwitchMode behaviour when network connection is lost: I use reed sensor at GPIO3 (or tried also with GPIO14 with same result) and therefore SwitchMode 2 is configured. It does not switch the relay as expected, but when newtork signal is lost it is like it fell back to SwitchMode 0 (or similar) as in this state closing/opening this GPIO pin with ground toggles the relay. 
Similar behaviour as described in feature request #1041. I use the REED sensor with garage door opener, so it can cause a lot of troubles. 

_Also, make sure these boxes are checked [x] before submitting your issue - Thank you!_
- [ ] _Searched the problem in issues and in the wiki_: yes
- [ ] _Hardware used_ : SONOFF basic with REED sensor 
- [ ] _Development/Compiler/Upload tools used_ : PlatformIO
- [ ] _If a pre-compiled release or development binary was used, which one?_ : NO
- [ ] _You have tried latest release or development binaries?_ : yes
- [ ] _Provide the output of command_``status 0`` :
```
cannot provide it at the moment, as the issue comes without network connection, and switch configured to GPIO
```

**To Reproduce**
_Steps to reproduce the behavior:_
Flash Tasmota 6.3.0 to a SONOFF basic, set switch to e.g. GPIO14, set SwitchMode 2, then mistype the wifi credentials so that it cannot connect to the wifi, restart SONOFF and try opening/closing the configured GPIO with ground.

**Expected behavior**
No switching of relay when open/close GPIO14 with ground

**Screenshots**

**Additional context**

**(Please, remember to close the issue when the problem has been addressed)**

ascillato, 2018-11-07:
Hi,

That is a designed behaviour. When there is no mqtt, the switch will control the relay. 

This fallback is the default. If you want another behaviour, you will need to set rules for that. 

Please, check the examples of rules in the wiki. 

Thanks

benzino77, 2018-11-07:
I have simple configuration: wemos d1 mini with relay connected to D1 (relay1) `pulsetime1 4`, and reed sensor as switch2 connected to D7 (with pulldown) - `switchmode2 2`
The rule:
```
on switch2#state=0 do publish stat/garage-wm01/GATE CLOSED endon 
on switch2#state=1 do publish stat/garage-wm01/GATE OPEN endon
```
I can remotely open the gate by sending `cmnd/garage-wm01/power1 on` 

@ascillato please correct me if I'm wrong
With this configuration there will be absolutely no chance to open the garage gate with remote controller (button on car key) in situation when wifi or mqtt connection is lost. When I push a button on my car key to open the gate, the reed sensor will be in `switchmode2 0` which will not be handled by the rules and it will toggle the relay which will result in ... stoping opening the gate. 

will this rule configuration change this behavior:
```
on switch2#state=0 do publish stat/garage-wm01/GATE CLOSED endon 
on switch2#state=1 do publish stat/garage-wm01/GATE OPEN endon 
on switch2#state=2 delay endon
```
(do nothing when TOGGLE)?

benzino77, 2018-11-07:
...or meybe define "fake" switch1 and "fake" relay2 and leave the rule as it is?

ascillato, 2018-11-07:
> please correct me if I'm wrong
> With this configuration there will be absolutely no chance to open the garage gate with remote controller (button on car key) in situation when wifi or mqtt connection is lost.

Sorry, I have no clue how your garage door is connected, so I can't say anything about your case.

If you use switchtopic, when there is no mqtt or wifi, Tasmota has a fallback behaviour to start controlling your relays from your switches. 

If you are using Tasmota in a different way than the default, you need to tell Tasmota what do you want Tasmota to do. So, for PIR sensors, reeds, etc., you need to use rules to make the behaviour you want.

ascillato, 2018-11-07:
In the third rules you post is missing a do for do delay

benzino77, 2018-11-07:
Thanks for replay.  
Me and @kbaras are talking about `switchmode` not `switchtopic`,  in my configuration I have not changed `switchtopic`  it is still `0`.

benzino77, 2018-11-07:
I've performed quick test with the above configuration: Wemos mini with relay conected to pin D1 (`relay1`) and reed switch connected to D7 (with pull down - `switch2`)
```
switchtopic 0
switchmode2 2
rule1 on switch2#state=0 do 
   publish stat/test11-wm01/GATE CLOSED 
endon 
on switch2#state=1 do 
   publish stat/test11-wm01/GATE/ OPEN 
endon 
```
Test 1
change `mqtthost` to wrong value - reed switch works as expected (does not change relay) it still `switchmode2 2` and trying to publish OPEN/CLOSED messages
Test 2 
change `ssid1` to wrong value - reed switch works as expected (does not change relay) it is still `switchmode2 2` and trying to publish OPEN/CLOSED messages

@ascillato could you please clarify whether  "fallback mode" is related to `switchtopic` or `switchmode`?

ascillato, 2018-11-07:
It is just for switchtopic

benzino77, 2018-11-07:
Thanks! I was a little bit... stressed that one day I won't be able to open my garage gate ;)

@kbaras could you please provide what is the value of `switchtopic` in your configuration?

ascillato, 2018-11-07:
switchtopic overrides rules. So switchtopic must be 0 to allow the rules `on switch1#state=1 do ...` to work

benzino77, 2018-11-07:
I do believe that the same situation is with buttontopic (it overwrites rules)?

ascillato, 2018-11-07:
yes, exactly

kbaras, 2018-11-07:
Hello!
SwitchMode=1
SwitchTopic=sonoffgaragesensor

At least when connected.
I cannot get output when it does not connect to network.

kbaras, 2018-11-07:
Sorry, and could you help, which rule to use? I have tried this one that benzino mentioned, but this has no effect on the behaviour.

benzino77, 2018-11-08:
What is a number of your switch in configuration? 
[image]

Do you have a pullup or pulldown? To use rules you cannot change switchtopic.