arendst/Tasmota #4302: Will displaytext support mqtt topics?

ceraz68, 2018-11-07:
**Have you look for this feature in other issues and in the wiki?**
Yes

**Is your feature request related to a problem? Please describe.**
No, it's a question on feature support

**Describe the solution you'd like**
The status of a sonoff switch (tele topic) to be displayed on an OLED screen connected to a wemos mini 
D1 running tasmota. I can use different hardware if necessary.
I've seen examples of displaytext used for locally connected peripherals on GPIO. I'd like to subscribe to an mqtt topic

**Describe alternatives you've considered**
Use different firmware but still under evaluation. I prefer tasmota!

**Additional context**
_Add any other context or screenshots about the feature request here._

**(Please, remember to close the issue when the problem has been addressed)**

andrethomas, 2018-11-07:
You should be able to use a rule to catch the switch and tell the rule to execute displaytext with its %value%

ceraz68, 2018-11-07:
Thanks andrethomas,

I did look at the rules for the DisplayText command but the wiki refers to "Display local sensors using rules".

For example in the wiki: "rule1 on tele-SHT3X-0x44#Temperature do DisplayText [f1p7x0y5]%value% C endon". Do you mean, I can replace "tele-SHT3X-0x44#Temperature" by an mqtt topic which would subscribe to the broker?

Otherwise, I thought of a remote software plugin to capture mqtt values from my switch and then use http to execute displaytext on the wemos but don't know if this is possible.

Thx

andrethomas, 2018-11-07:
You can send displaytext commands over mqtt also.

closed 2018-11-07 by ascillato2