arendst/Tasmota #4303: New hardware revision v2.3 BlitzWolf® BW-SHP2

aritmeester, 2018-11-07:
My new BW-SHP2 just arrived and it seems I have gotten a new hardware revision. I managed to flash Tasmota on it but it does not work with the default BW-SPH2 module type.
I switched to generic board and was able to use the relay on GPIO14. I didn't get the other functions working (power monitoring, button, led).

Any help to get it working completely is appreciated!

[image]

[image]

[image]

[image]

[image]

ascillato, 2018-11-07:
Please,

Find which gpios are connected to buttons, leds, power monitoring and also which power monitoring chip it uses.

Then modify the sonoff_template.h with your findings to test your device. 

Thanks

ascillato2, 2018-11-07:
Closing this issue as it has been answered.

### Support Information
See [Wiki](https://github.com/arendst/Sonoff-Tasmota/wiki) for more information.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.

closed 2018-11-07 by ascillato2

AugustH, 2018-11-07:
see -- Goland SP1 V2.3 seems to be a clone
https://github.com/arendst/Sonoff-Tasmota/issues/4297

AugustH, 2018-11-07:
Why is this closed? needs change/additional entry in sonoff_template.h -- see #4297

aritmeester, 2018-11-10:
I can confirm this unit working perfectly on the _Teckin_ module type

Roxxor91, 2018-11-24:
I just flashed it with the standard tasmota firmware. First I was able to get the hotspot functionality. (Nor sure if this was default, or I entered it successfully with 4 button presses.) Now it doesnt connect, maybe I entered something wrongly. I now want to reconfigure, but Im not able to enter hotspot mode via 4 button presses. Does this eman the button doesnt work and I have to reflash? I already removed all wires -_-

R-Guhr, 2018-11-27:
Hardware button is not working! Tasmota v6.3.0, BlitzWolf BW-SHP2 v2.3

Soulguard, 2018-12-01:
Same here at Gosund SP1 v2.3. All works, except the hardware button and the LED is constantly slow flashing in red regardless the plug is switched on or off.

AugustH, 2018-12-01:
which version did you flash? just checked the nightly (Sonoff-Tasmota 6.3.0.14) from
 http://thehackbox.org/tasmota/ 
with Konfiguration: 55 Gosund SP1 v23

Soulguard, 2018-12-01:
thank you very much AugustH, i had flashed 6.3.0. now with 6.3.0.14 the device works as expected. Hardware Button works, LED is blue now (properly configuring MQTT stops slowly flashing LED) Great work!

kbhuinfo, 2018-12-02:
Teckin working for me as well.
[image]
[image]

Silencer80, 2018-12-11:
I´ve just flashed tasmota to 4 Gosund SP1 v24. This one has Screws to and is not glued together like the Teckin. Pins/Layout seems equal to v23. I´ll test the newest nightly this afternoon and will post here, if everything works as expected.

naeramarth7, 2019-01-02:
Running latest tasmota firmware (v6.4.1), the button works for me using the "Gosund SP1 v23 (55)" config.

vb0, 2019-01-13:
I have a V2.4 that looks just like the pictures here, with a date from 201811. Works fine with "Gosund SP1 v23 (55)" /6.4.1 too.

SimoneBnc, 2019-01-19:
I got 2 v2.4 SHP2 and it works like the v2.3.  Has anyone measured how accurate is the voltage reading? Mine is 10v below what I measured with a true rms multimeter.

agsochi, 2019-02-22:
Hi. I have a V2.4. Button works with "Gosund SP1 v23 (55)" /6.4.1, blue LED is flashing 1sec period.
When i push button - relay is on, but no RED led is on. How to fix RED led?

riston, 2019-03-31:
I received also newer version BlitzWolf v2.4 and when using Gosound SP1 module type it's working great after calibration voltage is off by +-2V(mostly 1).

paucoma, 2019-04-03:
Hi All, 
I recieved mine (v2.4) opened it up and traced the board to create this schematic. Not Tested yet! but I'm pretty sure its right. 
In summary ESP pinout is the following : 
-IO1 | sink LED1 blue [out] | U0TXD
-IO3 | active low K1 (pushbutton) [in] | U0RXD
-IO4 | CF (BL0937) [in] | GPIO4
-IO5 | CF1 (BL0937) [in] | GPIO5
-IO12 | SEL (BL0937) [out] | MTDI
-IO13 | sink LED1 red [out] | MTCK
-IO14 | source RL2 relay [out] | MTMS
-RST | not connected | EXT_RSTB?
-AD | not connected | TOUT

Cheers,
Pau
[sch_BW-SHP2_(SP1-C_V2.4).pdf](https://github.com/arendst/Sonoff-Tasmota/files/3038647/sch_BW-SHP2_.SP1-C_V2.4.pdf)

scasic, 2019-04-12:
What are modes of button LED? When it should blink and with which color?
Mine:
MQTT enabled, power is OFF - blinking BLUE
MQTT enabled, power is ON - blinking RED/BLUE
MQTT disabled, power is OFF - no light
MQTT disabled, power in ON - solid red

Is it correct behavior?

mariantr, 2019-04-18:
I successfully flashed a BW-SPH2 with latest Tasmota, but after that the led slowly blinks red and the device does not connect to wifi (pre-configured SSID and password in config before flashing) and the button is not working
Anyone has any idea how to fix this?

SimoneBnc, 2019-04-18:
Start by reading the serial debug and look for what's going on

mszel42, 2019-05-17:
Since SHP2 is one of those Tuya based devices this OTA Tasmota installation method worked just fine:
https://github.com/ct-Open-Source/tuya-convert
The tuya-convert has a built in Tasmota v6.5.0 basic(but you can define any firmware during the install), remember, that basic version doesn't come with homeassistant auto discovery support, but Tasmota OTA works as expected.

shady2k, 2019-05-31:
> I received also newer version BlitzWolf v2.4 and when using Gosound SP1 module type it's working great after calibration voltage is off by +-2V(mostly 1).

@riston , How do you calibrate it? Are you convert data on server side?

sorianonet, 2019-06-03:
hi,
i've broken the IO0. is there another way to find it and enter in programmer mode?

Notrial, 2019-10-10:
> What are modes of button LED? When it should blink and with which color?
> Mine:
> MQTT enabled, power is OFF - blinking BLUE
> MQTT enabled, power is ON - blinking RED/BLUE
> MQTT disabled, power is OFF - no light
> MQTT disabled, power in ON - solid red
> 
> Is it correct behavior?

Did you by any chance figure out how to modify this? I have same led patterns.
Can we disable the leds altogether?

AugustH, 2019-10-11:
you could create a template without the leds

atico45, 2020-03-27:
Good afternoon, I have been able to flash this device through OTA but I do not know how to link it to the homeassistant I have done it through Mqtt, there is another hole.
Thanks and sorry for not knowing English, this is from the google translate

SimoneBnc, 2020-03-27:
It's written in the home assistant documentation, just enable the autodiscover in tasmota

atico45, 2020-03-27:
Ok, gracias