arendst/Tasmota #4309: Documentation Clarification

danobot, 2018-11-08:
Just using this issue to raise some documentation problems that may require clarification. From my understanding, the recommended way to configure is to use `user_config_override.h` and then pick and choose values from `my_user_confg.h` to overwrite by copy pasting. In `Sonoff-Tasmota/sonoff/my_user_config.h`, there is the following line:

```
#define MODULE                 SONOFF_BASIC      // [Module] Select default model from sonoff_template.h (Should not be changed)
```
What does "(Should not be changed)" referring to? This seems like a setting that users may overwrite without issues if they wish. Can confirm that overwriting this setting works as expected, so I am not sure what the quoted string means.

ascillato2, 2018-11-08:
You can change it without problems. You can test it.

ascillato2, 2018-11-08:
Closing this issue as it has been answered.

### Support Information
See [Wiki](https://github.com/arendst/Sonoff-Tasmota/wiki) for more information.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.

closed 2018-11-08 by ascillato2

danobot, 2018-11-08:
What does "(Should not be changed)" referring to?

ascillato, 2018-11-08:
It is old. Should not be there anymore.

danobot, 2018-11-08:
Ok, just wanted to point it out. You can remove it then.