arendst/Tasmota #4311: Need to change unit from kg -> kPa for HX711

wongnam, 2018-11-08:
I have implemented a Pressure sensor MPS20N0040D-S and using HX711 to interface it to NodeMCU but it's measured by Kg.
How to change it to measure by kPa or mV? Thank

[image]

[image]

[image]

```
STATUS 0 OUTPUT HERE - DO NOT DELETE THE MARKERS ABOVE AND BELOW THIS LINE
22:02:08 CMD: status 0
22:02:08 SRC: WebConsole from 192.168.12.116
22:02:08 RSL: Group 0, Index 1, Command STATUS, Data 0
22:02:08 MQT: stat/sonoff/STATUS = {"Status":{"Module":18,"FriendlyName":["Sonoff"],"Topic":"sonoff","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
22:02:08 MQT: stat/sonoff/STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"sonoffs","OtaUrl":"http://thehackbox.org/tasmota/release/sonoff.bin","RestartReason":"Software/System restart","Uptime":"0T00:52:47","StartupUTC":"2018-11-08T14:09:21","Sleep":0,"BootCount":4,"SaveCount":21,"SaveAddress":"F8000"}}
22:02:08 MQT: stat/sonoff/STATUS2 = {"StatusFWR":{"Version":"6.3.0.5(sonoff)","BuildDateTime":"2018-11-08T21:06:39","Boot":31,"Core":"2_3_0","SDK":"1.5.3(aec24ac9)"}}
22:02:08 MQT: stat/sonoff/STATUS3 = {"StatusLOG":{"SerialLog":0,"WebLog":3,"SysLog":0,"LogHost":"192.168.12.155","LogPort":514,"SSId":["Multimedia Wi-Fi Network",""],"TelePeriod":300,"SetOption":["00008009","55818000","00000000"]}}
22:02:08 MQT: stat/sonoff/STATUS4 = {"StatusMEM":{"ProgramSize":489,"Free":512,"Heap":18,"ProgramFlashSize":1024,"FlashSize":4096,"FlashMode":3,"Features":["00000809","0FCAE794","240183A0","23B617CE","00003BC0"]}}
22:02:08 MQT: stat/sonoff/STATUS5 = {"StatusNET":{"Hostname":"sonoff-6909","IPAddress":"192.168.12.133","Gateway":"192.168.12.1","Subnetmask":"255.255.255.0","DNSServer":"210.245.31.220","Mac":"68:C6:3A:84:1A:FD","Webserver":2,"WifiConfig":5}}
22:02:08 MQT: stat/sonoff/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.12.155","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_841AFD","MqttUser":"admin","MqttType":1,"MAX_PACKET_SIZE":1000,"KEEPALIVE":15}}
22:02:08 MQT: stat/sonoff/STATUS7 = {"StatusTIM":{"UTC":"Thu Nov 08 15:02:08 2018","Local":"Thu Nov 08 22:02:08 2018","StartDST":"Sun Mar 25 02:00:00 2018","EndDST":"Sun Oct 28 03:00:00 2018","Timezone":"+07:00","Sunrise":"05:46","Sunset":"17:27"}}
22:02:08 MQT: stat/sonoff/STATUS10 = {"StatusSNS":{"Time":"2018-11-08T22:02:08","HX711":{"Weight":458}}}
22:02:08 MQT: stat/sonoff/STATUS11 = {"StatusSTS":{"Time":"2018-11-08T22:02:08","Uptime":"0T00:52:47","Vcc":2.835,"Wifi":{"AP":1,"SSId":"Multimedia Wi-Fi Network","BSSId":"B0:C7:45:36:85:E0","Channel":11,"RSSI":84}}}
```
Thanks.

ascillato, 2018-11-08:
Hi,

As the HX711 can be used with a load cell or a pressure sensor, I think we need to add a setoption to select the display unit for HX711.

Now, to use hPa instead of Kg, you can change D_UNIT_KILOGRAM to D_UNIT_PRESSURE in the line

https://github.com/arendst/Sonoff-Tasmota/blob/2a4d82f4b69ab4f91ba2be1c3d4a3bb8983f5c08/sonoff/xsns_34_hx711.ino#L392

and also change

https://github.com/arendst/Sonoff-Tasmota/blob/2a4d82f4b69ab4f91ba2be1c3d4a3bb8983f5c08/sonoff/xsns_34_hx711.ino#L435

to

`dtostrfd((float)Settings.weight_item / 1000, 4, tmp);`

_( Hope it is ok for you to use **hPa** instead of **kPA** in order to use the already translated key **hPa** )_

wongnam, 2018-11-08:
Hi, @ascillato  I've just modified it but maybe the unit display in other place.

[image]

ascillato, 2018-11-13:
@wongnam 

Those changes in the code have worked fine for you?

Are you ok on using hPA?

wongnam, 2018-11-13:
Thanks for your support.
Update: In my case is to measure the pressure so It just changed the unit to hPa was not fix my case due to every power cycle that it will tare the value to 0 as it is for the weight scale.

closed 2018-11-16 by wongnam