arendst/Tasmota #4313: SHT3X sensor driver: invalid JSON identifier syntax when converted to a JSON object in node-red

CliveBull, 2018-11-08:
**Describe the bug**
Closed issues #1949 and #3758 refer - the issue raised in each bug report (with reference to a Home Assistant use case) has been closed but is not resolved. I have further input to make on the issue in a node-red use case. 

_Also, make sure these boxes are checked [x] before submitting your issue - Thank you!_
- [x] _Searched the problem in issues and in the wiki_
- [x] _Hardware used_ : Wemos D1 mini 
- [x] _Development/Compiler/Upload tools used_ : PlatformIO, web OTA updater (no issues with compile or upload)
- [x] _If a pre-compiled release or development binary was used, which one?_ : n/a
- [x] _You have tried latest release or development binaries?_ : master release 6.3.0 used
- [x] _Provide the output of command_``status 0`` :
```
10:40:09 MQT: stat/wemos01/STATUS = {"Status":{"Module":18,"FriendlyName":["Wemos [01]"],"Topic":"wemos01","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":1,"SwitchRetain":1,"SensorRetain":0,"PowerRetain":1}}
10:40:09 MQT: stat/wemos01/STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"sonoffs","OtaUrl":"http://thehackbox.org/tasmota/release/sonoff.bin","RestartReason":"Software/System restart","Uptime":"0T00:31:52","StartupUTC":"2018-11-08T21:08:17","Sleep":0,"BootCount":8,"SaveCount":22,"SaveAddress":"FA000"}}
10:40:09 MQT: stat/wemos01/STATUS2 = {"StatusFWR":{"Version":"6.3.0","BuildDateTime":"2018-11-09T10:02:03","Boot":31,"Core":"2_4_2","SDK":"2.2.1(cfd48f3)"}}
10:40:09 MQT: stat/wemos01/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"SysLog":0,"LogHost":"192.168.1.33","LogPort":514,"SSId":["XXXXX",""],"TelePeriod":30,"SetOption":["000080E9","44818000","00000000"]}}
10:40:09 MQT: stat/wemos01/STATUS4 = {"StatusMEM":{"ProgramSize":440,"Free":560,"Heap":24,"ProgramFlashSize":1024,"FlashSize":4096,"FlashMode":3,"Features":["00000809","0D002594","240183A1","000216C6","00003BC0"]}}
10:40:09 MQT: stat/wemos01/STATUS5 = {"StatusNET":{"Hostname":"wemos01-4464","IPAddress":"192.168.1.72","Gateway":"192.168.1.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.1.1","Mac":"84:F3:EB:0D:B1:70","Webserver":2,"WifiConfig":5}}
10:40:09 MQT: stat/wemos01/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.1.33","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_0DB170","MqttUser":"","MqttType":1,"MAX_PACKET_SIZE":1000,"KEEPALIVE":15}}
10:40:09 MQT: stat/wemos01/STATUS7 = {"StatusTIM":{"UTC":"Thu Nov 08 21:40:09 2018","Local":"Fri Nov 09 10:40:09 2018","StartDST":"Sun Sep 30 02:00:00 2018","EndDST":"Sun Apr 01 03:00:00 2018","Timezone":99}}
10:40:09 MQT: stat/wemos01/STATUS10 = {"StatusSNS":{"Time":"2018-11-09T10:40:09","SHT3X-0x44":{"Temperature":20.9,"Humidity":73},"TempUnit":"C"}}
10:40:09 MQT: stat/wemos01/STATUS11 = {"StatusSTS":{"Time":"2018-11-09T10:40:09","Uptime":"0T00:31:52","Vcc":2.952,"Wifi":{"AP":1,"SSId":"XXXXX","BSSId":"XX:XX:XX:XX:XX:XX","Channel":11,"RSSI":100}}}

```

**To Reproduce**
Using node-red, connect an MQTT input node to a JSON string to object converter node to a function node. In the (Javascript) function node, reference to the JSON object identifier "SHT3X-0x44" throws a syntax error. 

The error is that the dash between the sensor type and the i2c address is not a valid JSON object identifier within Javascript. This is discussed in more detail in the referenced closed issues but within a Home Assistant use case.

Refer: https://developer.mozilla.org/en-US/docs/Glossary/Identifier Quote: "In JavaScript, identifiers can contain only alphanumeric characters (or "$" or "_"), and may not start with a digit."

Thus, an underscore is correct but a dash is not.

I have proved the bug fix by editing the sensor driver xsns_14_sht3x.ino and modifying the PSTR in line 109 from PSTR("%s-0x%02X") to PSTR("%s_0x%02X"). This creates a correctly parsable JSON object identifier.

I don't have the knowledge to propose a change within github.

**Expected behavior**
The node-red function node should output correctly parsed JSON values for temperature and humidity for further processing in the flow.

**Screenshots**
n/a

**Additional context**
Closed issues #1949 and #3758 refer - the issue raised in each bug report (with reference to a Home Assistant use case) has been closed but not resolved.

**(Please, remember to close the issue when the problem has been addressed)**

ascillato, 2018-11-08:
Thanks for your report. Very complete :+1: 

Made the PR #4314 for solving this.

curzon01, 2018-11-09:
To be noted:
A Javascript idenitifier is not a JSON token so the error at all implementing a Javascript identifier as a JSON token!

Refering [RFC 7159](https://tools.ietf.org/html/rfc7159) a JSON token is a [String](https://tools.ietf.org/html/rfc7159#section-7) which needs only escaping a few characters - where a dash is not part of it.

This solution here is a workaroung for the wrong implemtation in node-re template. The correct way would be node-red using the suggestion of [#1949 Frogmore42](https://github.com/arendst/Sonoff-Tasmota/issues/1949#issuecomment-387925195).

RaymondMouthaan, 2018-11-09:
Not sure if it is mentioned earlier but to get the temperature and/or humidity of the sensor received from mqtt in node-red one can do the following:

mqtt node -> json node -> function node

[image]

where the function node has the following code:
```
var obj = {"payload": {}};
obj.payload.temp = msg.payload["SHT3X-0x45"].Temperature;
obj.payload.hum = msg.payload["SHT3X-0x45"].Humidity;

msg = obj;

return msg;
```

In other words, using the bracket-quote style notation one can access the key - value pair.

ascillato, 2018-11-09:
@curzon01 is right. There is no need for this fix. Thanks for pointing this. This is the good thing of posting a PR for revision.

Thanks to all.

closed 2018-11-09 by ascillato2