arendst/Tasmota #4324: IPv6 Support now possible ....

cawi2001, 2018-11-10:
Hello,

for Freifunk, IPv6 is mandatory to reach from outside the Tasmota device.

In the area of esp8266/Arduino there is a new change for IPv6 support:
https://github.com/esp8266/Arduino/pull/5136

Can this code used within Tasmota?

ascillato, 2018-11-10:
Up to now, it is not supported by the Arduino core, they are working on that but it is not released. Tasmota uses the Arduino core to work. When Arduino release a new core that supports IPv6, then the modifications needed can be done to Tasmota. Up to now the Stage core has in the debugging process. It is not released for production nor stable.

We don't know when Arduino will add IPv6 to the core and if it is going to be a stable core by 2.5.0 at the end of the year.

ascillato2, 2018-11-11:
For reference: https://github.com/arendst/Sonoff-Tasmota/issues/2521

closed 2018-11-11 by ascillato2

ascillato2, 2018-11-11:
### Support Information
See [Wiki](https://github.com/arendst/Sonoff-Tasmota/wiki) for more information.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.

akram, 2020-01-15:
@ascillato it seems that IPv6 is now supported in Arduino core: https://github.com/esp8266/Arduino/pull/5136

is it worth re-opening this issue ?

ascillato, 2020-01-15:
> is it worth re-opening this issue ?

No because it is already implemented in Tasmota. You have to compile by yourself enabling IPV6 support in platformio.ini

If you need further assistant, please, do not hesitate on asking on the Support Chat. Thanks.