arendst/Tasmota #4325: SR04 sensor, send data to domotic

Tresett, 2018-11-10:
Hl there.

Sonoff Basic + Ultrasonic sensor SR04.
Firmware: 5.13.1a
Domotics via MQTT
Domoticz IDX setted into Key IDX field.
But no data received in domoticz.

I see data from sensor on the main page.
I see telemetry info in console. Example:
02:23:48 MQT: tele/Sonoff_Basic_7.5/SENSOR = {"Time":"2018-11-11T02:23:48","SR04":{"Distance":117}}

Is it a bug or may be a future request?
I want receive information from sensor to domoticz.

P.S. I'm check to set domoticz IDX into all other field in the settings. No effect.

ascillato2, 2018-11-11:
Please, update to last version. Several changes were done that impact in your request. Actual version is 6.3.0.6. And then please report back if your issue was solved.

Also, please complete the troubleshooting template in order to give us information in order to properly help you.

ascillato2, 2018-11-11:
### Support Information
See [Wiki](https://github.com/arendst/Sonoff-Tasmota/wiki) for more information.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.

arendst, 2018-11-11:
The maintainer of the SR04 sensor did not implement domoticz functionality. I will today and provide the information as Count/PM1 Sensor idx 6.