arendst/Tasmota #4327: shelly1 -> no wifi connections

philko123, 2018-11-11:
**IMPORTANT NOTICE**
If you do not complete the template below it is likely that your issue will not be addressed. When providing information about your issue please be as extensive as possible so that it can be solved by as little as possible responses.

**FAILURE TO COMPLETE THE REQUESTED INFORMATION WILL RESULT IN YOUR ISSUE BEING CLOSED**

**Describe the bug**
shelly1 doesn' t connect to the defined AP and doesn't show up the wifi manager

_Also, make sure these boxes are checked [x] before submitting your issue - Thank you!_
- [ x] _Searched the problem in issues and in the wiki_
- [ x] _Hardware used_ : 
shelly1, FTD1232, MacBook Pro Mojave
- [ x] _Development/Compiler/Upload tools used_ :
Arduino IDE 1.8.7 all libraries updated
- [ ] _If a pre-compiled release or development binary was used, which one?_ :
- [x ] _You have tried latest release or development binaries?_ :
I tried developement, master and release, always the same issue
- [ ] _Provide the output of command_``status 0`` :
```
STATUS 0 OUTPUT HERE - DO NOT DELETE THE MARKERS ABOVE AND BELOW THIS LINE
```
~ld

00:00:00 CFG: Use defaults
00:00:00 Project shelly1_sz1 shelly1_sz1 (Topic shelly1_sz1, Fallback DVES_5B37F1, GroupTopic shelly1_sz1) Version 6.3.0-2_4_2
12:39:30.785 -> 00:00:00 UPL: Arduino OTA Enabled Port 8266
00:00:00 WIF: Connecting to AP1 my_ssid in mode 11N as shelly1_sz1-6129...
00:00:09 WIF: Connect failed as AP cannot be reached
00:00:09 WIF: Connecting to AP1 my_ssid in mode 11N as shelly1_sz1-6129...
00:00:18 WIF: Connect failed as AP cannot be reached
12:39:58.875 -> 00:00:18 WIF: WifiManager active for 3 minutes
00:00:18 HTP: Web server active on shelly1_sz1-6129 with IP address 192.168.4.1
00:00:32 CMD: status 0
12:40:21.514 -> 00:00:32 RSL: stat/shelly1_sz1/STATUS = {"Status":{"Module":46,"FriendlyName":["shelly1_sz1"],"Topic":"shelly1_sz1","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
12:40:21.514 -> 00:00:32 RSL: stat/shelly1_sz1/STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"shelly1_sz1","OtaUrl":"http://thehackbox.org/tasmota/release/sonoff.bin","RestartReason":"Software/System restart","Uptime":"0T00:00:32","StartupUTC":"","Sleep":0,"BootCount":1,"SaveCount":3,"SaveAddress":"F9000"}}
12:40:21.514 -> 00:00:32 RSL: stat/shelly1_sz1/STATUS2 = {"StatusFWR":{"Version":"6.3.0","BuildDateTime":"2018-11-11T12:38:22","Boot":31,"Core":"2_4_2","SDK":"2.2.1(cfd48f3)"}}
12:40:21.514 -> 00:00:32 RSL: stat/shelly1_sz1/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["my_ssid",""],"TelePeriod":300,"SetOption":["00008009","55818000","00000000"]}}
12:40:21.514 -> 00:00:32 RSL: stat/shelly1_sz1/STATUS4 = {"StatusMEM":{"ProgramSize":499,"Free":504,"Heap":23,"ProgramFlashSize":1024,"FlashSize":2048,"FlashMode":3,"Features":["00000809","475027B0","00000000","000000C6","000000C0"]}}
12:40:21.514 -> 00:00:32 RSL: stat/shelly1_sz1/STATUS5 = {"StatusNET":{"Hostname":"shelly1_sz1-6129","IPAddress":"0.0.0.0","Gateway":"192.168.178.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.178.1","Mac":"80:7D:3A:5B:37:F1","Webserver":2,"WifiConfig":2}}
12:40:21.514 -> 00:00:32 RSL: stat/shelly1_sz1/STATUS6 = {"StatusMQT":{"MqttHost":"","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_5B37F1","MqttUser":"DVES_USER","MqttType":1,"MAX_PACKET_SIZE":1000,"KEEPALIVE":15}}
12:40:21.514 -> 00:00:32 RSL: stat/shelly1_sz1/STATUS7 = {"StatusTIM":{"UTC":"Thu Jan 01 00:00:32 1970","Local":"Thu Jan 01 00:00:32 1970","StartDST":"Thu Jan 01 00:00:00 1970","EndDST":"Thu Jan 01 00:00:00 1970","Timezone":1,"Sunrise":"07:43","Sunset":"16:03"}}
12:40:21.514 -> 00:00:32 RSL: stat/shelly1_sz1/STATUS10 = {"StatusSNS":{"Time":"1970-01-01T00:00:32","Switch1":"OFF"}}
12:40:21.514 -> 00:00:32 RSL: stat/shelly1_sz1/STATUS11 = {"StatusSTS":{"Time":"1970-01-01T00:00:32","Uptime":"0T00:00:32","Vcc":3.461,"POWER":"OFF","Wifi":{"AP":1,"SSId":"my_ssid","BSSId":"FF:FF:FF:FF:FF:FF","Channel":1,"RSSI":100}}}
00:02:00 RSL: tele/shelly1_sz1/UPTIME = {"Time":"1970-01-01T00:02:00","Uptime":"0T00:02:00"}
00:03:20 APP: Restarting

12:44:39.050 ->  ets Jan  8 2013,rst cause:4, boot mode:(1,7)
12:44:39.050 -> 
12:44:39.050 -> wdt reset

12:44:49.954 ->  ets Jan  8 2013,rst cause:4, boot mode:(1,7)
12:44:49.954 -> 
12:44:49.954 -> wdt reset

**To Reproduce**
_Steps to reproduce the behavior:_

**Expected behavior**
_A clear and concise description of what you expected to happen._

**Screenshots**
_If applicable, add screenshots to help explain your problem._

**Additional context**
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

ascillato, 2018-11-11:
Please try the latest version with a precompiled bin from http://thehackbox.org/tasmota/sonoff.bin

Also please erase all flash before upload using esptool.py. 

Thanks

philko123, 2018-11-11:
#1
esptool.py --port /dev/cu.usbserial-00000000 erase_flash
esptool.py v2.5.1-dev
Serial port /dev/cu.usbserial-00000000
Connecting....
Detecting chip type... ESP8266
Chip is ESP8266EX
Features: WiFi
MAC: 12:34:56:78:90:XX
Uploading stub...
Running stub...
Stub running...
Erasing flash (this may take a while)...
Chip erase completed successfully in 2.4s
Hard resetting via RTS pin...

#2
esptool.py --port /dev/cu.usbserial-00000000 write_flash -fs 1MB -fm dout 0x0 sonoff.bin
esptool.py v2.5.1-dev
Serial port /dev/cu.usbserial-00000000
Connecting....
Detecting chip type... ESP8266
Chip is ESP8266EX
Features: WiFi
MAC: 12:34:56:78:90:XX
Uploading stub...
Running stub...
Stub running...
Configuring flash size...
Compressed 503984 bytes to 345785...
Wrote 503984 bytes (345785 compressed) at 0x00000000 in 30.7 seconds (effective 131.3 kbit/s)...
Hash of data verified.

Leaving...
Hard resetting via RTS pin...

#3
power down and power up

#4 
wifi manager shows up, insert wifi credentials und it works!

thanks a lot, you saved my day :-)

closed 2018-11-11 by philko123