arendst/Tasmota #4329: WEBSEND command: SetOption for it to change behaviour, request/GET any URL

th-s-op, 2018-11-11:
I'd like to be able to use the WEBSEND command to request/GET any URL.

Currently, WEBSEND resolves the host, then GETs the URL defined by concatenating "cm?cmnd=" and the given command.

My feature request would be to optionally change the WEBSEND behaviour, which in this case just would NOT add the "cm?cmnd=" String. Instead, the WEBSEND "command" argument contains a complete URI.

Example:
**`WebSend [myserver.com] /fancy/data/logger.php?log=1234`**

With this, what I would expect a small, change, one could do fancy stuff with rules + WEBSEND. In my case, I would send data to an external data logger.

ascillato, 2018-11-11:
Interesting.

An idea could be to implement a WEBSEND2 command with that behavior.

What do you think?

th-s-op, 2018-11-11:
SYSLOG doesn't work for me as I really have to send something to a PHP script on my hosted server.

th-s-op, 2018-11-12:
**You are a genius! Thanks a lot for this and for this great project!!!**

closed 2018-11-12 by th-s-op

HoerMirAuf, 2019-01-13:
Seem'snot to work?

[sonoff,admin:joker] /any/link/starting/with/a/slash.php?log=123 --> Sends http://sonoff/any/link/starting/with/a/slash.php?log=123

Sending a websend:
websend [192.168.178.30,admin:0000] /cgi-bin/query.sh?test=0

will always transmit the /cm Parameter?
192.168.178.241 - - [13/Jan/2019:19:45:03 +0100] "GET /cm HTTP/1.1" 404 - "" ""

ascillato, 2019-01-13:
Please elaborate. I can't understand what you want to achieve and what if your issue. What is the console output? what is the command?

FraatRamski, 2019-03-10:
Hallo,
I'm using Sonoff-Tasmota 6.4.1.18 by Theo Arends.

when I'm trying to use websend on a special port i.e. 8181 so it is not running
21:11:28 CMD: WebSend [10.12.14.36:8181] /fancy/data/logger.php?log=1234
21:11:33 MQT: stat/sensor/RESULT = {"WebSend":"Connect failed"}

when chnaging the server port to 80 and trying websend seem to be working
21:11:10 CMD: WebSend [10.10.10.10] /fancy/data/logger.php?log=1234
21:11:10 MQT: stat/sensor/RESULT = {"WebSend":"Done"}

bonsai308, 2020-04-13:
Is it possible that the new function only works with http addresses, but not with https?

ascillato, 2020-04-13:
Exactly. Tasmota only supports http.

alanjds, 2024-03-14:
> **`WebSend [myserver.com] /fancy/data/logger.php?log=1234`**

Thanks for this issue. From the docs I though that `[<ip>]` was optional, not that `[` and `]` were mandatory!

Your example with `[myserver.com]` really clarified it. Kudos.

alanjds, 2024-03-14:
Funny that you mentioned WebQuery. I found it first, then moved for WebSend for conciseness.

But We query accepts TLS and WebSend don't, correct?

sfromis, 2024-03-14:
Partially true. On ESP32 only, you can use `https` with WebQuery, but not WebSend.

barbudor, 2024-03-14:
WebSend beng originally designed to send commands to other tasmota devices through HTTP doesn't need https

WebQuery on the other is meant as a generic http/https interface