arendst/Tasmota #4330: Auto disable WiFi

tty74it, 2018-11-11:
**Have you look for this feature in other issues and in the wiki?**

**Is your feature request related to a problem? Please describe.**
_A clear and concise description of what the problem is._

**Describe the solution you'd like**
_A clear and concise description of what you want to happen._

**Describe alternatives you've considered**
_A clear and concise description of any alternative solutions or features you've considered._

**Additional context**
_Add any other context or screenshots about the feature request here._

**(Please, remember to close the issue when the problem has been addressed)**

tty74it, 2018-11-11:
It would be useful to turn off the wifi in the t1 touch sonoffs installed in the bedroom near the head of the bed, so that during the night you avoid having additional wifi waves, of course maintaining the functionality of the switch. As soon as the button is pressed, the wifi is re-locked and then automatically disabled.
like this project:
https://github.com/omersiar/esp-rfid
https://bitadvise.com/esp-rfid/

Thank's
Tty74it

ascillato, 2018-11-11:
Hi,

It is not good to have a wifi source over the head. I agree. Your device needs to be installed in another place.

But also, turning off the wifi of your smart device make it not a home automation device. That is out of the scope of Tasmota. Please see readme for more information.

ascillato2, 2018-11-12:
Closing this issue as it has been answered.

### Support Information
See [Wiki](https://github.com/arendst/Sonoff-Tasmota/wiki) for more information.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.

closed 2018-11-12 by ascillato2

ascillato, 2018-11-12:
Another thing you can try. Besides changing the place of your device, is to set wificonfig to 5 and when you don't want your device to use wifi just turn off your wifi router.

With that configuration, if there is no wifi, your sonoff will be looking for a wifi signal but will not transmit anything. As soon as your wifi router is turned on, your device will connect again.

ascillato, 2018-11-12:
Yes, understood.

I think you need another type of hardware for what you want or another type of switches. Wifi signals close to your head is not good.

One idea could be to have a normal switch (or a capacitive one if you like) close to your bed but wired up to your roof light (far from your head), and in that place, put a sonoff basic, or a shelly, or a nodemcu connected to your switches. And that device can have Tasmota and wifi.

Remember, if you want, you can turn off your router and all your wifi devices will stop transmitting.

Also, you can see https://github.com/arendst/Sonoff-Tasmota/issues/1805 and https://github.com/arendst/Sonoff-Tasmota/issues/3609

If you have further questions or ideas, please do not hesistate on asking here or in the Tasmota chat. Thanks

tty74it, 2018-11-12:
Or using a RF433 battery powered wall switch, and sonoff rf bridge. My problem is find a similar touch t1. But modem sleep can be considered?

Thank's
tty74it

ascillato, 2018-11-13:
Modem sleep is already implemented in Tasmota if it is compiled using core 2.3.0 (the actual precompiled bins) or the actual arduino stage core.

But modem sleep is the light_modem_sleep that will put the cpu to sleep (less power) for up to 3 DTIM and then it will wake up for receiving and transmitting from the router. So, the sleep feature just reduce the CPU energy consumption, but the wifi signals will stay there.

If you set sleep to 1 (using command sleep in the console), your Tasmota device will use less energy to work and all the functions will work fine. If you increase that value up to 200 ms for example, all functions will work, but you will notice a delay in response when pressing any button.

tty74it, 2018-11-13:
I was thinking about it, but because it would not be useful to have the possibility of having an autodisable wifi after x seconds after launching a mqtt command. If you think of a device sonoff t1 that acts only as a button to turn on remote relays in toggle mode. It would work similarly to an RF433 remote control. If you need to change the configuration of the sonoff t1, it would be enough to increase the autodisable time to 5 minutes and you would have all the time to configure it. Each time you press the toggle button the wifi will turn on anyway for 5 minutes.

Thanks again for your patience

Greetings
tty74

arendst, 2018-11-13:
So you seriously want a switch that once toggled has to re-connect wifi taking at least 5 seconds and then performs the action you want. That means it has to store these actions and execute once connected. That's a whole new ballpark.

Why not buy these https://www.itead.cc/smart-home/t433-sticky-rf-remote.html and have a bridge somewhere to execute your requests.

tty74it, 2018-11-13:
I had not calculated the connection times of the wifi at least 5 sec.

Ok i try to buy it

Thank's
tty74

palto42, 2019-08-01:
I would also be interested in this feature, mainly to safe power if WiFi is not required.
With the great integration of RF433 transmitter/receiver there would still be a way for remote control (using less power). The other main advantage of Tasmota w/o WiFi compared to dumb remote are the advanced features Tasmota supports, e.g. rule based actions or autonomous time based switching.

ascillato, 2020-06-16:
Now it is available the command `WIFI` to turn ON and OFF the wifi from the Tasmota Console. Please, use latest development version.

Lukas1818, 2020-06-23:
It is possible to turn WiFi automaticlly on again? As Example at a specific time or after x hour?

ascillato, 2020-06-23:
Yes, by the use of rules and the new command. Please check the docs for more information about rules. If you need further assistant on that, please address this to the Tasmota support chat. Thanks.

Lukas1818, 2020-06-23:
Thanks you too

jd710313, 2021-03-21:
> Now it is available the command `WIFI` to turn ON and OFF the wifi from the Tasmota Console. Please, use latest development version.

Thanks, but I have flashed the latest development version (9.3.1.1) using Tasmotizer and choosing the development release, but the console says "{"Command":"Unknown"}" if I try and use the WIFI command?

ruzickav, 2021-04-01:
> Thanks, but I have flashed the latest development version (9.3.1.1) using Tasmotizer and choosing the development release, but the console says "{"Command":"Unknown"}" if I try and use the WIFI [...]

Same there - no idea how to use "wifi" command. Nothing in documentation...

DerDom, 2022-01-13:
I would be interested in the feature to reduce WIFI signals during the night. Did someone find the documentation or could provide it here?

barbudor, 2022-01-13:
Have you read [`Wifi`](https://tasmota.github.io/docs/Commands/#wi-fi) command ?
However be ready to have Tasmota working erraticaly without Wifi

ascillato, 2022-01-13:
Hi, it is documented in the Tasmota Oficial Docs.

Please go to: https://tasmota.github.io/docs/Commands/#wi-fi

Thanks.