arendst/Tasmota #4331: Bug: HTTP Host Header of WEBSEND command against the spec

th-s-op, 2018-11-11:
When using the "WebSend" command and using a host NAME like [myweb], the name is correctly resolved to an IP number (1.2.3.4 here as an example) but when capturing the HTTP request packet, I saw that for example the command "WebSend [myweb] status" sends the following headers:

GET /cm?cmnd=status HTTP/1.1
Host: 1.2.3.4
Connection: close

While in a singe server configuration, sending the resolved IP number in the "Host: " Header works fine (which in fact a single target Tasmota device is), but if behind the host name there is a web server with multiple virtual hosts, the "Host: " Header must contain the non-resolved name. So the correct header would be:

GET /cm?cmnd=status HTTP/1.1
Host: myweb
Connection: close

See the HTTP 1.1 specification - it must be the host name, not the resolved IP. Always.
https://tools.ietf.org/html/rfc7230#section-5.4

ascillato, 2018-11-12:
That behavior is in the code at line 1953 in _webserver.ino_:

https://github.com/arendst/Sonoff-Tasmota/blob/671231053afeb20a9fd9a5286b14b217816a8283/sonoff/xdrv_01_webserver.ino#L1950-L1958

we will look into it.

arendst, 2018-11-12:
The main reason for this is that many users have no name resolving (DNS) active or use mDNS. By providing the IP address this solves their issue.

I'll rewrite the code and allow hostnames in addition to plain ip address

th-s-op, 2018-11-12:
Wow, that was extremely fast, thanks!

I am just not 100% sure if the fix is completely correct in the way it handles the Port number:
From the standard, I read that the "Host: " request header does contain the Port number, too. At least as long as it's not the standard Port 80.

So an examples would be:

```
Host: myserver.com
Host: myserver.com:8082
```

In addition, I don't see that there is a "Port: " request header in the specification.
You can double check here: https://en.wikipedia.org/wiki/List_of_HTTP_header_fields

th-s-op, 2018-11-12:
Hi @arendst, after some final checks, I can confirm that you have to do the following changes to produce a compliant request header:

- remove the "Port: " request header
- instead, append a ":xxx" to the "Host: xyz" header
- remove all leading and trailing blanks from the headers - these also stopped it from working with my hosted web space

The resulting code would be:
```
if (connected) {
        String url = F("GET /cm?");
        url += uri;
        url += F(" HTTP/1.1\r\nHost: ");
        //url += IPAddress(host_ip).toString();
        url += host;
        if (port) {
          url += F(":");
          url += port;
        }
        url += F("\r\nConnection: close\r\n\r\n");
```

arendst, 2018-11-12:
Thx. Will update together with #4329

th-s-op, 2018-11-12:
**You are a genius! Thanks a lot for this and for this great project!!!**

closed 2018-11-12 by th-s-op