arendst/Tasmota #4332: Support for Electrodragon ESP LED Strip Board, Mosfet Drive

empenoso, 2018-11-12:
I bought ESP LED Strip Board, Mosfet Drive https://www.electrodragon.com/product/esp-led-strip-board/. Board works with Sonoff-Tasmota. RGB LED Strip also works.
But do not work together.

I can't send "+12V" to exit "V". 
Wiki: https://www.electrodragon.com/w/ESP_Light

Any ideas? 
Thank!

P.S. https://github.com/arendst/Sonoff-Tasmota/issues/1241 - similar topic.

ascillato, 2018-11-12:
Please, complete the troubleshooting template.

Also provide how you have configured your module.

empenoso, 2018-11-12:
**IMPORTANT NOTICE**
If you do not complete the template below it is likely that your issue will not be addressed. When providing information about your issue please be as extensive as possible so that it can be solved by as little as possible responses.

**FAILURE TO COMPLETE THE REQUESTED INFORMATION WILL RESULT IN YOUR ISSUE BEING CLOSED**

Make sure these boxes are checked [x] before submitting your issue - Thank you!

- [x] Searched the problem in issues (https://github.com/arendst/Sonoff-Tasmota/issues)
- [x] Searched the problem in the wiki (https://github.com/arendst/Sonoff-Tasmota/wiki/Troubleshooting)
- [ ] Searched the problem in the forum (https://groups.google.com/d/forum/sonoffusers)
- [ ] Searched the problem in the chat (https://discord.gg/Ks2Kzd4)
- [x] Development/Compiler/Upload tools used : Visual Studio Code/ESP8266Flasher
- [x] Hardware used : Electrodragon ESP LED Strip Board, Mosfet Drive https://www.electrodragon.com/product/esp-led-strip-board/
- [ ] If a pre-compiled release or development binary was used, which one? :
- [ ] You have tried latest release or development binaries? :
- [x] Provide the output of command ``status 0`` :
```
{
Status: {
Module: 18,
FriendlyName: [
"CorridorRGBW"
],
Topic: "CorridorRGBW",
ButtonTopic: "0",
Power: 1,
PowerOnState: 3,
LedState: 1,
SaveData: 1,
SaveState: 1,
SwitchTopic: "0",
SwitchMode: [
0,
0,
0,
0,
0,
0,
0,
0
],
ButtonRetain: 0,
SwitchRetain: 0,
SensorRetain: 0,
PowerRetain: 0
},
StatusPRM: {
Baudrate: 115200,
GroupTopic: "sonoffs",
OtaUrl: "http://thehackbox.org/tasmota/release/sonoff.bin",
RestartReason: "Software/System restart",
Uptime: "0T00:00:40",
StartupUTC: "2018-11-12T04:47:26",
Sleep: 0,
BootCount: 22,
SaveCount: 114,
SaveAddress: "FA000"
},
StatusFWR: {
Version: "6.3.0",
BuildDateTime: "2018-11-11T16:40:21",
Boot: 31,
Core: "2_3_0",
SDK: "1.5.3(aec24ac9)"
},
StatusLOG: {
SerialLog: 2,
WebLog: 2,
SysLog: 0,
LogHost: "",
LogPort: 514,
SSId: [
"ofnet",
"rinet"
],
TelePeriod: 300,
SetOption: [
"00008009",
"55818000",
"00000000"
]
},
StatusMEM: {
ProgramSize: 491,
Free: 512,
Heap: 17,
ProgramFlashSize: 4096,
FlashSize: 4096,
FlashMode: 2,
Features: [
"00000809",
"0FDAE794",
"240183A0",
"23B617CE",
"00003BC0"
]
},
StatusNET: {
Hostname: "CorridorRGBW-2480",
IPAddress: "192.168.15.54",
Gateway: "192.168.15.1",
Subnetmask: "255.255.255.0",
DNSServer: "212.33.225.212",
Mac: "5C:XX:7F:XX:E9:B0",
Webserver: 2,
WifiConfig: 5
},
StatusMQT: {
MqttHost: "192.168.15.15",
MqttPort: 1883,
MqttClientMask: "DVES_%06X",
MqttClient: "DVES_65E9B0",
MqttUser: "DVES_USER",
MqttType: 1,
MAX_PACKET_SIZE: 1000,
KEEPALIVE: 15
},
StatusTIM: {
UTC: "Mon Nov 12 04:48:07 2018",
Local: "Mon Nov 12 09:48:07 2018",
StartDST: "Sun Mar 25 02:00:00 2018",
EndDST: "Sun Oct 28 03:00:00 2018",
Timezone: 5,
Sunrise: "11:54",
Sunset: "21:14"
},
StatusSNS: {
Time: "2018-11-12T09:48:07"
},
StatusSTS: {
Time: "2018-11-12T09:48:07",
Uptime: "0T00:00:41",
Vcc: 3.178,
POWER: "ON",
Dimmer: 33,
Color: "545454",
HSBColor: "0,0,33",
Channel: [
33,
33,
33
],
Scheme: 0,
Fade: "OFF",
Speed: 1,
LedTable: "OFF",
Wifi: {
AP: 1,
SSId: "ofnet",
BSSId: "64:D1:XX:F5:XX:B4",
Channel: 7,
RSSI: 74
}
}
}
```
To Reproduce
Steps to reproduce the behavior:
I configured as follows: http://192.168.15.54/cm?cmnd=gpio:
{
GPIO0: "0 (None)",
GPIO1: "0 (None)",
GPIO2: "0 (None)",
GPIO3: "0 (None)",
GPIO4: "0 (None)",
GPIO5: "0 (None)",
GPIO12: "39 (PWM3)",
GPIO13: "38 (PWM2)",
GPIO14: "0 (None)",
GPIO15: "37 (PWM1)",
GPIO16: "0 (None)"
}

Expected behavior
A clear and concise description of what you expected to happen.
Wiki https://www.electrodragon.com/w/ESP_Light have nothing to do with the output V. I can’t understand how to physically apply for it "+".

Screenshots
If applicable, add screenshots to help explain your problem.

Additional context
Add any other context about the problem here.

(Please, remember to close the issue when the problem has been addressed)

**(Please, remember to close the issue when the problem has been addressed)**

ascillato, 2018-11-12:
> Wiki https://www.electrodragon.com/w/ESP_Light have nothing to do with the output V. I can’t understand how to physically apply for it "+".

Can you explain a bit more. I can't understand what is your issue.
Please attach pictures also

ascillato, 2018-11-12:
What you have tried? pwm command? dimmer command?

brainnex, 2018-11-12:
Nice and cheap board. I think I'll buy one too :-). 
V - is the input for 12-24V DC. (not output!)
W (GPIO14)- is the output for white channel. just set this as relais1 in Tasmota
R,G,B like you configured it already.

Hope this helps.

ascillato2, 2018-11-12:
Closing this issue as it has been answered.

### Support Information
See [Wiki](https://github.com/arendst/Sonoff-Tasmota/wiki) for more information.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.

closed 2018-11-12 by ascillato2

empenoso, 2018-11-13:
> Nice and cheap board. I think I'll buy one too :-).
> V - is the input for 12-24V DC. (not output!)
> W (GPIO14)- is the output for white channel. just set this as relais1 in Tasmota
> R,G,B like you configured it already.
> 
> Hope this helps.

Thank! It worked for me. But on another module. The first was apparently broken.