arendst/Tasmota #4333: MCP23xxx Output command by MQTT

letnic, 2018-11-12:
- Development/Compiler/Upload tools used : Arduino IDE 1.8.5
- Hardware used :  ESP8266 with Tasmota 6.3.0

Hi everyone, i got a problem with MCP23xxx output. 
I try to make a MQTT request to make output to ON and OFF. 

If i'm in the web interface, in console, i'm able to send command via sensor29 15,ON or sensor29 15,OFF has example. Now i want to make MQTT request to execute that task. 

I try google, i don't find anything in the wiki about MQTT message to trigger output pin on MCP23xxx chips.

Is anyone has do this before ? Is it already program ? If not can i make a rule to do this ? 

Thanks in advance for your help.

This is some output on the console window
23:04:23 MQT: tele/Sprinkler/SENSOR = {"Time":"2018-11-11T23:04:23","ANALOG":{"A0":1},"DS18B20":{"Temperature":20.8},"DHT11":{"Temperature":20.0,"Humidity":51.0},"MCP230XX":{"D0":1,"D1":1,"D2":1,"D3":1,"D4":1,"D5":1,"D6":1,"D7":1,"D8":0,"D9":0,"D10":0,"D11":0,"D12":0,"D13":0,"D14":0,"D15":0},"TempUnit":"C"}
23:04:23 MQT: tele/Sprinkler/SENSOR = {"Time":"2018-11-11T23:04:23","MCP230_OUT": {"OUT_D8":"OFF","OUT_D9":"OFF","OUT_D10":"OFF","OUT_D11":"OFF","OUT_D12":"OFF","OUT_D13":"OFF","OUT_D14":"OFF","OUT_D15":"OFF","END":1}}

ascillato, 2018-11-12:
Please, complete the troubleshooting template

ascillato2, 2018-11-12:
To send any Tasmota command using mqtt just send from your broker

cmnd/yoursonofftopic/sensor29 15,ON

closed 2018-11-12 by ascillato2