arendst/Tasmota #4334: OLED SSD1306 not working

helmar74, 2018-11-12:
**IMPORTANT NOTICE**
If you do not complete the template below it is likely that your issue will not be addressed. When providing information about your issue please be as extensive as possible so that it can be solved by as little as possible responses.

**FAILURE TO COMPLETE THE REQUESTED INFORMATION WILL RESULT IN YOUR ISSUE BEING CLOSED**

Make sure these boxes are checked [x] before submitting your issue - Thank you!

- [X] Searched the problem in issues (https://github.com/arendst/Sonoff-Tasmota/issues)
- [X] Searched the problem in the wiki (https://github.com/arendst/Sonoff-Tasmota/wiki/Troubleshooting)
- [X] Searched the problem in the forum (https://groups.google.com/d/forum/sonoffusers)
- [X] Searched the problem in the chat (https://discord.gg/Ks2Kzd4)
- [X] Development/Compiler/Upload tools used : Arduino IDE
- [X] Hardware used : Wemos Mini D
- [X] If a pre-compiled release or development binary was used, which one? :sonoff-display.bin
- [X] You have tried latest release or development binaries? : 6.3.0.4
- [ ] Provide the output of command ``status 0`` :
```
13:12:35 MQT: stat/sonoffWemosMini_6/STATUS = {"Status":{"Module":18,"FriendlyName":["WemosMini-6"],"Topic":"sonoffWemosMini_6","ButtonTopic":"0","Power":1,"PowerOnState":3,"LedState":1,"SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
13:12:35 MQT: stat/sonoffWemosMini_6/STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"sonoffStandardGruppe","OtaUrl":"http://xxx.xxx.xxx2:5555/data/firmwares/sonoff.bin","RestartReason":"Software/System restart","Uptime":"0T00:02:27","StartupUTC":"2018-11-12T12:10:08","Sleep":0,"BootCount":55,"SaveCount":97,"SaveAddress":"FA000"}}
13:12:35 MQT: stat/sonoffWemosMini_6/STATUS2 = {"StatusFWR":{"Version":"6.3.0","BuildDateTime":"2018-10-30T17:36:38","Boot":31,"Core":"2_3_0","SDK":"1.5.3(aec24ac9)"}}
13:12:35 MQT: stat/sonoffWemosMini_6/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["SSID_EG","sonoffap"],"TelePeriod":300,"SetOption":["00008009","55818000","00000000"]}}
13:12:35 MQT: stat/sonoffWemosMini_6/STATUS4 = {"StatusMEM":{"ProgramSize":485,"Free":516,"Heap":13,"ProgramFlashSize":1024,"FlashSize":4096,"FlashMode":3,"Features":["00000809","0FFAA59C","000187A0","23B617CA","00003BC0"]}}
13:12:35 MQT: stat/sonoffWemosMini_6/STATUS5 = {"StatusNET":{"Hostname":"sonoffWemosMini_6","IPAddress":"xxx.xxx.xxx166","Gateway":"xxx.xxx.xxx1","Subnetmask":"255.255.255.0","DNSServer":"xxx.xxx.xxx2","Mac":"84:F3:EB:0C:2D:65","Webserver":2,"WifiConfig":5}}
13:12:35 MQT: stat/sonoffWemosMini_6/STATUS6 = {"StatusMQT":{"MqttHost":"xxx.xxx.xxx50","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_0C2D65","MqttUser":"DVES_USER","MqttType":1,"MAX_PACKET_SIZE":1000,"KEEPALIVE":15}}
13:12:35 MQT: stat/sonoffWemosMini_6/STATUS7 = {"StatusTIM":{"UTC":"Mon Nov 12 12:12:35 2018","Local":"Mon Nov 12 13:12:35 2018","StartDST":"Sun Mar 25 02:00:00 2018","EndDST":"Sun Oct 28 03:00:00 2018","Timezone":99,"Sunrise":"07:12","Sunset":"16:33"}}
13:12:35 MQT: stat/sonoffWemosMini_6/STATUS10 = {"StatusSNS":{"Time":"2018-11-12T13:12:35"}}
13:12:35 MQT: stat/sonoffWemosMini_6/STATUS11 = {"StatusSTS":{"Time":"2018-11-12T13:12:35","Uptime":"0T00:02:27","Vcc":2.711,"POWER":"ON","Wifi":{"AP":1,"SSId":"SSID","BSSId":"9C:C7:A6:F7:72:A2","Channel":6,"RSSI":84}}}
```

**(Please, remember to close the issue when the problem has been addressed)**
I connected a SSD1306  OLED SPI Display to my Wemos Mini D. But I didn't get it running.
Simply nothing is shown.
I used a precompiled Image (sonoff-displays.bin)

Here's the output of the display command:
18:14:51 CMD: display
18:14:51 MQT: stat/sonoffWemosMini_6/RESULT = {"Display":{"Model":2,"Mode":0,"Dimmer":15,"Size":1,"Font":1,"Rotate":2,"Refresh":2,"Cols":[16,8],"Rows":2}}

Here's my GPIO config:
[image]

GPIO4 is connected to D1 on Display (SDA)
and 
GPIO5 is connected to D0 on Display (SCL)

Anybody an idea, what to try also?

smittna, 2018-11-12:
Are you sure about the SPI connection? What's the result from I2Cscan (web console)? Valid OLED addresses are 0x3C (OLED 128x32) or 0x3D (OLED 128x64) Read the Hardware connections and DisplayAdress at https://github.com/arendst/Sonoff-Tasmota/wiki/DisplayText-Command (SPI or I2C)

helmar74, 2018-11-12:
Hi, 

thanks a lot.
Here's the result of i2cscan:
```
17:58:21 CMD: i2cscan
17:58:21 MQT: stat/sonoffWemosMini_6/RESULT = {"I2CScan":"No devices found"}
```

Im getting insane ;-)
I have the following display:
[https://de.aliexpress.com/item/1pcs-blue-color-0-96-inch-128X64-OLED-Display-Module-For-arduino-0-96-IIC-SPI/32802648947.html](https://de.aliexpress.com/item/1pcs-blue-color-0-96-inch-128X64-OLED-Display-Module-For-arduino-0-96-IIC-SPI/32802648947.html)

I soldered R3 to R1 and make a shortwire for R8
(as described in the link above)
That should set the display to I2C Mode...

Maybe I've got a wiring problem?

helmar74, 2018-11-12:
I tried to set Displayaddress to 109 (which is the decimal value of 0x3d):
8:09:54 CMD: i2cscan
18:09:54 MQT: stat/sonoffWemosMini_6/RESULT = {"I2CScan":"Device(s) found at 0x3d"}
18:10:02 CMD: displayaddress
18:10:02 MQT: stat/sonoffWemosMini_6/RESULT = {"DisplayAddress1":109}

Is this correct?

Jason2866, 2018-11-12:
I think you soldered wrong resistors. The perfect english description says for I2C
IIC interface design      
In the iic interface to configure BS1 is 1, BS0 0;       So R1 and R4 welding, R2, R3 not welding, the R8 weldable      
      Also don't welding.

helmar74, 2018-11-12:
I removed shortcut on R8:
```
19:51:41 CMD: i2cscan
19:51:41 MQT: stat/sonoffWemosMini_6/RESULT = {"I2CScan":"No devices found"}
```

Jason2866, 2018-11-12:
What have you connected? I understand it that way you have to solder R1 and R4 for I2C

Jason2866, 2018-11-12:
I left you a message in DISCOR with a link to the modul i bought and is working without any soldering

Jason2866, 2018-11-13:
What do you get now for I2Cscan?

ascillato2, 2018-11-14:
Hi,

Have you managed to solve your issue?

helmar74, 2018-11-14:
Hi,

not until now. I ordered a other OLED Display, will give you feedback, when it arrives.
In the meantime I tried to get a 2,9" E-Paper (SPI) Display running. 
Also without success. I used a Wemos Mini D as Basis.

I connected GPIO13 to DIN (MOSI?), GPIO14 to CLK, and GPIO15 to CS of the Display.
GPIO15 was selected as CS Pin in Tasmota.
I choosed displaymodel 5 and displaymode 0. 
I wasn't able display anything....

Jason2866, 2018-11-14:
E-paper displays are connected via 3 wire SPI (CS,SCLK,MOSI)  the other 3 Interface lines of the display (DC,Reset,busy) may be left unconnected. The jumper on the circuit board of the display should be set to 3 wire SPI. You can NOT use GPIO16 for software spi!

MOSI=GPIO13, SCLK=GPIO14, CS=GPIO15

[image]

[image]

helmar74, 2018-11-14:
Something is wrong. This is a Screenshot of my settings:
[image]

You say CS=GPIO15.
But "Your" CS  (orange cable) is connected to D5 which is GPIO14.
You said CS=GPIO15 (which should be GPIO14)....

Jason2866, 2018-11-14:
You are right, but it will work with this setup too. I changed my setup a little
Dont get irretated i have a I2C sensor connected  (BMP 280)
[image]

helmar74, 2018-11-14:
Thanks!
I don't have the possibility to choose "103 SSPI_MOSI", "104 SSPI_SCLK" and "102 SSPI_CS".
I Think I have to set some compile option?

Jason2866, 2018-11-14:
https://github.com/gemu2015/Sonoff-Tasmota
Use branch Display
#define USE_SOFTSPI
#define USE_DISPLAY
#define USE_DISPLAY_EPAPER29

helmar74, 2018-11-15:
Hi thanks again.

I downloaded the zip file, but I couldn't find the 3 defines.
In which Files they must be defined?
Can I also use the Tasmota development branch?

Jason2866, 2018-11-15:
You have to add it in 
`user_config.h`
or if you have one in
`user_config_override.h`
No latest display drivers are in GEMU Fork only

Jason2866, 2018-11-15:
After a look in the Data Sheet of the SSD1306. I do find the soldering of the resistors strange.
I2C need BS1 "1" and and SPI 4 wire all "0". 
[image]

helmar74, 2018-11-15:
Got the SSD1306 OLED (new one I2C only version running.
Thanks to Jason2866 and

closed 2018-11-15 by helmar74

FCustoms, 2018-12-25:
In order to use the DisplayText command the DisplayMode must be set to zero or other modes must be disabled before compilation with #undef USE_DISPLAY_MODES1TO5.

ShaggyDog18, 2020-06-05:
trying to connect I2C: SSD1306 with no success. Observe the following mismatch:
display is connected correctly, result of display commend:
```
22:32:18 CMD: display
22:32:18 RSL: stat/tasmota_CAF0B7/RESULT = {"Display":{"Model":2,"Width":128,"Height":64,"Mode":2,"Dimmer":15,"Size":1,"Font":1,"Rotate":0,"Refresh":2,"Cols":[16,8],"Rows":2}}
```
At start up I2C address found - **0x3d**:
```
00:00:00 CFG: Loaded from flash at F4, Count 6858
00:00:00 **I2C: SSD1306 found at 0x3d**
00:00:00 Project tasmota Tasmota Version 8.3.1(tasmota)-2_7_1
00:00:00 UPL: Arduino OTA Enabled Port 8266
00:00:00 I2C: BME280 found at 0x76
```
but in run i2cscan, it returns **0x3C** address:
```
22:30:40 CMD: i2cscan
22:30:40 RSL: stat/tasmota_CAF0B7/RESULT = {"I2CScan":"Device(s) found at 0x3c 0x76"}
```

vinibali, 2023-11-04:
> address

I don't know if you resolved that, but I had exactely the same issue.
The i2C display got detected and the settings were correct, but it remained black.
Original result with displaymode 0
```
04:57:19.717 RSL: RESULT = {"Display":{"Model":2,"Type":0,"Width":128,"Height":64,"Mode":0,"Dimmer":50,"Size":1,"Font":1,"Rotate":0,"Invert":0,"Refresh":2,"Cols":[16,8],"Rows":2}}
```
So I changed the displaymode to 1 and I had to restart the device for get the OLED to work. Pretty strange, now the displaymode change works at runtime.
*So it might worth a try to restart after change the displaymode.*
```
05:17:23.775 RSL: RESULT = {"Display":{"Model":2,"Type":0,"Width":128,"Height":64,"Mode":1,"Dimmer":50,"Size":1,"Font":1,"Rotate":0,"Invert":0,"Refresh":2,"Cols":[16,8],"Rows":2}}
```