arendst/Tasmota #4335: No reads from SDS011 after physical restart

mgwoj, 2018-11-12:
**Describe the bug**
I've noticed, that after physical restart of Tasmota device (i.e. after unplugging and pluging to the main source), there are no readings from SDS011 sensor. Uptime status is send correctly via MQTT every few minutes. I need to initiate _Restart_ via web interface or MQTT command to start it working. 
It happens every time I do restart.

- [x] _Searched the problem in issues and in the wiki_
- [x] _Hardware used_ : NodeMCU Amica
- [x] _Development/Compiler/Upload tools used_ : VSCode 1.28
- [x] _If a pre-compiled release or development binary was used, which one?_ : development 39efb2aa5309d8116fd47172f994898bcace1060
- [x] _You have tried latest release or development binaries?_ : dev
- [x] _Provide the output of command_``status 0`` :
```
14:00:34 CMD: status 0
14:00:34 SRC: WebConsole from 192.168.10.25
14:00:34 RSL: Group 0, Index 1, Command STATUS, Data 0
14:00:34 MQT: stat/dust/STATUS = {"Status":{"Module":18,"FriendlyName":["Dust sensor"],"Topic":"dust","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
14:00:34 MQT: stat/dust/STATUS1 = {"StatusPRM":{"Baudrate":9600,"GroupTopic":"sonoffs","OtaUrl":"http://thehackbox.org/tasmota/release/sonoff.bin","RestartReason":"Software/System restart","Uptime":"0T02:29:05","StartupUTC":"2018-11-12T10:31:29","Sleep":0,"BootCount":31,"SaveCount":45,"SaveAddress":"3F7000"}}
14:00:34 MQT: stat/dust/STATUS2 = {"StatusFWR":{"Version":"6.3.0.2","BuildDateTime":"2018-11-05T20:01:56","Boot":31,"Core":"2_4_2","SDK":"2.2.1(cfd48f3)"}}
14:00:34 MQT: stat/dust/STATUS3 = {"StatusLOG":{"SerialLog":0,"WebLog":3,"SysLog":3,"LogHost":"192.168.xx.xx","LogPort":514,"SSId":["Irena","AiM"],"TelePeriod":300,"SetOption":["00008009","55818000","00000000"]}}
14:00:34 MQT: stat/dust/STATUS4 = {"StatusMEM":{"ProgramSize":533,"Free":2536,"Heap":18,"ProgramFlashSize":4096,"FlashSize":4096,"FlashMode":2,"Features":["00000809","0FDAE794","040183A0","23B617CE","00003BC0"]}}
14:00:34 MQT: stat/dust/STATUS5 = {"StatusNET":{"Hostname":"dust-4110","IPAddress":"192.168.10.xxx","Gateway":"192.168.10.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.xx.xx","Mac":"18:FE:34:DF:xx:xx","Webserver":2,"WifiConfig":4}}
14:00:34 MQT: stat/dust/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.xx.xx","MqttPort":1883,"MqttClientMask":"dust","MqttClient":"dust","MqttUser":"DVES_USER","MqttType":1,"MAX_PACKET_SIZE":1000,"KEEPALIVE":15}}
14:00:34 MQT: stat/dust/STATUS7 = {"StatusTIM":{"UTC":"Mon Nov 12 13:00:34 2018","Local":"Mon Nov 12 14:00:34 2018","StartDST":"Sun Mar 25 02:00:00 2018","EndDST":"Sun Oct 28 03:00:00 2018","Timezone":"+01:00","Sunrise":"07:54","Sunset":"17:14"}}
14:00:34 MQT: stat/dust/STATUS10 = {"StatusSNS":{"Time":"2018-11-12T14:00:34","SDS0X1":{"PM2.5":19.3,"PM10":26.1}}}
14:00:34 MQT: stat/dust/STATUS11 = {"StatusSTS":{"Time":"2018-11-12T14:00:34","Uptime":"0T02:29:05","Vcc":2.921,"Wifi":{"AP":1,"SSId":"Ixxx","BSSId":"00:26:18:72:xx:xx","Channel":6,"RSSI":38}}}
```

**To Reproduce**
* Unplug and plug back the device to the main source
* No readings from Nova SDS011 sensor
* Initiate _Restart_ via Web interface
* Readings works as expected

**Expected behavior**
* Readings should appear right after device restart

**Screenshots**
N/A

**Additional context**

ascillato2, 2018-11-12:
Please, type weblog 4 in the console in order to have more debug information and then please try again both cases when your sensor works and when it doesn't and paste the console output please

mgwoj, 2018-11-12:
After hard restart
```
00:00:00 CFG: Loaded from flash at 3F6, Count 46
00:00:00 APP: Boot Count 32
00:00:00 SRC: Restart
00:00:00 Project sonoff Dust sensor (Topic dust, Fallback dust, GroupTopic sonoffs) Version 6.3.0.2-2_4_2
00:00:00 SNS: Hardware Serial
00:00:00 CFG: Saved to flash at 3F5, Count 47, Bytes 3584
00:00:00 WIF: Checking connection...
00:00:00 WIF: Attempting connection...
00:00:00 WIF: Connecting to AP1 Ixxx in mode 11N as dust-4110...
00:00:01 WIF: Checking connection...
00:00:01 WIF: Attempting connection...
00:00:02 WIF: Checking connection...
00:00:02 WIF: Attempting connection...
00:00:03 WIF: Checking connection...
00:00:03 WIF: Attempting connection...
00:00:04 WIF: Checking connection...
00:00:04 WIF: Attempting connection...
00:00:05 WIF: Checking connection...
00:00:05 WIF: Attempting connection...
00:00:06 WIF: Checking connection...
00:00:06 WIF: Connected
00:00:06 DNS: Initialized
00:00:06 HTP: Web server active on dust-4110.local with IP address 192.168.xx.xx
00:00:08 MQT: Attempting connection...
00:00:08 MQT: Connected
00:00:08 MQT: tele/dust/LWT = Online (retained)
00:00:08 MQT: cmnd/dust/POWER = 
00:00:08 MQT: Subscribe to cmnd/dust/#
00:00:08 MQT: Subscribe to cmnd/sonoffs/#
00:00:08 MQT: Subscribe to cmnd/dust/#
00:00:08 MQT: tele/dust/INFO1 = {"Module":"Generic","Version":"6.3.0.2","FallbackTopic":"dust","GroupTopic":"sonoffs"}
00:00:08 MQT: tele/dust/INFO2 = {"WebServerMode":"Admin","Hostname":"dust-4110","IPAddress":"192.168.xx.xx"}
00:00:08 MQT: tele/dust/INFO3 = {"RestartReason":"External System"}
00:00:08 SRC: MQTT
00:00:08 RSL: Received Topic cmnd/dust/RESTART, Data Size 3, Data OFF
00:00:08 RSL: Group 0, Index 1, Command RESTART, Data OFF
00:00:08 MQT: stat/dust/RESULT = {"Restart":"1 to restart"}
00:00:08 SRC: MQTT
00:00:08 RSL: Received Topic cmnd/dust/RESTART, Data Size 3, Data OFF
00:00:08 RSL: Group 0, Index 1, Command RESTART, Data OFF
00:00:08 MQT: stat/dust/RESULT = {"Restart":"1 to restart"}
00:00:08 APP: (UTC) Mon Nov 12 19:15:46 2018, (DST) Sun Mar 25 02:00:00 2018, (STD) Sun Oct 28 03:00:00 2018
20:15:54 MQT: tele/dust/STATE = {"Time":"2018-11-12T20:15:54","Uptime":"0T00:00:16","Vcc":2.928,"Wifi":{"AP":1,"SSId":"Ixxx","BSSId":"00:26:18:72:XX:XX","Channel":6,"RSSI":50}}
20:16:04 WIF: Checking connection...
20:16:04 WIF: Connected
20:16:24 WIF: Checking connection...
20:16:24 WIF: Connected
20:16:44 WIF: Checking connection...
20:16:44 WIF: Connected
20:16:58 HTP: Main Menu
20:17:04 WIF: Checking connection...
20:17:04 WIF: Connected
20:17:05 HTP: Console
```

After restar via web gui
```
00:00:00 CFG: Loaded from flash at 3F5, Count 47
00:00:00 APP: Boot Count 33
00:00:00 SRC: Restart
00:00:00 Project sonoff Dust sensor (Topic dust, Fallback dust, GroupTopic sonoffs) Version 6.3.0.2-2_4_2
00:00:00 SNS: Hardware Serial
00:00:00 CFG: Saved to flash at 3F4, Count 48, Bytes 3584
00:00:00 WIF: Checking connection...
00:00:00 WIF: Attempting connection...
00:00:00 WIF: Connecting to AP1 Ixxx in mode 11N as dust-4110...
00:00:00 SER: Received C5 02 01 01 00 A4 9E 46
00:00:01 WIF: Checking connection...
[... 58 lines cut ...]
20:17:58 SER: Received C0 D8 00 08 01 A4 9E 23
20:17:59 SER: Received C0 D8 00 08 01 A4 9E 23
20:18:00 SER: Received C0 D8 00 08 01 A4 9E 23
20:18:01 SER: Received C0 D8 00 08 01 A4 9E 23
20:18:02 SER: Received C0 D8 00 08 01 A4 9E 23
20:18:03 SER: Received C0 D8 00 08 01 A4 9E 23
20:18:03 WIF: Checking connection...
20:18:03 WIF: Connected
20:18:04 SER: Received C0 D8 00 08 01 A4 9E 23
20:18:05 SER: Received C0 D8 00 08 01 A4 9E 23
20:18:06 SER: Received C0 D8 00 08 01 A4 9E 23
20:18:07 SER: Received C0 D8 00 08 01 A4 9E 23
20:18:08 SER: Received C0 D8 00 08 01 A4 9E 23
20:18:09 SER: Received C0 D8 00 08 01 A4 9E 23
20:18:10 SER: Received C0 D8 00 08 01 A4 9E 23
20:18:11 SER: Received C0 D8 00 08 01 A4 9E 23
20:18:12 SER: Received C0 D8 00 08 01 A4 9E 23
20:18:13 SER: Received C0 D8 00 08 01 A4 9E 23
20:18:14 SER: Received C0 D8 00 08 01 A4 9E 23
20:18:15 SER: Received C0 D8 00 08 01 A4 9E 23
20:18:16 SER: Received C0 D8 00 08 01 A4 9E 23
20:18:17 SER: Received C0 D8 00 08 01 A4 9E 23
20:18:18 SER: Received C0 D8 00 08 01 A4 9E 23
20:18:19 SER: Received C0 D8 00 08 01 A4 9E 23
```

ascillato, 2018-11-12:
Ok, seems that the sensor is not giving you any information in the hard boot case. How do you have connected the power of the sensor?

mgwoj, 2018-11-12:
NodeMCU GND - SDS GND (5)
NodeMCU Vin (5V) - SDS 5V (3)

ascillato, 2018-11-12:
Ok, so you are using the vin connector to provide 5V to the sensor?

That is the issue. Vin is the INPUT for voltage to the nodemcu, It can not give enough power to your sensor from that. Please, connect another 5V source to the sensor in order to provide enough power to your sensor.

Frogmore42, 2018-11-13:
Assuming the NodeMCU is connected to a USB adapter, power is probably not the reason in this case.

There are several things it could be.
It could be the sensor is coming out of reset later than Tasmota and doesn't automatically start talking.  In this case the initialization that Tasmota does will never be heard by the sensor.

Resetting Tasmota causes it to redo the initialization, but this time the sensor is already awake and ready when Tasmota starts talking to it.

It could be the sensor is not coming up correctly on power up, but the second time Tasmota does the initialization the sensor is working.

It could be a marginal power supply.  Using the VIN pin of NodeMCU means you have a diode voltage drop lower to your sensor and it might not like that.

If you have a logic analyzer, you could look at the serial lines and see what is different between the good and the bad cases.

arendst, 2018-11-13:
I agree with @Frogmore42 that the most obvious reason is the fact that Tasmota probes the device before it is ready. As this probe is only done once at restart the device won't show up over time as you noticed.

With some other devices I re-probe once every 100 seconds so implementing this for the SDS might solve your issue too.

ascillato2, 2018-11-13:
@mgwoj 

Hi, Theo has just added some fixes. Could you please test it. You can download the precompiled bins of the lastest version from: http://thehackbox.org/tasmota/

Thanks

mgwoj, 2018-11-13:
After flashing the image, it is no longer reading values from SDS. It is trying to read Analog0
```
22:52:48 MQT: tele/dust/SENSOR = {"Time":"2018-11-13T22:52:48","ANALOG":{"A0":0}}
```

W was uploading sensor-4m image, I also tried with compiled version from latest sources.
What is strange I have this issue even after built once again version for commit 39efb2a.

Probably I need to connect it to computer via USB and wipe the flash.

ascillato, 2018-11-13:
The analog has nothing to do with your sds sensor. If you use the sonoff-sensor.bin, it has ADC enabled for pin A0. If you use any other like sonoff.bin, it has pin A0 disabled and adc is used for reading internal Vcc.

Please, do a erase before flash and flash again with esptool.py using the precompiled bin sonoff.bin from http://thehackbox.org/tasmota/

Thanks

ascillato2, 2018-11-14:
Hi,

Have you managed to solve your issue?

mgwoj, 2018-11-14:
Hello,

I cleaned the flash using `esptool.py --port COM4 erase_flash` then uploaded the image using `esptool.py --port COM4 write_flash -fs 1MB -fm dout 0x0 sonoff.bin`.
Configured the device, it works, but:
1. after hard reset the main page shows nothing, then value of 0, and finally the correct readings. However the post to MQTT topic is done 5 mintues later. 
1. after soft reset, webpage shows values almost imediately and a few seconds later it is published to MQTT topic

In both cases webpage shows new readings a few minutes before they are published to MQTT.

Wiki page https://github.com/arendst/Sonoff-Tasmota/wiki/Wemos-D1-Mini-and-Nova-Fitness-SDS011-Laser-Dust-Sensor shows mapping
```
TX (GPIO01) Serial Out : 70 SDS0X1 Rx
RX (GPIO03) Serial In : 101 SDS0X1 Tx
```
however I had to swap it to
```
TX (GPIO01) Serial Out : 101 SDS0X1 Tx
RX (GPIO03) Serial In : 70 SDS0X1 Rx
```
Was it changed recently?

ascillato, 2018-11-14:
Great to know that it is working now for you.

TX and RX were not changed.

closed 2018-11-14 by ascillato2

arendst, 2018-11-16:
@mgwoj Yes it was changed recently. See https://github.com/arendst/Sonoff-Tasmota/wiki/What's-New#20181105---6303