arendst/Tasmota #4336: Multiple Sonoff Touch with common light

Lodeli, 2018-11-12:
**IMPORTANT NOTICE**
If you do not complete the template below it is likely that your issue will not be addressed. When providing information about your issue please be as extensive as possible so that it can be solved by as little as possible responses.

**FAILURE TO COMPLETE THE REQUESTED INFORMATION WILL RESULT IN YOUR ISSUE BEING CLOSED**

Make sure these boxes are checked [x] before submitting your issue - Thank you!

- [x] Searched the problem in issues (https://github.com/arendst/Sonoff-Tasmota/issues)
- [x] Searched the problem in the wiki (https://github.com/arendst/Sonoff-Tasmota/wiki/Troubleshooting)
- [ ] Searched the problem in the forum (https://groups.google.com/d/forum/sonoffusers)
- [ ] Searched the problem in the chat (https://discord.gg/Ks2Kzd4)
- [ ] Development/Compiler/Upload tools used :
- [x] Hardware used : maybe Touch US 2CH (will buy if a solution is found)
- [ ] If a pre-compiled release or development binary was used, which one? :
- [ ] You have tried latest release or development binaries? :
- [ ] Provide the output of command ``status 0`` :
```
STATUS 0 OUTPUT HERE
```

**(Please, remember to close the issue when the problem has been addressed)**

Hi all,
I've been looking for a way to address this problem without the need for my HA rules.
Let's say I have two rooms, with a corridor in common.
I want to install two Sonoff Touch with 2CH in each room; from these Sonoff, I need the 1st button to control the local own room light, and the 2nd button to control the light in the corridor keeping their on/off status updated (if I turn corridor light on from the Room#1 I want the Room#2 Sonoff to show this condition).

I checked the ButtonTopic feature, that would be a correct behavior for the corridor light but is not button-depending: I'd end up turning on the Room#1 light from the Room#2 Sonoff 2nd button and vice-versa.
Searching around it seems there's no way to have different ButtonTopic (or no ButtonTopic at all) for different buttons.
That's really a pity, IMHO.

Now I'm thinking about rules + Sonoff Basic device.
I could set a rule in each Sonoff Touch to send something like "cmnd/corridor/POWER %value%" on button 1 press, and let the button 2 behavior to default.
Then, a Sonoff Basic could listen to "corridor" topic and pilot the corridor light.
Maybe I'd lose the corridor status on Sonoff Touch buttons 1?

Is there any better way to achieve what I'm trying for?

ascillato, 2018-11-12:
Hi,

> Now I'm thinking about rules

That exactly is your solution. Please see the wiki at rules and the examples. Also search in issues. This was addressed several times. Thanks.

Lodeli, 2018-11-12:
Thank you @ascillato 
surely my fault, but I haven't been finding any issue that addresses these requirements: no switches (just buttons), a Sonoff gang for a group and other for another group (or none).
I've just found Sonoff grouped together to control the same light (easy task with ButtonTopic) or with switches wired to wall physical buttons.
But none to group part of a device.

Would be really great to have ButtonTopic specific per button, to create groups with mixed channels of multichannel devices, but I think there are reasons behind this limit.

Anyway, I've tried thinking the rules-way, but I don't know if the solution would work.
First rule:
`rule on button1#state do publish cmnd/corridor/power %value% endon`

Now, pressing the button 1 will trigger a Sonoff Basic to turn on the corridor light and update the state in HA system.
But.
I can't find a way to update the Sonoff Touch devices led status.
I could write a rule on MQTT connect to send a subscription on "corridor" topic, is this possible?

ascillato, 2018-11-12:
yes, you can sync devices using rules. just put a rule in one device and another rule in the other device.

When you buy your devices, give it a try and if you need help on the rules, please use the Tasmota Chat.