arendst/Tasmota #4337: Rules Stopped Functioning

meingraham, 2018-11-12:
- [X] Searched the problem in issues (https://github.com/arendst/Sonoff-Tasmota/issues)
- [X] Searched the problem in the wiki (https://github.com/arendst/Sonoff-Tasmota/wiki/Troubleshooting)
- [X] Searched the problem in the forum (https://groups.google.com/d/forum/sonoffusers)
- [X] Searched the problem in the chat (https://discord.gg/Ks2Kzd4)
- Development/Compiler/Upload tools used:
Arduino IDE
- Hardware used:
Sonoff 4Ch Pro
- You have tried latest release or development binaries?
No

I tried using Discord to troubleshoot this but nobody responded to my posts. This may be too detailed for an interactive chat.

BUTTONn#State is not being triggered by any means (Physical, Web, RF).

I originally flashed 6.0.0c (self-compiled) and I know the rules were operational then. The only change I have made is to upgrade the device to 6.1.1b (self-compiled). I upgraded all my devices at once - it wasn't anything specific I was doing to one device or another, just a firmware upgrade to all my devices. I don't know if the rules have been working after the firmware update. I had put this project on the back burner for a few weeks and am just now returning to it.

**STATUS 0**
```
stat/fan/STATUS = {"Status":{"Module":23,"FriendlyName":["Sonoff","Sonoff","Sonoff","Sonoff"],"Topic":"fan","ButtonTopic":"fan","Power":8,"PowerOnState":3,"LedState":0,"SaveData":1,"SaveState":1,"ButtonRetain":0,"PowerRetain":0}}
stat/fan/STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"sonoffs","OtaUrl":"http://sonoff.maddox.co.uk/tasmota/sonoff.bin","RestartReason":"Software/System restart","Uptime":"0T00:50:45","StartupUTC":"2018-11-12T00:55:37","Sleep":250,"BootCount":87,"SaveCount":1395,"SaveAddress":"F9000"}}
stat/fan/STATUS2 = {"StatusFWR":{"Version":"6.1.1b","BuildDateTime":"2018-07-19T22:02:08","Boot":6,"Core":"2_3_0","SDK":"1.5.3(aec24ac9)"}}
stat/fan/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["mySSID",""],"TelePeriod":300,"SetOption":["00008109","55818000"]}}
stat/fan/STATUS4 = {"StatusMEM":{"ProgramSize":424,"Free":576,"Heap":14,"ProgramFlashSize":1024,"FlashSize":1024,"FlashMode":3,"Features":["00000809","0F002B90","00000001","00000096","00000000"]}}
stat/fan/STATUS5 = {"StatusNET":{"Hostname":"fan-xxxx","IPAddress":"192.168.1.196","Gateway":"192.168.1.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.1.1","Mac":"xx:xx:xx:xx:xx:xx","Webserver":2,"WifiConfig":2}}
stat/fan/STATUS6 = {"StatusMQT":{"MqttHost":"openhabianpi","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_808B56","MqttUser":"openhabian","MqttType":2,"MAX_PACKET_SIZE":1000,"KEEPALIVE":15}}
stat/fan/STATUS7 = {"StatusTIM":{"UTC":"Mon Nov 12 01:46:22 2018","Local":"Sun Nov 11 20:46:22 2018","StartDST":"Sun Mar 11 02:00:00 2018","EndDST":"Sun Nov 04 02:00:00 2018","Timezone":99,"Sunrise":"06:57","Sunset":"17:24"}}
stat/fan/STATUS10 = {"StatusSNS":{"Time":"2018-11-11T20:46:22"}}
stat/fan/STATUS11 = {"StatusSTS":{"Time":"2018-11-11T20:46:22","Uptime":"0T00:50:45","Vcc":3.135,"POWER1":"OFF","POWER2":"OFF","POWER3":"OFF","POWER4":"ON","Wifi":{"AP":1,"SSId":"dd-wrt","RSSI":72,"APMac":"xx:xx:xx:xx:xx:xx"}}}
```

**Rules**
```
stat/fan/RESULT = {"Rule1":"ON","Once":"OFF","Free":270,"Rules":"on BUTTON1#State do Event toggling1=%var1% endon on event#toggling1<1 do Backlog Power2 OFF; Power3 OFF; Power1 ON; var1 1; var2 0; var3 0 endon on event#toggling1>0 do Backlog Power1 OFF; Power2 OFF; Power3 OFF; var1 0; var2 0; var3 0 endon"}
stat/fan/RESULT = {"Rule2":"ON","Once":"OFF","Free":271,"Rules":"on BUTTON2#State do Event toggling2=%var2% endon on event#toggling2<1 do Backlog Power3 OFF; Power1 ON; Power2 ON; var1 0; var2 1; var3 0 endon on event#toggling2>0 do Backlog Power1 OFF; Power2 OFF; Power3 OFF; var1 0; var2 0; var3 0 endon"}
stat/fan/RESULT = {"Rule3":"ON","Once":"OFF","Free":270,"Rules":"on BUTTON3#State do Event toggling3=%var3% endon on event#toggling3<1 do Backlog Power1 OFF; Power2 OFF; Power3 ON; var1 0; var2 0; var3 1 endon on event#toggling3>0 do Backlog Power1 OFF; Power2 OFF; Power3 OFF; var1 0; var2 0; var3 0 endon"}
```

**Weblog 4 logging**
_Clicked 'Toggle 1' button on Web GUI_
```
CMD: weblog 4
MQT: stat/fan/RESULT = {"WebLog":4}
CFG: Saved to flash at F7, Count 1397, Bytes 3584
SRC: WebGui from 192.168.1.250
MQT: stat/fan/RESULT = {"POWER1":"ON"}
MQT: stat/fan/POWER1 = ON
CFG: Saved to flash at F6, Count 1398, Bytes 3584
```

_Clicked the button on my RF remote linked to channel 1_
```
APP: Button1 level 1-0
MQT: cmnd/fan/POWER1 = OFF
SRC: MQTT
RSL: Received Topic cmnd/fan/POWER1, Data Size 3, Data OFF
RSL: Group 0, Index 1, Command POWER, Data OFF
MQT: stat/fan/RESULT = {"POWER1":"OFF"}
MQT: stat/fan/POWER1 = OFF
APP: Button1 level 0-1
CFG: Saved to flash at F5, Count 1399, Bytes 3584
```

It seems that TASMOTA **_is_** 'seeing' the button press (at least in the second event). I would have thought that a BUTTON#State would have been triggered for both. I seem to recall that when the rules were working, any "button" press triggered the rules.

One thing I don't understand about the second event is that it's saying that the source is MQTT and that it received a `POWER1` `cmnd` topic. What was transmitted? I didn't see a `stat` topic first.

Regards.

Mike

ascillato, 2018-11-12:
Hi, button is the Physical button. The webGUI and the RF are not the button.

ascillato, 2018-11-12:
> APP: Button1 level 1-0
> MQT: cmnd/fan/POWER1 = ON

That means that you have buttontopic different than 0.

Buttontopic and switchtopic OVERRIDES rules.

You need to have those to 0 in order to use rules

meingraham, 2018-11-12:
```
CMD: buttontopic
SRC: WebConsole from 192.168.1.250
RSL: Received Topic /buttontopic, Data Size 0, Data 
RSL: Group 0, Index 1, Command BUTTONTOPIC, Data 
MQT: stat/fan/RESULT = {"ButtonTopic":"fan"}
CMD: switchtopic
SRC: WebConsole from 192.168.1.250
RSL: Received Topic /switchtopic, Data Size 0, Data 
RSL: Group 0, Index 1, Command SWITCHTOPIC, Data 
MQT: stat/fan/RESULT = {"SwitchTopic":""0""}
```

ascillato, 2018-11-12:
MQT: stat/fan/RESULT = {"ButtonTopic":"fan"}

that's it. need to be 0 for using rules

meingraham, 2018-11-12:
Yup. Thanks.

So... I know I didn't change that... although I may have set that long ago when I was first testing this button triggering.

The RF does behave as if though the physical button was pressed... which is what I needed.

And, clicking the web GUI toggle does not trigger BUTTON#State (which I know used to trigger with 6.0.0c). But, I'll just walk a few steps so I'm close enough for the RF remote to work ;-) I can stand to burn one or two calories!

Thanks again.

Mike

closed 2018-11-12 by meingraham