arendst/Tasmota #4340: Feature request - Shelly 2LED

trommegutten, 2018-11-12:
I have successfully been able to flash this device with Tasmota. I tried to choose the Tasmota “generic” device but I couldn’t figure out how to configure it. Anyone?

ascillato, 2018-11-12:
Shelly 2 is already supported. See release notes. Thanks

ascillato2, 2018-11-12:
Closing this issue as it has been answered.

### Support Information
See [Wiki](https://github.com/arendst/Sonoff-Tasmota/wiki) for more information.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.

closed 2018-11-12 by ascillato2

trommegutten, 2018-11-12:
Both Shelly1 and Shelly2 are supported, but this post was regarding the Shelly 2LED. A led dimmer.

https://shelly.cloud/shelly-2-led/

[image]
[image]
[image]