arendst/Tasmota #4341: Shelly 2LED

trommegutten, 2018-11-12:
I have successfully been able to flash this device with Tasmota. I tried to choose the Tasmota “generic” device but I couldn’t figure out how to configure it. Anyone? 

https://shelly.cloud/shelly-2-led/

[image]
[image]
[image]

ascillato, 2018-11-12:
You can choose generic and put every GPIO as relay. Then go one by one turning on. Then you should find if it uses PWM and which color channel. Then you change that to PWM and try dimming.

If it uses another mcu to control dimming it is going to be more difficult

FrankX0, 2018-11-12:
From your photo: there is certainly another MCU.
A Nuvoton [N76E003](https://www.nuvoton.com/hq/products/microcontrollers/8bit-8051-mcus/low-pin-count-8051-series/n76e003/?__locale=en), which is a generic 8051 variant.

ascillato2, 2018-11-13:
Hi,

If it has another MCU the integration to Tasmota will be more difficult. Someone will need to do some reverse engineering in the communication between the stock firmware for the esp8266 and that MCU to emulate the same communication in Tasmota, and then make a driver for that.

Did you saved the stock firmware before flashing Tasmota?

Have you the digital analyzer to do that communication sniffing?

ascillato2, 2018-11-14:
@trommegutten 

Hi

Any news on this?

digiblur, 2018-11-14:
Looks like the same chip I've seen in some other dimmers, I do bet the communications protocol is different though given it isn't Tuya based.  They aren't too hard to reverse engineer usually give the communication is usually serial based.  I'd do one but I don't see them available on Amazon yet.

trommegutten, 2018-11-14:
Hi,
Sorry late answer. I have the stock software on another item, but I don’t know how to do sniffing. 
I tried to post on the support page yesterday, but no answer yet.
https://www.facebook.com/groups/1686781668087857/permalink/1829792280453461/

chaosmaster, 2018-11-29:
I've just added support for another Dimmer that uses N76E003.
https://github.com/arendst/Sonoff-Tasmota/wiki/PS-16-DZ-Dimmer
You can check if that works with this device.
Although I doubt it uses the same protocol.

ascillato, 2018-11-30:
@trommegutten

Hi

Any news on this?

ascillato2, 2018-12-14:
Closing this issue as nobody is working on this.

If anyone wants to add support for this, please feel free to ask to reopen this issue.

Thanks everyone for sharing their ideas.

### Support Information
See [Wiki](https://github.com/arendst/Sonoff-Tasmota/wiki) for more information.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.

closed 2018-12-14 by ascillato2