arendst/Tasmota #4348: Management of RF 433 MHz by means of RfKey and Rules

ilya89vet, 2018-11-13:
I have a board Wemos D1 and the RF transmitter is connected to it. In a configuration Sonoff Bridge and GPIO04 - RfSend is chosen
The RfSend <code> teams I can operate devices 433 MHz
I want to operate these devices by means of the RfKey1 buttons... RfKey16
Whether it is possible to realize it by means of scenarios? For example somehow so 
"**rule on RfKey1#State do RfSend 12345 endon**"
Or can you suggest any other option

arendst, 2018-11-13:
Sonoff Bridge is dedicated hardware using an ESP8266 and a RF microcontroller. This doesn't look like a Wemos D1 and a RF transmitter...

Forget Sonoff Bridge in your case and use the RcSwitch implementation.

ascillato, 2018-11-13:
RFKEY is not a trigger for rules. You can read the wiki for the available triggers.

andrethomas2, 2018-11-13:
Closing this issue as it has been answered.

### Support Information
See [Wiki](https://github.com/arendst/Sonoff-Tasmota/wiki) for more information.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.

closed 2018-11-13 by andrethomas2

ilya89vet, 2018-11-14:
This configuration works at esp8266 normally.
I ask you make the RfKey trigger

ilya89vet, 2018-11-14:
If this Trigger is realized, then it will be possible to operate RF devices
from the web interface that and is necessary for me!