arendst/Tasmota #4350: WiFi microphone, what do you think ?

araczkowski, 2018-11-13:
hi, 
I'm thinking about a possibility to add some microphone to Sonoff-Tasmota device.
Should be possible, according to this repo https://github.com/hjltu/esp8266-wifi-microphone

What do you think? Please share your thoughts
Thanks

ascillato, 2018-11-13:
Hi,

Sorry, but as Tasmota is not realtime software, that can not be done without dedicated hardware.

ascillato2, 2018-11-14:
Closing this issue as it has been answered.

### Support Information
See [Wiki](https://github.com/arendst/Sonoff-Tasmota/wiki) for more information.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.

closed 2018-11-14 by ascillato2

ascillato, 2018-11-14:
The other sensors takes 1 sample per second. The microfone needs at least 16000 samples per second for voice recognition quality and also the processing power for making a wav file for that. Not enough memory nor cpu power for doing all that plus Tasmota

araczkowski, 2018-11-14:
@ascillato but what about this https://github.com/hjltu/esp8266-wifi-microphone/blob/master/esp8266-wifi-mic.ino
is only 130 lines of code, the mqtt lib is already included in Tasmota 
maybe I missing something here?
Thanks

andrethomas, 2018-11-14:
@araczkowski Interesting link, but I agree with @ascillato - This is not implementable in Tasmota in the way in which the link you provided is using doing it.

andrethomas, 2018-11-14:
To be more specific, this function (for example)
```

void my_record()
{
	bt=false;
	for(int y=0;y<11;y++)
	{
		for(int i=0;i<1000;i++)
		{
			//rec = char(analogRead(A0)/4);
			//if(rec<200)
				//str_payload += rec;
			str_payload += char(analogRead(A0)/4);
			//delayMicroseconds(25);	//125-8kb
		}
	my_print();
	}
}
```

Will completely cripple the Tasmota firmware during that 11 second period.

hemangjoshi37a, 2022-07-10:
> Hi,
> 
> Sorry, but as Tasmota is not a realtime software, that can not be done without dedicated hardware.
> 
> The ESP8266 can be used as wifi microfone, but for that to be without glitches and with good performance, the cpu has to do only that.
> 
> This case is similar to the request of a mp3 decoder for the ESP8266. That also works in an ESP8266 but for only doing that.
> 
> All the tasks performed by Tasmota leaves some cpu power unused but that is not constant for doing something that requires realtime processing.
> 
> So, I think that the best for that is to have a dedicated hardware for that specific task and leave Tasmota for just home automation.

ESP32 has multiple cores. Can we take advantage of that and use one core dedicatedly for the audio purpose and another for communicating with wifi.

s-hadinger, 2022-07-10:
Esp32 also has DMA support which makes it less critical on timing. Whoever it may not be enough