arendst/Tasmota #4351: Start check on/off

voodooo82, 2018-11-14:
Hi i have buy more soft touch t1 3ch, with tasmota is all ok, is possible command to Google home with HA ecc ecc. I have install in my home only one for try, and 1 and 3 button la ok bit the second i have used with pulsetime because is used with other normale pulsant. The second Button when press switch on the relays and the button return to off. But when the Wi-Fi for yours problem disconnect to the tasmota and reconnect in all 3 channels the tasmota send a on/off check and for 1 and 3 channels isn't problem because is a simple switch but in secondo channels start the relays and the light la on... i have set PowerOnState in off after the start but is the same... tuo have a suggestion? 

Best regards

ascillato, 2018-11-14:
Sorry. I don't understand. What do you want to achieve?

ascillato, 2018-11-14:
Please complete the troubleshooting template and also post the output of your console