arendst/Tasmota #4353: OFF - door is closed. ON - door is open. Swap arbitrarily

empenoso, 2018-11-14:
**IMPORTANT NOTICE**
If you do not complete the template below it is likely that your issue will not be addressed. When providing information about your issue please be as extensive as possible so that it can be solved by as little as possible responses.

**FAILURE TO COMPLETE THE REQUESTED INFORMATION WILL RESULT IN YOUR ISSUE BEING CLOSED**

Make sure these boxes are checked [x] before submitting your issue - Thank you!

- [X] Searched the problem in issues (https://github.com/arendst/Sonoff-Tasmota/issues)
- [X] Searched the problem in the wiki (https://github.com/arendst/Sonoff-Tasmota/wiki/Troubleshooting)
- [ ] Searched the problem in the forum (https://groups.google.com/d/forum/sonoffusers)
- [ ] Searched the problem in the chat (https://discord.gg/Ks2Kzd4)
- [X] Development/Compiler/Upload tools used : Visual Studio Code / ESP8266Flasher
- [X] Hardware used : WEMOS D1 mini
- [ ] If a pre-compiled release or development binary was used, which one? :
- [ ] You have tried latest release or development binaries? :
- [X] Provide the output of command ``status 0`` :
```
{
Status: {
Module: 18,
FriendlyName: [
"AleksejDoor"
],
Topic: "AleksejDoor",
ButtonTopic: "0",
Power: 0,
PowerOnState: 3,
LedState: 1,
SaveData: 1,
SaveState: 1,
ButtonRetain: 0,
PowerRetain: 0
},
StatusPRM: {
Baudrate: 115200,
GroupTopic: "sonoffs",
OtaUrl: "http://sonoff.maddox.co.uk/tasmota/sonoff.bin",
RestartReason: "Software/System restart",
Uptime: "0T01:51:17",
StartupUTC: "2018-11-14T10:46:47",
Sleep: 0,
BootCount: 2106,
SaveCount: 3744,
SaveAddress: "F4000"
},
StatusFWR: {
Version: "6.1.1",
BuildDateTime: "2018-07-26T11:22:24",
Boot: 31,
Core: "2_3_0",
SDK: "1.5.3(aec24ac9)"
},
StatusLOG: {
SerialLog: 2,
WebLog: 2,
SysLog: 0,
LogHost: "",
LogPort: 514,
SSId: [
"ofnet",
""
],
TelePeriod: 300,
SetOption: [
"00008009",
"55818000"
]
},
StatusMEM: {
ProgramSize: 537,
Free: 464,
Heap: 13,
ProgramFlashSize: 4096,
FlashSize: 4096,
FlashMode: 2,
Features: [
"00000809",
"0FDAE794",
"0C000000",
"23B6179E",
"00000000"
]
},
StatusNET: {
Hostname: "AleksejDoor-1997",
IPAddress: "192.168.15.60",
Gateway: "192.168.15.1",
Subnetmask: "255.255.255.0",
DNSServer: "212.33.225.212",
Mac: "A0:20:XX:06:XX:CD",
Webserver: 2,
WifiConfig: 3
},
StatusMQT: {
MqttHost: "192.168.15.15",
MqttPort: 1883,
MqttClientMask: "DVES_%06X",
MqttClient: "DVES_0687CD",
MqttUser: "DVES_USER",
MqttType: 1,
MAX_PACKET_SIZE: 1000,
KEEPALIVE: 15
},
StatusTIM: {
UTC: "Wed Nov 14 12:38:04 2018",
Local: "Wed Nov 14 17:38:04 2018",
StartDST: "Sun Mar 25 02:00:00 2018",
EndDST: "Sun Oct 28 03:00:00 2018",
Timezone: 5,
Sunrise: "09:03",
Sunset: "16:54"
},
StatusSNS: {
Time: "2018-11-14T17:38:04",
Switch1: "ON",
Switch2: "ON"
},
StatusSTS: {
Time: "2018-11-14T17:38:04",
Uptime: "0T01:51:17",
Vcc: 2.766,
Wifi: {
AP: 1,
SSId: "ofnet",
RSSI: 46,
APMac: "64:XX:XX:F5:XX:B4"
}
}
}
```

A few months ago I flashed WEMOS D1 mini. A reed switch is connected to board. WEMOS is configured as "9 (Switch1)". I use openHAB and everything works:
`Switch	AleksejDoor_door  "Door [MAP(door_invert.map):%s]"    <door>   { mqtt=">[mosquitto:cmnd/AleksejDoor/POWER:command:*:default], <[mosquitto:stat/AleksejDoor/POWER:state:default]" }     `

WEMOS settings:
```http://192.168.15.60/cm?cmnd=gpio
{
GPIO0: "0 (None)",
GPIO1: "0 (None)",
GPIO2: "0 (None)",
GPIO3: "0 (None)",
GPIO4: "0 (None)",
GPIO5: "0 (None)",
GPIO12: "10 (Switch2)",
GPIO13: "0 (None)",
GPIO14: "9 (Switch1)",
GPIO15: "0 (None)",
GPIO16: "0 (None)"
}
```

BUT! Periodically OFF and ON change places themselves. Usually OFF - closed, ON - open. But they can change themselves. Then OFF is open, ON is closed.
How can I fix the positions OFF - closed, ON - open?

**(Please, remember to close the issue when the problem has been addressed)**

ascillato, 2018-11-14:
Sorry, I don't understand, what is on and off? What is position?

ascillato, 2018-11-14:
Which switchmode are you using?

ascillato, 2018-11-14:
Seems to be an issue in your openhab config.

May be would be better to use a rule for that. 

Go to the Tasmota console and type:
Switchmode1 1
Rule1 1 
Rule1 on switch1#state=0 do publish cmnd/AleksejDoor/POWER OFF endon on switch1#state=1 do publish cmnd/AleksejDoor/POWER ON endon

empenoso, 2018-11-14:
Sorry my english. If door is closed firmware return "POWER":"ON", if door is open firmware returns "POWER":"OFF".
But these values sometimes change and then with the door open firmware return "POWER":"ON".

ascillato, 2018-11-14:
May be would be better to use a rule for that.

Go to the Tasmota console and type:
Switchmode1 1
Rule1 1
Rule1 on switch1#state=0 do publish cmnd/AleksejDoor/POWER OFF endon on switch1#state=1 do publish cmnd/AleksejDoor/POWER ON endon

ascillato, 2018-11-14:
Ah, sorry, may be you will also need to update to latest for using rules. Your version is old.

empenoso, 2018-11-14:
Thanks for your help, I will try.

empenoso, 2018-11-15:
> May be would be better to use a rule for that.
> 
> Go to the Tasmota console and type:
> 
> Switchtopic1 0
> Switchmode1 1
> Rule1 1
> Rule1 on switch1#state=0 do publish cmnd/AleksejDoor/POWER OFF endon on switch1#state=1 do publish cmnd/AleksejDoor/POWER ON endon

Thanks for your reply. I updated the firmware to the latest version. But an error occurred. Here is the log:

> 10:02:44 CMD: Switchmode1 1
> 10:02:44 MQT: stat/ToiletDoor/RESULT = {"SwitchMode1":1}
> 10:02:50 CMD: Rule1 1
> 10:02:50 MQT: stat/ToiletDoor/RESULT = {"Rule1":"ON","Once":"OFF","StopOnError":"OFF","Free":389,"Rules":"on switch1#state=0 do publish cmnd/ToiletDoor/POWER OFF endon on switch1#state=1 do publish cmnd/ToiletDoor/POWER ON endon"}
> 10:03:02 CMD: Rule1 on switch1#state=0 do publish cmnd/ToiletDoor/POWER OFF endon on switch1#state=1 do publish cmnd/ToiletDoor/POWER ON endon
> 10:03:02 MQT: stat/ToiletDoor/RESULT = {"Rule1":"ON","Once":"OFF","StopOnError":"OFF","Free":389,"Rules":"on switch1#state=0 do publish cmnd/ToiletDoor/POWER OFF endon on switch1#state=1 do publish cmnd/ToiletDoor/POWER ON endon"}
> 10:03:21 RUL: SWITCH1#STATE=0 performs "publish cmnd/ToiletDoor/POWER OFF"
> 10:03:21 MQT: cmnd/ToiletDoor/POWER = OFF
> 10:03:22 MQT: stat/ToiletDoor/RESULT = {"Command":"Unknown"}
> 10:03:23 RUL: SWITCH1#STATE=1 performs "publish cmnd/ToiletDoor/POWER ON"
> 10:03:23 MQT: cmnd/ToiletDoor/POWER = ON
> 10:03:23 MQT: stat/ToiletDoor/RESULT = {"Command":"Unknown"}
> 10:03:25 RUL: SWITCH1#STATE=0 performs "publish cmnd/ToiletDoor/POWER OFF"
> 10:03:25 MQT: cmnd/ToiletDoor/POWER = OFF
> 10:03:25 MQT: stat/ToiletDoor/RESULT = {"Command":"Unknown"}
> 10:03:26 RUL: SWITCH1#STATE=1 performs "publish cmnd/ToiletDoor/POWER ON"
> 10:03:26 MQT: cmnd/ToiletDoor/POWER = ON
> 10:03:26 MQT: stat/ToiletDoor/RESULT = {"Command":"Unknown"}
> 10:03:49 MQT: tele/ToiletDoor/STATE = {"Time":"2018-11-15T10:03:49","Uptime":"0T00:15:14","Vcc":2.726,"Wifi":{"AP":1,"SSId":"rinet","BSSId":"6C:3B:6B:FE:BC:D1","Channel":1,"RSSI":86}}
> 10:03:49 MQT: tele/ToiletDoor/SENSOR = {"Time":"2018-11-15T10:03:49","Switch1":"ON"}
> 
Firmware returns now 
10:30:32 MQT: cmnd/ToiletDoor/POWER = OFF
10:30:32 MQT: stat/ToiletDoor/RESULT = {"Command":"Unknown"}
10:30:34 RUL: SWITCH1#STATE=1 performs "publish cmnd/ToiletDoor/POWER ON"
10:30:34 MQT: cmnd/ToiletDoor/POWER = ON
10:30:34 MQT: stat/ToiletDoor/RESULT = {"Command":"Unknown"}

wongnam, 2018-11-15:
Hi @empenoso 
Switchtopic1 0
Switchmode1 1
Rule1 on switch1#state do publish stat/AleksejDoor/POWER %value% endon
Rule 1

pls notice the topic stat not cmnd.

ascillato, 2018-11-15:
The publish is working fine according to your logs. Now you should be able to see the "power on" in your broker.

The command unknown that you see is because the topic of your publish is exactly the same of your device, so, the publish goes to your broker and it also come back as a command to your device. And as you don't have configured any relay, the command power is not recognized. 

So, you can change the topic in publish to not match your device.

ascillato, 2018-11-15:
@empenoso

Have yoy checked your broker? Are you receiving the status in the right way? 

Have you changed the topic of your sonoff to something different than the publish command you are using? 

Seems to me that now it is working fine due to your console output. 

Please check it and if it is working close this issue. Thanks

ascillato2, 2018-11-15:
Closing this issue as it has been addressed.

closed 2018-11-15 by ascillato2