arendst/Tasmota #4354: Wemos D1 with multiple 18B20 not recognised by Tasmota Admin in Hassio

smokey1300, 2018-11-14:
**IMPORTANT NOTICE**
If you do not complete the template below it is likely that your issue will not be addressed. When providing information about your issue please be as extensive as possible so that it can be solved by as little as possible responses.

**FAILURE TO COMPLETE THE REQUESTED INFORMATION WILL RESULT IN YOUR ISSUE BEING CLOSED**

**Describe the bug**
When using Tasmota 6.1 on a Wemos D1 mini multiple 18B20 sensors the Hassio interface does not recognize the sonoff during auto scan. Also manually adding the device causes error an does not save the address.

_Also, make sure these boxes are checked [x] before submitting your issue - Thank you!_
- [ x ] _Searched the problem in issues and in the wiki_
- [ x ] _Hardware used_ : Wemos D1 mini
- [ ] _Development/Compiler/Upload tools used_ : 
- [ ] _If a pre-compiled release or development binary was used, which one?_ :
- [ ] _You have tried latest release or development binaries?_ :
- [ ] _Provide the output of command_``status 0`` :
```
STATUS 0 OUTPUT HERE - DO NOT DELETE THE MARKERS ABOVE AND BELOW THIS LINE
```

**To Reproduce**
_Steps to reproduce the behavior:_
In Hassio using the admistration UI the Wemos is not recognised.

**Expected behavior**
_A clear and concise description of what you expected to happen._
To identify the usage/device.

**Screenshots**
_If applicable, add screenshots to help explain your problem._

**Additional context**
(copied from info screen)

Tasmota Program Version | 6.1.0

ESP Chip Id	1466106
Flash Chip Id	1458400
Flash Size	4096kB
Program Flash Size	4096kB
Program Size	559kB
Free Program Space	444kB
Free Memory	17kB

(copied from the error while manually adding the Wemos D1)

Device not found

JSON ERROR => 4: Syntax error
Please copy the whole error message and copy it in a new issue on GitHub. Somehow the Tasmota Firmware answered a wrong formatted JSON string which needs to get adjusted in TasmoAdmin.
JSON Answer => {"Status":{"Module":18,"FriendlyName":["Wemos52"],"Topic":"wemos52","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"SaveData":1,"SaveState":1,"ButtonRetain":0,"PowerRetain":0},"StatusPRM":{"Baudrate":115200,"GroupTopic":"sonoffs","OtaUrl":"http://sonoff.maddox.co.uk/tasmota/sonoff.bin","RestartReason":"Software/System restart","Uptime":"3T03:56:25","StartupUTC":"","Sleep":0,"BootCount":64,"SaveCount":89,"SaveAddress":"FB000"},"StatusFWR":{"Version":"6.1.0","BuildDateTime":"2018-07-06T21:03:23","Boot":31,"Core":"2_3_0","SDK":"1.5.3(aec24ac9)"},"StatusLOG":{"SerialLog":2,"WebLog":2,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["Polytarp24","Polytarp50"],"TelePeriod":300,"SetOption":["00008009","55818000"]},"StatusMEM":{"ProgramSize":559,"Free":444,"Heap":17,"ProgramFlashSize":4096,"FlashSize":4096,"FlashMode":2,"Features":["00000809","0FDEE794","0C000004","37FFBFCE","00000000"]},"StatusNET":{"Hostname":"Polytarp24","IPAddress":"192.168.1.52","Gateway":"192.168.1.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.2.27","Mac":"60:01:94:16:5E:FA","Webserver":2,"WifiConfig":3},"StatusMQT":{"MqttHost":"192.168.1.27","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_165EFA","MqttUser":"mqtt15","MqttType":1,"MAX_PACKET_SIZE":1000,"KEEPALIVE":15},"StatusTIM":{"UTC":"Sun Jan 04 03:59:52 1970","Local":"Sun Jan 04 03:59:52 1970","StartDST":"Thu Jan 01 00:00:00 1970","EndDST":"Thu Jan 01 00:00:00 1970","Timezone":1,"Sunrise":"07:43","Sunset":"16:06"},"StatusSNS":{"Time":"1970-01-04T03:59:52","DS18B20-1":{"Id":"0000003EC08A","Temperature":24.1},"DS18B20-2":{"Id":"0000003EDFB2","Temperature":-2.4},"DS18B20-3":{"Id":"03167264A8FF","Temperature":19.4},"DS18B20-4":{"Id":"051671B5E6FF","Temperature":22.8},"DS18B20-5":{"Id":"051671C28DFF","Temperature":22.5},"DS18B20-6":{"Id":"051671C68FFF","Temperature":21.6},"DS18B20-7":{"Id":"051671DDE4FF","Temperature":21.3},"DS18B20-8":{"Id":"0516720A4BFF","Temperature":22.1},"TempUnit": ..,"StatusSTS":{"Time":"1970-01-04T03:59:52","Uptime":"3T03:56:25","Vcc":2.751,"Wifi":{"AP":1,"SSId":"Polytarp24","RSSI":92,"APMac":"00:18:E7:ED:BD:C3"}}}

**(Please, remember to close the issue when the problem has been addressed)**

smokey1300, 2018-11-15:
Wrong repository

reloxx13, 2018-11-15:
but still, its kind of a tasmota issue, cuz there are characters in the json response which are breaking the json standard, leading in other app cannot parse it.

here it is formated and you will see that temp unit is missing.

@smokey1300 could you post a new status 0 from the device directly
http//192.168.1.52/cm?cmnd=status%200

```
{
  "Status": {
    "Module": 18,
    "FriendlyName": [
      "Wemos52"
    ],
    "Topic": "wemos52",
    "ButtonTopic": "0",
    "Power": 0,
    "PowerOnState": 3,
    "LedState": 1,
    "SaveData": 1,
    "SaveState": 1,
    "ButtonRetain": 0,
    "PowerRetain": 0
  },
  "StatusPRM": {
    "Baudrate": 115200,
    "GroupTopic": "sonoffs",
    "OtaUrl": "http://sonoff.maddox.co.uk/tasmota/sonoff.bin",
    "RestartReason": "Software/System restart",
    "Uptime": "3T03:56:25",
    "StartupUTC": "",
    "Sleep": 0,
    "BootCount": 64,
    "SaveCount": 89,
    "SaveAddress": "FB000"
  },
  "StatusFWR": {
    "Version": "6.1.0",
    "BuildDateTime": "2018-07-06T21:03:23",
    "Boot": 31,
    "Core": "2_3_0",
    "SDK": "1.5.3(aec24ac9)"
  },
  "StatusLOG": {
    "SerialLog": 2,
    "WebLog": 2,
    "SysLog": 0,
    "LogHost": "",
    "LogPort": 514,
    "SSId": [
      "xxxxxx24",
      "xxxxx50"
    ],
    "TelePeriod": 300,
    "SetOption": [
      "00008009",
      "55818000"
    ]
  },
  "StatusMEM": {
    "ProgramSize": 559,
    "Free": 444,
    "Heap": 17,
    "ProgramFlashSize": 4096,
    "FlashSize": 4096,
    "FlashMode": 2,
    "Features": [
      "00000809",
      "0FDEE794",
      "0C000004",
      "37FFBFCE",
      "00000000"
    ]
  },
  "StatusNET": {
    "Hostname": "xxx24",
    "IPAddress": "192.168.1.52",
    "Gateway": "192.168.1.1",
    "Subnetmask": "255.255.255.0",
    "DNSServer": "192.168.2.27",
    "Mac": "60:01:94:16:5E:FA",
    "Webserver": 2,
    "WifiConfig": 3
  },
  "StatusMQT": {
    "MqttHost": "192.168.1.27",
    "MqttPort": 1883,
    "MqttClientMask": "DVES_%06X",
    "MqttClient": "DVES_165EFA",
    "MqttUser": "mqtt15",
    "MqttType": 1,
    "MAX_PACKET_SIZE": 1000,
    "KEEPALIVE": 15
  },
  "StatusTIM": {
    "UTC": "Sun Jan 04 03:59:52 1970",
    "Local": "Sun Jan 04 03:59:52 1970",
    "StartDST": "Thu Jan 01 00:00:00 1970",
    "EndDST": "Thu Jan 01 00:00:00 1970",
    "Timezone": 1,
    "Sunrise": "07:43",
    "Sunset": "16:06"
  },
  "StatusSNS": {
    "Time": "1970-01-04T03:59:52",
    "DS18B20-1": {
      "Id": "0000003EC08A",
      "Temperature": 24.1
    },
    "DS18B20-2": {
      "Id": "0000003EDFB2",
      "Temperature": -2.4
    },
    "DS18B20-3": {
      "Id": "03167264A8FF",
      "Temperature": 19.4
    },
    "DS18B20-4": {
      "Id": "051671B5E6FF",
      "Temperature": 22.8
    },
    "DS18B20-5": {
      "Id": "051671C28DFF",
      "Temperature": 22.5
    },
    "DS18B20-6": {
      "Id": "051671C68FFF",
      "Temperature": 21.6
    },
    "DS18B20-7": {
      "Id": "051671DDE4FF",
      "Temperature": 21.3
    },
    "DS18B20-8": {
      "Id": "0516720A4BFF",
      "Temperature": 22.1
    },
    "TempUnit": ..,
    "StatusSTS": {
      "Time": "1970-01-04T03:59:52",
      "Uptime": "3T03:56:25",
      "Vcc": 2.751,
      "Wifi": {
        "AP": 1,
        "SSId": "xxxxx24",
        "RSSI": 92,
        "APMac": "00:18:E7:xx:xx:xx"
      }
    }
  }
```

arendst, 2018-11-15:
It's a Tasmota thing caused by string length limitations. When 8 DS18B20 are used the string becomes too long and will be truncated leading to an invalid JSON string.

As it's only a few characters I'm thinking of changing the text TEMPERATURE to TEMP when more than 2 DS18B20 are connected.

Still thinking....

reloxx13, 2018-11-15:
rather change it for all and not only if > 2.

will be easier to implement in other apps ^^

if not, others need to check if TEMPERATURE or TEMP exists.
(but for backward compatibility it still need to check boths in the future)

but u r the boss :D

ascillato, 2018-11-15:
I think that changing to TEMP for all sensors will brake backwards compatibility but it will leave more character for using this sensors and others all together.

curzon01, 2018-11-15:
backwards compatibility will be an issue - at least for me - this issue only happens on http output, not for mqtt

arendst, 2018-11-15:
@curzon01 thx for the research. 

The http output uses the log buffer and as this buffer is "only" 512 bytes long it truncates and extents with double dot. 512 bytes is more than enough for logging but I will test with a slightly longer one (it only needs +5 bytes for "C"}} but then if people have even more sensors connected it would be too short again).

For a quick solution I'll extent it to 520 bytes so this issue can then be closed ;-)

ascillato2, 2018-11-15:
Closing as has been fixed

closed 2018-11-15 by ascillato2