arendst/Tasmota #4356: DS18b20 randomly display 0C until power cycled.

thucar, 2018-11-15:
- [x] Searched the problem in issues (https://github.com/arendst/Sonoff-Tasmota/issues)
- [x] Searched the problem in the wiki (https://github.com/arendst/Sonoff-Tasmota/wiki/Troubleshooting)
- [x] Searched the problem in the forum (https://groups.google.com/d/forum/sonoffusers)
- [x] Searched the problem in the chat (https://discord.gg/Ks2Kzd4)
- [x] Development/Compiler/Upload tools used : Arduino IDE
- [x] Hardware used : Sonoff Basic + 6x DS18B20 + 4K7 pull-up resistor
- [x] If a pre-compiled release or development binary was used, which one? : http://thehackbox.org/tasmota/sonoff.bin
- [x] You have tried latest release or development binaries? : Latest release
- [x] Provide the output of command ``status 0`` :
```
08:34:11 MQT: sonoff/stat/boiler/STATUS = {"Status":{"Module":1,"FriendlyName":["Boiler Temps"],"Topic":"boiler","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"SaveData":1,"SaveState":1,"ButtonRetain":1,"PowerRetain":0}}
08:34:11 MQT: sonoff/stat/boiler/STATUS1 = {"StatusPRM":{"Baudrate":19200,"GroupTopic":"sonoffs","OtaUrl":"http://sonoff.maddox.co.uk/tasmota/sonoff.bin","RestartReason":"Power on","Uptime":"0T00:06:16","StartupUTC":"2018-11-01T06:27:55","Sleep":0,"BootCount":48,"SaveCount":181,"SaveAddress":"F6000"}}
08:34:11 MQT: sonoff/stat/boiler/STATUS2 = {"StatusFWR":{"Version":"6.2.1.7","BuildDateTime":"2018-10-30T05:12:49","Boot":6,"Core":"2_4_1","SDK":"2.2.1(cfd48f3)"}}
08:34:11 MQT: sonoff/stat/boiler/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["Indome Office","4G-Gateway-8FAB"],"TelePeriod":300,"SetOption":["00008049","55818000","00000000"]}}
08:34:11 MQT: sonoff/stat/boiler/STATUS4 = {"StatusMEM":{"ProgramSize":484,"Free":516,"Heap":15,"ProgramFlashSize":1024,"FlashSize":1024,"FlashMode":3,"Features":["00000809","4708AB80","00000000","020000C6","000000C0"]}}
08:34:11 MQT: sonoff/stat/boiler/STATUS5 = {"StatusNET":{"Hostname":"boiler-6146","IPAddress":"192.168.32.11","Gateway":"192.168.32.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.1.254","Mac":"BC:DD:C2:0F:18:02","Webserver":2,"WifiConfig":2}}
08:34:11 MQT: sonoff/stat/boiler/STATUS6 = {"StatusMQT":{"MqttHost":"xxxx","MqttPort":xxxx,"MqttClientMask":"SOF_BOILER_TEMPS","MqttClient":"SOF_BOILER_TEMPS","MqttUser":"xxxx","MqttType":2,"MAX_PACKET_SIZE":1000,"KEEPALIVE":15}}
08:34:11 MQT: sonoff/stat/boiler/STATUS7 = {"StatusTIM":{"UTC":"Thu Nov 01 06:34:11 2018","Local":"Thu Nov 01 08:34:11 2018","StartDST":"Sun Mar 25 03:00:00 2018","EndDST":"Sun Oct 28 04:00:00 2018","Timezone":99,"Sunrise":"07:33","Sunset":"16:24"}}
08:34:11 MQT: sonoff/stat/boiler/STATUS10 = {"StatusSNS":{"Time":"2018-11-01T08:34:11","DS18B20-1":{"Id":"000005957B1D","Temperature":93.4},"DS18B20-2":{"Id":"0000059625CB","Temperature":16.0},"DS18B20-3":{"Id":"00000596501B","Temperature":39.2},"DS18B20-4":{"Id":"0000059691D5","Temperature":104.2},"DS18B20-5":{"Id":"0014517554FF","Temperature":41.0},"DS18B20-6":{"Id":"0014517FE8FF","Temperature":40.5},"TempUnit":"C"}}
08:34:11 MQT: sonoff/stat/boiler/STATUS11 = {"StatusSTS":{"Time":"2018-11-01T08:34:11","Uptime":"0T00:06:16","Vcc":3.538,"POWER":"OFF","Wifi":{"AP":2,"SSId":"xxxx","BSSId":"xxxx","Channel":1,"RSSI":54}}}
```

I have 6 DS18B20 sensors hooked up to a Sonoff Basics GPIO14. Included a 4K7 pull-up resistor. They work perfectly fine for most of the time. However every couple days all the sensors readings drop to 0.0C Soft restarting the module does not help. Power cycling fixes the issue immediately.

NB! I am aware this is a duplicate of https://github.com/arendst/Sonoff-Tasmota/issues/4237 However the original has been closed for 14 days yet the issue still remains.

reloxx13, 2018-11-15:
if this happens again, post a status 0 with the 0 temps. guess this will be hard to debug.

arendst, 2018-11-15:
As you will have to power cycle to resolve this to me it seems one of your DS18B20 is sensors is blocking the bus which can only be solved by a power cycle/reset of the offending DS18B20. 

Tasmota has no means of resetting the bus when one of the sensors is locks it.

This blocking may well be caused by insufficient power too.

For debugging purposes try to remove some sensors and see what happens.

ascillato, 2018-11-15:
Are you powering them using the vcc from your sonoff basic? The basics are known to have a very limited power regulator, so if taking energy from there may be it is not enough for all your sensors.

Seems that you need to have another 3.3v power source for your sensors.

ascillato2, 2018-11-15:
@thucar 

Have you managed to test your sensors individually?

thucar, 2018-11-16:
Thanks for the tip, I will start swapping out individual sensors each time the issue manifests again. Once I have replaced them all, or it is apparent that the issue is resolved, I will let you guys know.

KPK75, 2018-11-16:
Mine too. After 6 days of work sensor suddenly shows 0.0 ( Wemos mini clone with fw 6.2.1.18, ds18x20 lib ) After soft restart DS18B20 is not detected. I'll try to do more tests.

ascillato2, 2018-11-16:
@thucar great, thanks. Please let us know your results.

closed 2018-11-16 by ascillato2

KPK75, 2018-11-16:
Mine was bad solder on the pull-up resistor

thucar, 2019-01-07:
I think we can safely say this was caused by either a faulty sensor or a connection. After replacing 4 out of 6 sensors I've been using, the problem has stopped.

edelstahlratte, 2019-01-07:
@thucar I have repeatedly experienced the same: 3 orders, 10 sensors each, 3 different sellers = 7 defective sensors total until now... because they are cheap, I just switch them out... Tooo lazy to measure something or investigate further... I have wondered whether these are maybe counterfeit DS18B20...

thucar, 2019-09-22:
So, about a year later, after double and triple-checking all my connections, replacing sensors and regular reboots of the module, I finally know what the cause of my problem was - incorrect SleepMode settings.

I was running version 6.3.something on it and it did not display Load information. I suspect I had set the sleep time to something very short when initially setting up the node. Today I was doing some re-wiring in the basement and cut the power for an hour. After that the temps would revert back to 0 in about a minute after a hard reboot. Every time. So I decided to try to upgrade to version 6.6 when I noticed it was not able to download the update file. It would reboot before the download could finish.

So going on a hunch, I set sleepmode to dynamic and sleep time to 50, as is the default in the current version. After that I immediately noticed a huge improvement in the responsiveness and stability of the node.

The temps have been stable for 4 hours since, hopefully they will stay that way.