arendst/Tasmota #4357: connection from source port 0

or78rXxZ, 2018-11-15:
i install sonoff.bin from thehackbox
Program Version | 6.3.0.7(sonoff)
Build Date & Time | 2018-11-14T06:32:49

everything seems to work quite well except that when i have the browser interface open i get a stream of attempted connections hitting my firewall about every 2 seconds
 kernel: [ 4611.114127] REJECT: IN=eth0 OUT= MAC=00:26:18:aa:a2:da:84:0d:8e:45:23:8f:08:00 SRC=192.168.2.60 DST=192.168.2.15 LEN=40 TOS=0x00 PREC=0x00 TTL=255 ID=8895 PROTO=TCP SPT=0 DPT=42616 WINDOW=53270 RES=0x00 ACK RST URGP=0

any ideas what this is
thanks

arendst, 2018-11-15:
How about the Ajax data update telling you the current state of the device ;-)

or78rXxZ, 2018-11-15:
got it
it only occurs when i am at the main menu
but the display is fully functional with that port blocked
maybe it's just a belt and braces thing
thanks