arendst/Tasmota #4358: Support HASS topic prefix

emontnemery, 2018-11-15:
HASS has introduced support for topic prefix which together with #4349 will allow to announce full feature of RGB(WW) lights, e.g. effects, without out-of-memory issues.
The concept of topic prefix allows the start or end of a topic to be replaced by a constant string, see example below.
Unfortunately, this is not compatible with the default `MQTT_FULLTOPIC = "%prefix%/%topic%/"` because the constant string %topic% is in the middle of the topic.

I've implemented this support, but I'd like some feedback before finishing and opening a PR.

Proposal:
- Add support for HASS topic prefix concept to allow announcing full capabilities without running into out-of-memory issues.
- Modify `HAssDiscovery()` from:
  ```
  void HAssDiscovery(uint8_t mode)
  {
    // Configure Tasmota for default Home Assistant parameters to keep discovery message as short as possible
    if (Settings.flag.hass_discovery) {
      Settings.flag.mqtt_response = 0;     // Response always as RESULT and not as uppercase command
      Settings.flag.decimal_text = 1;      // Respond with decimal color values
    }
    ...
   ```
  to:
    ```
  void HAssDiscovery(uint8_t mode)
  {
      // Configure Tasmota for default Home Assistant parameters to keep discovery message as short as possible
      if (Settings.flag.hass_discovery) {
        Settings.flag.mqtt_response = 0;     // Response always as RESULT and not as uppercase command
        Settings.flag.decimal_text = 1;      // Respond with decimal color values
        // TODO: Maybe only do this if Settings.mqtt_fulltopic does not end with "%topic%/"?
        strncpy_P(Settings.mqtt_fulltopic,     // Make topic compatible with HASS topic prefix
                  PSTR("%topic%/%prefix%"),
                  sizeof(Settings.mqtt_fulltopic));
      }
    ...
   ```
- (OPTIONAL) : Change default `MQTT_FULLTOPIC` to `"%topic%/%prefix%"` :)

Example of full length discovery message:
``` JSON
{
"name":"Sonoff",
"command_topic":"tasmota/sonoff_9D3B0B/cmnd/POWER",
"state_topic":"tasmota/sonoff_9D3B0B/stat/RESULT",
"value_template":"{{value_json.POWER}}",
"payload_off":"OFF",
"payload_on":"ON",
"availability_topic":"tasmota/sonoff_9D3B0B/tele/LWT",
"payload_available":"Online",
"payload_not_available":"Offline",
"brightness_command_topic":"tasmota/sonoff_9D3B0B/cmnd/Dimmer",
"brightness_state_topic":"tasmota/sonoff_9D3B0B/stat/RESULT",
"brightness_scale":100,
"on_command_type":"brightness",
"brightness_value_template":"{{value_json.Dimmer}}",
"rgb_command_topic":"tasmota/sonoff_9D3B0B/cmnd/Color2",
"rgb_state_topic":"tasmota/sonoff_9D3B0B/stat/RESULT",
"rgb_value_template":"{{value_json.Color}}",
"color_temp_command_topic":"tasmota/sonoff_9D3B0B/cmnd/CT",
"color_temp_state_topic":"tasmota/sonoff_9D3B0B/stat/RESULT",
"color_temp_value_template":"{{value_json.CT}}"
}
```

Shortened:
``` JSON
{
"~":"tasmota/sonoff_9D3B0B/",
"name":"Sonoff",
"command_topic":"~cmnd/POWER",
"state_topic":"~stat/RESULT",
"value_template":"{{value_json.POWER}}",
"payload_off":"OFF",
"payload_on":"ON",
"availability_topic":"~tele/LWT",
"payload_available":"Online",
"payload_not_available":"Offline",
"brightness_command_topic":"~cmnd/Dimmer",
"brightness_state_topic":"~stat/RESULT",
"brightness_scale":100,
"on_command_type":"brightness",
"brightness_value_template":"{{value_json.Dimmer}}",
"rgb_command_topic":"~cmnd/Color2",
"rgb_state_topic":"~stat/RESULT",
"rgb_value_template":"{{value_json.Color}}",
"color_temp_command_topic":"~cmnd/CT",
"color_temp_state_topic":"~stat/RESULT",
"color_temp_value_template":"{{value_json.CT}}"
}
```

emontnemery, 2018-11-15:
@arendst Some comments/feedback on this would be great!

arendst, 2018-11-15:
I like the shortend version very much.

To not change the default, which is probably in use by many people not using Hass ;-) I opt to make the change in HassDiscovery as you suggest too.

So if people want to use HassDiscovery they also need to agree on changing their FullTopic.

Perhaps to satisfy users not yet on Hass 0.81 we could use a SetOption55 flag toggle to choose either the old or new discovery method.

arendst, 2018-11-15:
- Document in Wiki and if issues arise point to it
- Start using the old feature. Switch by January and wait for issues to appear
- Remove SetOption55 by April

emontnemery, 2018-11-15:
Great, I'll update #4349 with SetOption55!

ascillato2, 2018-11-15:
Closing as PR #4349 was merged

closed 2018-11-15 by ascillato2

emontnemery, 2018-11-15:
@ascillato2 Please reopen, this is not same as #4349