arendst/Tasmota #4360: LC04 with rgbww 4in1

timtimsson, 2018-11-15:
Hello,
i use this [LC04](https://github.com/arendst/Sonoff-Tasmota/wiki/Arilux-LC04) module.
How can i control my rgbww strip 4in1? 
With LC11 i can use ct+dimmer -  with LC04 only dimmer...
Is the color-temperature changeable?
who can help me?
[image]

ascillato, 2018-11-16:
Have you tried the commands?

Please, read the wiki. Thanks

ascillato2, 2018-11-16:
Have you asked for support at the Tasmota chat? 

### Support Information
See [Wiki](https://github.com/arendst/Sonoff-Tasmota/wiki) for more information.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.

ascillato, 2018-11-16:
LC11 has 5 pwm channels

LC04 has 4 pwm channels

So the color temperature can not be changed using your LC04 and your led strip.

closed 2018-11-16 by ascillato2