arendst/Tasmota #4363: Sonoff POW R2 works on USB power, not on mains

pcb1962, 2018-11-16:
Tasmota 6.0.3.6
ESP core 2_4_2
Sonoff POW R2
Arduino IDE self compiled

Strange issue. Flashed a new POW R2 with the latest Tasmota, I've done many Sonoff devices successfully in the past. After flash it works perfectly on the USB connection.
When I plug it into the mains it connects to wifi, I can ping it, but it has no web interface. 
nmap shows no ports open (shows port 80 open on USB power).
Switch is also unresponsive and nothing sent to syslog.
I see the blue LED flash briefly during startup.
Seems it's hanging somewhere during the startup after connecting to wifi and before creating the web server.
Any ideas what's going wrong?
Next I will try downgrade core to 2.3.0, and also try precompiled binary but I'd rather just have it working as is.

- [x ] Searched the problem in issues (https://github.com/arendst/Sonoff-Tasmota/issues)
- [ x] Searched the problem in the wiki (https://github.com/arendst/Sonoff-Tasmota/wiki/Troubleshooting)
- [x ] Searched the problem in the forum (https://groups.google.com/d/forum/sonoffusers)
- [ x] Searched the problem in the chat (https://discord.gg/Ks2Kzd4)
- [ x] Development/Compiler/Upload tools used :
- [ x] Hardware used : 
- [ x] If a pre-compiled release or development binary was used, which one? :
- [ ] You have tried latest release or development binaries? :
- [ x] Provide the output of command ``status 0`` :
```
00:37:13 CMD: status 0
00:37:13 MQT: stat/KitchenHeater/STATUS = {"Status":{"Module":43,"FriendlyName":["KitchenHeater"],"Topic":"KitchenHeater","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
00:37:13 MQT: stat/KitchenHeater/STATUS1 = {"StatusPRM":{"Baudrate":4800,"GroupTopic":"sonoffs","OtaUrl":"http://www.xxxxxx.net/tasmota/KitchenHeater.bin","RestartReason":"Power on","Uptime":"0T00:00:35","StartupUTC":"2018-11-15T00:36:38","Sleep":0,"BootCount":24,"SaveCount":51,"SaveAddress":"F9000"}}
00:37:13 MQT: stat/KitchenHeater/STATUS2 = {"StatusFWR":{"Version":"6.3.0.6(sonoff)","BuildDateTime":"2018-11-13T19:52:07","Boot":31,"Core":"2_4_2","SDK":"2.2.1(cfd48f3)"}}
00:37:13 MQT: stat/KitchenHeater/STATUS3 = {"StatusLOG":{"SerialLog":0,"WebLog":2,"SysLog":3,"LogHost":"syslog.xxxxxx.net","LogPort":514,"SSId":["ssid","ssid"],"TelePeriod":300,"SetOption":["00008009","55C18000","00000000"]}}
00:37:13 MQT: stat/KitchenHeater/STATUS4 = {"StatusMEM":{"ProgramSize":461,"Free":540,"Heap":17,"ProgramFlashSize":1024,"FlashSize":4096,"FlashChipId":"1640EF","FlashMode":3,"Features":["00000809","0F002780","00010000","000000CE","000013C0"]}}
00:37:13 MQT: stat/KitchenHeater/STATUS5 = {"StatusNET":{"Hostname":"kitchenheater","IPAddress":"192.168.1.95","Gateway":"192.168.1.1","Subnetmask":"255.255.255.0","DNSServer":"81.187.228.195","Mac":"84:F3:EB:4A:D1:3E","Webserver":2,"WifiConfig":4}}
00:37:13 MQT: stat/KitchenHeater/STATUS6 = {"StatusMQT":{"MqttHost":"mqtt.xxxxxx.net","MqttPort":1883,"MqttClientMask":"kitchen_heater","MqttClient":"kitchen_heater","MqttUser":"sonoff","MqttType":1,"MAX_PACKET_SIZE":1000,"KEEPALIVE":15}}
00:37:13 MQT: stat/KitchenHeater/STATUS7 = {"StatusTIM":{"UTC":"Thu Nov 15 00:37:13 2018","Local":"Thu Nov 15 00:37:13 2018","StartDST":"Sun Mar 25 02:00:00 2018","EndDST":"Sun Oct 28 03:00:00 2018","Timezone":99,"Sunrise":"06:59","Sunset":"16:10"}}
00:37:13 MQT: stat/KitchenHeater/STATUS9 = {"StatusPTH":{"PowerDelta":80,"PowerLow":0,"PowerHigh":0,"VoltageLow":0,"VoltageHigh":0,"CurrentLow":0,"CurrentHigh":0}}
00:37:13 MQT: stat/KitchenHeater/STATUS10 = {"StatusSNS":{"Time":"2018-11-15T00:37:13","ENERGY":{"TotalStartTime":"2018-11-13T19:55:23","Total":0.000,"Yesterday":0.000,"Today":0.000,"Power":0,"ApparentPower":0,"ReactivePower":0,"Factor":0.00,"Voltage":0,"Current":0.000}}}
00:37:13 MQT: stat/KitchenHeater/STATUS11 =
{"StatusSTS":{"Time":"2018-11-15T00:37:13","Uptime":"0T00:00:35","Vcc":3.447,"POWER":"OFF","Wifi":{"AP":1,"SSId":"ssid","BSSId":"2C:54:2D:38:AD:89","Channel":8,"RSSI":62}}}
```

**(Please, remember to close the issue when the problem has been addressed)**

ascillato, 2018-11-16:
Seems that your firmware has something wrong in your self compiled version.

Please, try the precompiled bins from development branch at http://thehackbox.org/tasmota/

ascillato2, 2018-11-16:
Please, let us know if the precompiled bins has solved your issue. Thanks

closed 2018-11-16 by ascillato2

arendst, 2018-11-16:
Try version 6.3.0.7 and up as in version 6.3.0.1 until 6.3.0.6 serial comms on Sonoff Pow R2 and S31 will hang device.

See _changelog.ino

pcb1962, 2018-11-16:
Thanks Theo, I did have serial logging off in the config ("StatusLOG":{"SerialLog":0 ) and my UART completely disconnected when running on mains.
It's working with the precompiled binary now so I will try 6.3.0.7 and if that doesn't fix it I'll downgrade my build env to core 2_3_0. Thanks again.

pcb1962, 2018-11-16:
> Please, let us know if the precompiled bins has solved your issue. Thanks

Yes, the precompiled binary is running ok so I will try building the same versions (6.3.0.8 with core 2_3_0).

arendst, 2018-11-16:
Core 2.4.2 should work just fine:
```
00:00:00 Project sonoff Pow3 (Topic pow3, Fallback DVES_668F99, GroupTopic sonoffs) Version 6.3.0.8(sonoff)-2_4_2
00:00:00 WIF: Connecting to AP2 indebuurt2 in mode 11N as pow3...
00:00:02 RSL: tele/pow3/SENSOR = {"Time":"1970-01-01T00:00:02","ENERGY":{"TotalStartTime":"2018-10-30T16:45:16","Total":1.938,"Yesterday":0.157,"Today":0.069,"Power":7.45,"ApparentPower":15.98,"ReactivePower":14.10,"Factor":0.47,"Voltage":225.5,"Current":0.071}}
00:00:04 WIF: Connected
00:00:04 HTP: Web server active on pow3 with IP address 192.168.2.194
10:36:02 MQT: Attempting connection...
10:36:03 MQT: Connected
10:36:03 MQT: tele/pow3/LWT = Online (retained)
10:36:03 MQT: cmnd/pow3/POWER = 
10:36:03 MQT: tele/pow3/INFO1 = {"Module":"Sonoff Pow R2","Version":"6.3.0.8(sonoff)","FallbackTopic":"DVES_668F99","GroupTopic":"sonoffs"}
10:36:03 MQT: tele/pow3/INFO2 = {"WebServerMode":"Admin","Hostname":"pow3","IPAddress":"192.168.2.194"}
10:36:03 MQT: tele/pow3/INFO3 = {"RestartReason":"Software/System restart"}
10:36:03 MQT: stat/pow3/RESULT = {"POWER":"ON"}
10:36:03 MQT: stat/pow3/POWER = ON
10:36:11 MQT: tele/pow3/STATE = {"Time":"2018-11-16T10:36:11","Uptime":"0T00:00:14","Vcc":3.446,"POWER":"ON","Wifi":{"AP":2,"SSId":"indebuurt2","BSSId":"00:22:6B:FE:8E:20","Channel":6,"RSSI":72}}
10:36:11 MQT: tele/pow3/SENSOR = {"Time":"2018-11-16T10:36:11","ENERGY":{"TotalStartTime":"2018-10-30T16:45:16","Total":1.938,"Yesterday":0.157,"Today":0.070,"Period":0.02,"Power":7.42,"ApparentPower":15.47,"ReactivePower":13.60,"Factor":0.48,"Voltage":225.5,"Current":0.069}}
```
Keeping in mind the correct wifi parameter: ``  -DPIO_FRAMEWORK_ARDUINO_LWIP2_HIGHER_BANDWIDTH``

pcb1962, 2018-11-16:
I built with v1.4 Higher Bandwidth, I guess I should I use v2 Higher Bandwidth?

arendst, 2018-11-16:
I have trouble with v1.4
Others do not have trouble with v1.4
Same is true for v2 Higher Bandwidth.

Any way. I guess it makes no difference but if v1.4 fails just try v2

pcb1962, 2018-11-17:
Self compiled 6.3.0.8 with core 2_4_2 and "v2 Higher Bandwidth" is working perfectly, thanks for your help.