arendst/Tasmota #4364: Pow Module Rule for Dishwasher ?

zibous, 2018-11-16:
Is there a way to define a rule that turns off the power when the power measurement is not providing data. I would need the same for the washing machine and dryer.

[image]

I would like to use this to turn power off the dishwasher after the washing has finished.

Jason2866, 2018-11-16:
Yes, using rules https://github.com/arendst/Sonoff-Tasmota/wiki/Rules#rule-syntax
based on this possibilitys https://github.com/arendst/Sonoff-Tasmota/wiki/Commands#sonoff-pow-sonoff-s31-shelly2-and-pzem004t

zibous, 2018-11-16:
@Jason2866 
Thanks but I can't find any example for CurrentLow setting and rules.

curzon01, 2018-11-16:
@zibous - read the wiki - slowly :smiley: 

quote from [Rules Trigger](https://github.com/arendst/Sonoff-Tasmota/wiki/Rules#trigger)

> All connected sensors can be a trigger in the form as they are represented in the teleperiod or Status 8 JSON message like:
>
> - ``Energy#Power>100`` whenever the power used is more than 100W
...

so `Energy#Current<1` or whatever you want should work:

tamtaramtam

`rule1 on Energy#Current<1 do power1 0 endon`

zibous, 2018-11-16:
@curzon01
Thanks for the example, but I do not work for my device:

Rule: ***on Energy#Current<1 do power1 0 endon***
```
rule1 on Energy#Current<1 do power1 0 endon

21:33:32 MQT: stat/GSP100/RESULT = 
     {"Rule1":"ON",
	  "Once":"ON",
	  "StopOnError":"OFF",
	  "Free":474,
	  "Rules":"on Energy#Current<1 do power1 0 endon"
	  }
```

Value: ***Current":0.058***
```
21:34:20 MQT: tele/GSP100/SENSOR = 
     {"Time":"2018-11-16T21:34:20",
	  "ENERGY":{"Total":149.607,
	  "Yesterday":0.485,
	  "Today":0.566,
	  "Period":0,
	  "Power":3,
	  "ApparentPower":13,
	  "ReactivePower":13,
	  "Factor":0.23,
	  "Voltage":230,
	  "***Current":0.058***}}
```

State: ***"POWER1":"ON"***
```	  
21:37:20 MQT: tele/GSP100/STATE = 
      {"Time":"2018-11-16T21:37:20",
	  "Uptime":"7T01:13:31",
	  "Vcc":3.208,
	  "POWER1":"ON",.....
	  }
```

andrethomas, 2018-11-16:
Read the wiki again... this time slower ;)

ascillato2, 2018-11-16:
Hi, you have rule once enabled, so for the rule current<1 to trigger, first, your current needs to be >= 1.

andrethomas, 2018-11-16:
There are periods during the wash cycle when the machine will draw less than 1 amp... i.e. when it is filling with water and waiting for the dry cycle to complete.

I recommend figuring out how long a wash cycle takes in minutes (mine is 35 minutes) and then start a timer when current > 1 to count down the pre-determined time and then turn off.

Using the method that you used in your previous post may result in premature turn off's

curzon01, 2018-11-16:
@zibous it was an example not a solution. I fully agree with Andre suggestion but anyway I'm wondering why to power of the machine on program finish. Our is consuming < 0,4 W - a Sonoff n
to disable tge power needs the same. Do your machine consumes so much after program finish?