arendst/Tasmota #4365: Home Assistant MQTT Discovery - Add support for switches and sensors

HeWhoWas, 2018-11-16:
**Have you look for this feature in other issues and in the wiki?**
Yes

**Is your feature request related to a problem? Please describe.**
_Home Assistant is unable to auto-discovery sensors and switches._

**Describe the solution you'd like**
_Support to be added for this._

**Describe alternatives you've considered**
_Current workaround is to add them manually. It would be nice to have it all automated._

**Additional context**
_xdrv_12_home_assistant.ino:314_

Jason2866, 2018-11-16:
For switches it is there.
https://github.com/arendst/Sonoff-Tasmota/wiki/Home-Assistant#discovery

Please read the wiki ;-)

HeWhoWas, 2018-11-16:
I'm actually interested in sensors, but I'm inferring the above title from this code block:

```
if (Settings.flag.hass_discovery || (1 == mode)) {
    // Send info about relays and lights
    HAssDiscoverRelay();
    // Send info about buttons
    HAssDiscoverButton();
    // TODO: Send info about switches

    // TODO: Send info about sensors
  }```

andrethomas, 2018-11-16:
I think the correct question is why does Home Assistant not use telemetry data to discover sensors/switches - Almost all the sensors/switches are in json format over telemetry which offloads the discovery function from the Tasmota device to the Home Assistant software.

HeWhoWas, 2018-11-16:
Given that the application already supports the Home-Assistant specific implementation of discovery, I feel it makes sense to complete the implementation. I'm happy to do leg-work myself, but I'm not an embedded systems developer and the intricacies of C/C++ are a bit beyond me.

ascillato2, 2018-11-16:
Hi, as you saw in the code, it is in the TO DO list. You are very welcome to provide the needed code for that if you want. 

As Andre has explained. By now, you can use the sensors by the Tasmota telemetry configuring HA manually as explained in the wiki.

closed 2018-11-16 by ascillato2

HeWhoWas, 2018-11-16:
Is this not a valid feature request? At the very least, it would be helpful if the documentation was updated to reflect this limitation.

ascillato, 2018-11-16:
It is valid and it is in the TO DO list as explained in the code that you found

ascillato, 2018-11-16:
It is not implementad yet. It requires time and a lot of testing for doing that. If you need now that feature, please start working on that, we will help you. Remember that you can find us also in the Tasmota Chat.

ascillato, 2018-11-16:
If you don't want to program that, at least help us finding and posting which mqtt messages is expecting HA for every sensor type.

andrethomas, 2018-11-16:
> As Andre has explained. By now, you can use the sensors by the Tasmota telemetry configuring HA manually as explained in the wiki.

I actually meant the reverse - the sensors are already outputting data via mqtt - why can Home Assistant not be modified by its developers to learn from the telemetry data ;)