arendst/Tasmota #4367: Sonoff SV - CAME ZF1 gate open pilot lamp ports to manage open / close state

zonehun, 2018-11-16:
Hi All,

I'm using a Sonoff SV to open my gate whichis controlledby a CAME ZF1 board. The opening and closing process is smooth. I want to handle the state of the gate with the Sonoff as well. I found DrZzs youtube video how to do this with a reed switch. It is working flawless with my Hörmann Garage opener.... but I have no chance to add a reed switch to my gate. I figured out that there are a "gate open pilot lamp" on the CAME ZF1 board. This is working with two wires, 24 vac and with maximum 3W lamp. When the gate opened the lamp turning on, when the gate closed the lamp turning off. I think it can be used with the Sonoff as well.

I tried:
- Making an optocoupler circuit to make contactless connection between the ZF1 and the Sonoff GPIO14. 24 vac switches the Sonoff GPIO.
- I tried all of the possible values in Tasmota to configure a ghost relay and a switch to handle the electrical signals coming from the ZF1. But nothing working :(
- Tried everything what I found: pull-up resistor, capacitor, etc...
- I'm using 4n25 optocoupler.

I think my main problem is that the GPIO14 outputs 3.3v when I setup it as a switch. I need 0v to make it work perfectly. Or am I miss something? (I'm not an electrical guy. I'm a software developer.)

If anybody has any tips I would really appreciate it.
Thank you.

**IMPORTANT NOTICE**
If you do not complete the template below it is likely that your issue will not be addressed. When providing information about your issue please be as extensive as possible so that it can be solved by as little as possible responses.

**FAILURE TO COMPLETE THE REQUESTED INFORMATION WILL RESULT IN YOUR ISSUE BEING CLOSED**

Make sure these boxes are checked [x] before submitting your issue - Thank you!

- [x] Searched the problem in issues (https://github.com/arendst/Sonoff-Tasmota/issues)
- [x] Searched the problem in the wiki (https://github.com/arendst/Sonoff-Tasmota/wiki/Troubleshooting)
- [x] Searched the problem in the forum (https://groups.google.com/d/forum/sonoffusers)
- [x] Searched the problem in the chat (https://discord.gg/Ks2Kzd4)
- [x] Development/Compiler/Upload tools used : ESPTool
- [x] Hardware used : Sonoff SV
- [x] If a pre-compiled release or development binary was used, which one? : pre-compiled, 6.2.1
- [x] You have tried latest release or development binaries? : no
- [x] Provide the output of command ``status 0`` :
```
STATUS 0 OUTPUT HERE
```
21:21:51 CMD: status 0
21:21:51 RSL: STATUS = {"Status":{"Module":3,"FriendlyName":["CAME","CAME2"],"Topic":"sonoff","ButtonTopic":"0","Power":0,"PowerOnState":0,"LedState":1,"SaveData":1,"SaveState":1,"ButtonRetain":0,"PowerRetain":0}}
21:21:51 RSL: STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"sonoffs","OtaUrl":"http://sonoff.maddox.co.uk/tasmota/sonoff.bin","RestartReason":"Software/System restart","Uptime":"1T02:31:05","StartupUTC":"2018-11-15T17:50:46","Sleep":0,"BootCount":11,"SaveCount":229,"SaveAddress":"F7000"}}
21:21:51 RSL: STATUS2 = {"StatusFWR":{"Version":"6.2.1","BuildDateTime":"2018-09-09T16:50:26","Boot":7,"Core":"2_3_0","SDK":"1.5.3(aec24ac9)"}}
21:21:51 RSL: STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["otthoni_halo",""],"TelePeriod":300,"SetOption":["00008001","55818000","00000000"]}}
21:21:51 RSL: STATUS4 = {"StatusMEM":{"ProgramSize":471,"Free":532,"Heap":15,"ProgramFlashSize":1024,"FlashSize":1024,"FlashMode":3,"Features":["00000809","0FDAE794","000003A0","23B617CE","00000000"]}}
21:21:51 RSL: STATUS5 = {"StatusNET":{"Hostname":"sonoff-2104","IPAddress":"192.168.178.52","Gateway":"192.168.178.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.178.42","Mac":"CC:50:E3:02:A8:38","Webserver":2,"WifiConfig":5}}
21:21:51 RSL: STATUS7 = {"StatusTIM":{"UTC":"Fri Nov 16 20:21:51 2018","Local":"Fri Nov 16 21:21:51 2018","StartDST":"Sun Mar 25 02:00:00 2018","EndDST":"Sun Oct 28 03:00:00 2018","Timezone":1,"Sunrise":"08:00","Sunset":"17:09"}}
21:21:51 RSL: STATUS10 = {"StatusSNS":{"Time":"2018-11-16T21:21:51","Switch2":"OFF"}}
21:21:51 RSL: STATUS11 = {"StatusSTS":{"Time":"2018-11-16T21:21:51","Uptime":"1T02:31:05","Vcc":3.186,"POWER1":"OFF","POWER2":"OFF","Wifi":{"AP":1,"SSId":"otthoni_halo","RSSI":48,"APMac":"C8:0E:14:16:32:33"}}}

**(Please, remember to close the issue when the problem has been addressed)**

zonehun, 2018-11-16:
Attached my circuit and my tasmota settings page.

[image]

[image]

Jason2866, 2018-11-16:
I think your optocoupler output doesnt work.
[image]

andrethomas, 2018-11-16:
The AC side is also incorrect.

24V AC needs to be rectified to DC which you almost got right except for the miscalculation of RMS... 24vAC * 1.414 = 34V rectified... then you need to smooth it with a capacitor to make it a solid useful input which will provide a solid logic high

[image]

The increase of voltage requires you to change the resistor to ~1.6K to limit the current to 20mA and that combined with the smoothing capacitor will provide a solid trigger for the opto isolator when 24AC is present.

When 24AC is present the opto isolator's P will break down causing the resistance between the collector and emittor to collapse which will pull your switch pin down to logic LOW, whilst when there is no 24AC the opto isolator will not be triggered causing the P to be higher than the 10k pull-up resistor causing a logic HIGH on the switch pin.

The LED you have on the GPIO switch would be better suited on the pin used as your virtual relay.. remember to consider the limit of 10mA on the GPIO4 so @ 3.3V I would recommend a 330 ohm resistor from GPIO4 -> resistor -> LED.

ascillato2, 2018-11-16:
Closing this issue as it has been answered.

### Support Information
See [Wiki](https://github.com/arendst/Sonoff-Tasmota/wiki) for more information.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.

closed 2018-11-16 by ascillato2

zonehun, 2018-11-17:
@andrethomas Thank you for your detailed answer. 
Now I try to make the first half of the circuit to test how is the capacitor working. Switching on and off the VAC meanwhile measuring the DC voltage. If it looks OK I will build the other half. Just for sure :)
Do I need 470uF? I have 1uF (63V) capacitors at hand.

Do you have any suggestions how can I test my 4n25? Because maybe you are right and I blowed it up. (I have a multimeter only which can be used to measure voltage, power, resistance and disruption.)

andrethomas, 2018-11-17:
> Do I need 470uF? I have 1uF (63V) capacitors at hand.

The lower the capacitance, the more ripple you will get but perhaps it will not be too critical. Here are some tests - The solid horizontal line is GND - each vertical dotted line above ground represents 2V on this scope... so the ripple occurs as fast as the AC.. 50hz, or whatever you use in your country.

1uF:
[image]

2 x 1uF in parallel:
[image]

Remember we have two LED's here so likely our dropout voltage is around 1.2V per LED, so that makes for a dropout of 2.4V which is the absolute minimum voltage we need to maintain.

So I think for the best part you should get away with it but might have some multiple triggers in the first few milliseconds after 24AC is present.

As for testing the 4N25 - power the led side of it with a 330R resistor and 3.3V and measure what resistance you get on the decoupled side (the emitter-collector) - then remove the 3.3V supply and see if the resistance goes high/infinite. If that happens then it should be OK.

zonehun, 2018-11-17:
I created the firsthalf with the 1uF capacitor. It is working! I get 32 VDC.

I tested the 4N25. Powered the led side with 515 ohm resistor (I found that only) and 3.3V. Measured on the other side and I got 5.0 value in the 200Kohm measuring range. When I powerd of I measured high (1). So I think the 4N25 is in good condition.

I will continue with the circuit and I hope everything will be OK. I will try out what you suggested about the led and resitor between the GND and GPIO4 (ghost relay).

andrethomas, 2018-11-17:
> I created the first half with the 1uF capacitor. It is working! I get 32 VDC.

Yes, until you put 20mA load on it through the resistor then the power across the LED side of the optoisolator will look like the scope images I posted... so a bit of ripple but probably not enough to cause a change in LOGIC state on the ESP side of the optoisolator. 

Without the capacitor in place you would get a change in logic state every 20ms ;)

> I tested the 4N25. Powered the led side with 515 ohm resistor (I found that only) and 3.3V. Measured on the other side and I got 5.0 value in the 200Kohm measuring range. When I powerd of I measured high (1). So I think the 4N25 is in good condition.

Looks like you have not damaged it yet... that is good news.

andrethomas, 2018-11-17:
Here I scope'd 4 capacitor scenario's for your 2 x LED (1 LED + the LED inside the OPTO Isolator) so you can better understand what the capacitance is doing on a 20mA load - Perhaps others will find it useful also.

[image]

zonehun, 2018-11-17:
@andrethomas You are a genius! :) I put togeother everything and it is working like a charm! I'm very happy now, thank you for helping me. Just for the record I used a 22uF capacitor.

Issue should be closed.

andrethomas, 2018-11-17:
@zonehun I've been called that, amonst other things ;) Happy to hear it worked out for you. Enjoy!