arendst/Tasmota #4368: 1 Switch/Button - 2 Relais

Domepo, 2018-11-16:
Hi, I try to control 2 Relais with 1 button on tasmoto. If button pressed then power on relai 1 and power off relai 2, if button pressed again power off relai 1 and power on relai 2. On Arduino Is it easy because you can youse variables, but how can use it on tasmota,  I've tried to do this with rules but there is only an option that i must press it longer or 2 times . Thanks for your help :)

andrethomas, 2018-11-16:
https://github.com/arendst/Sonoff-Tasmota/wiki/Button-usage

1 short press for relay 1, 2 short presses for relay 2

ascillato, 2018-11-16:
Hi

The troubleshooting template is missing so there is a lot of information required in order to help you properly that we don't have. 

I will suppose that you are using last version of Tasmota from the development branch.

I suppose that you have configured your gpio as button. 

Go to the Tasmota console and type:

Buttontopic1 0
Setoption14 1
Rule1 1 
Rule1 on button1#state do backlog power1 2; power2 2 endon

For more information please read the wiki. 
Remember that you have also the Tasmota Chat for support.

closed 2018-11-16 by ascillato2