arendst/Tasmota #4369: Extrem long time to connect

marcohorstmann, 2018-11-17:
Hi,

I have a really strange behavior. I flashed today 11 sonoffs. 9x S20 and the last are Sonoff basics.
Only with the basics I have this issue. After restarting or unplug power it takes ages that Wifi will maybe have sometimes a little bit Wifi . My Fritzbox v7.0.1 (has Mesh) told me that the basic is connected but it seems that the network of this sonoff not really work stable. 

- [X] Searched the problem in issues (https://github.com/arendst/Sonoff-Tasmota/issues)
- [X] Searched the problem in the wiki (https://github.com/arendst/Sonoff-Tasmota/wiki/Troubleshooting)
- [X] Searched the problem in the forum (https://groups.google.com/d/forum/sonoffusers)
- [ ] Searched the problem in the chat (https://discord.gg/Ks2Kzd4)
- [X ] Development/Compiler/Upload tools used : ESPeasy FlashESP8266.exe
- [X ] Hardware used : FTDI Adapter on Windows 10
- [X] If a pre-compiled release or development binary was used, which one? : sonoff.bin in version 6.3.0
- [X] You have tried latest release or development binaries? : This is the lastest build
- [X] Provide the output of command ``status 0`` :
```
01:50:17 MQT: stat/sonoff2_01/STATUS = {"Status":{"Module":1,"FriendlyName":["sonoff2_01"],"Topic":"sonoff2_01","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
01:50:17 MQT: stat/sonoff2_01/STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"sonoffs","OtaUrl":"http://thehackbox.org/tasmota/release/sonoff.bin","RestartReason":"Software/System restart","Uptime":"0T00:16:23","StartupUTC":"2018-11-17T00:33:54","Sleep":0,"BootCount":19,"SaveCount":55,"SaveAddress":"F5000"}}
01:50:17 MQT: stat/sonoff2_01/STATUS2 = {"StatusFWR":{"Version":"6.3.0","BuildDateTime":"2018-10-30T17:33:53","Boot":6,"Core":"2_3_0","SDK":"1.5.3(aec24ac9)"}}
01:50:18 MQT: stat/sonoff2_01/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["INTERN-HORSTMANN",""],"TelePeriod":300,"SetOption":["00008009","55818000","00000000"]}}
01:50:18 MQT: stat/sonoff2_01/STATUS4 = {"StatusMEM":{"ProgramSize":489,"Free":512,"Heap":13,"ProgramFlashSize":1024,"FlashSize":1024,"FlashMode":3,"Features":["00000809","0FDAE794","000183A0","23B617CE","00003BC0"]}}
01:50:18 MQT: stat/sonoff2_01/STATUS5 = {"StatusNET":{"Hostname":"sonoff2_01","IPAddress":"192.168.1.181","Gateway":"192.168.1.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.1.1","Mac":"BC:DD:C2:0E:21:36","Webserver":2,"WifiConfig":5}}
01:50:18 MQT: stat/sonoff2_01/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.1.200","MqttPort":1883,"MqttClientMask":"sonoff2_01","MqttClient":"sonoff2_01","MqttUser":"sonoff","MqttType":1,"MAX_PACKET_SIZE":1000,"KEEPALIVE":15}}
01:50:18 MQT: stat/sonoff2_01/STATUS7 = {"StatusTIM":{"UTC":"Sat Nov 17 00:50:18 2018","Local":"Sat Nov 17 01:50:18 2018","StartDST":"Sun Mar 25 02:00:00 2018","EndDST":"Sun Oct 28 03:00:00 2018","Timezone":1,"Sunrise":"08:02","Sunset":"17:07"}}
01:50:18 MQT: stat/sonoff2_01/STATUS10 = {"StatusSNS":{"Time":"2018-11-17T01:50:18"}}
01:50:18 MQT: stat/sonoff2_01/STATUS11 = {"StatusSTS":{"Time":"2018-11-17T01:50:18","Uptime":"0T00:16:24","Vcc":3.190,"POWER":"OFF","Wifi":{"AP":1,"SSId":"INTERN-HORSTMANN","BSSId":"C8:0E:14:F7:68:6B","Channel":11,"RSSI":66}}}
```

Ping results
```
C:\WINDOWS\system32>ping 192.168.1.181 -t

Ping wird ausgeführt für 192.168.1.181 mit 32 Bytes Daten:
Antwort von 192.168.1.64: Zielhost nicht erreichbar.
Zeitüberschreitung der Anforderung.
Zeitüberschreitung der Anforderung.
Zeitüberschreitung der Anforderung.
Antwort von 192.168.1.181: Bytes=32 Zeit=1128ms TTL=128
Antwort von 192.168.1.181: Bytes=32 Zeit=25ms TTL=128
Antwort von 192.168.1.181: Bytes=32 Zeit=9ms TTL=128
Antwort von 192.168.1.181: Bytes=32 Zeit=4ms TTL=128
Antwort von 192.168.1.181: Bytes=32 Zeit=9ms TTL=128
Antwort von 192.168.1.181: Bytes=32 Zeit=26ms TTL=128
Zeitüberschreitung der Anforderung.
Zeitüberschreitung der Anforderung.
Zeitüberschreitung der Anforderung.
Zeitüberschreitung der Anforderung.
Zeitüberschreitung der Anforderung.
Zeitüberschreitung der Anforderung.
Zeitüberschreitung der Anforderung.
Zeitüberschreitung der Anforderung.
Zeitüberschreitung der Anforderung.
Zeitüberschreitung der Anforderung.
Antwort von 192.168.1.181: Bytes=32 Zeit=15ms TTL=128
Zeitüberschreitung der Anforderung.
Zeitüberschreitung der Anforderung.
Zeitüberschreitung der Anforderung.
Zeitüberschreitung der Anforderung.
Zeitüberschreitung der Anforderung.
Zeitüberschreitung der Anforderung.
Antwort von 192.168.1.181: Bytes=32 Zeit=221ms TTL=128
Zeitüberschreitung der Anforderung.
Zeitüberschreitung der Anforderung.
Zeitüberschreitung der Anforderung.
Zeitüberschreitung der Anforderung.
Zeitüberschreitung der Anforderung.
Zeitüberschreitung der Anforderung.
Zeitüberschreitung der Anforderung.
Zeitüberschreitung der Anforderung.
Zeitüberschreitung der Anforderung.
Zeitüberschreitung der Anforderung.
[... 46 lines cut ...]
I flashed a second basic module which reports the same on serial line:

```
00:00:09 CMD: 
00:00:09 RSL: stat/sonoff2_02/RESULT = {"Command":"Unknown"}
00:00:14 WIF: Connect failed with AP timeout
status 0
00:00:16 CMD: status 0
00:00:16 RSL: stat/sonoff2_02/STATUS = {"Status":{"Module":1,"FriendlyName":["sonoff2_02"],"Topic":"sonoff2_02","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
00:00:16 RSL: stat/sonoff2_02/STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"sonoffs","OtaUrl":"http://thehackbox.org/tasmota/release/sonoff.bin","RestartReason":"Power on","Uptime":"0T00:00:16","StartupUTC":"","Sleep":0,"BootCount":3,"SaveCount":6,"SaveAddress":"F6000"}}
00:00:16 RSL: stat/sonoff2_02/STATUS2 = {"StatusFWR":{"Version":"6.3.0","BuildDateTime":"2018-10-30T17:33:53","Boot":6,"Core":"2_3_0","SDK":"1.5.3(aec24ac9)"}}
00:00:16 RSL: stat/sonoff2_02/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["INTERN-HORSTMANN",""],"TelePeriod":300,"SetOption":["00008009","55818000","00000000"]}}
00:00:16 RSL: stat/sonoff2_02/STATUS4 = {"StatusMEM":{"ProgramSize":489,"Free":512,"Heap":20,"ProgramFlashSize":1024,"FlashSize":1024,"FlashMode":3,"Features":["00000809","0FDAE794","000183A0","23B617CE","00003BC0"]}}
00:00:16 RSL: stat/sonoff2_02/STATUS5 = {"StatusNET":{"Hostname":"sonoff2_02","IPAddress":"0.0.0.0","Gateway":"192.168.1.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.1.1","Mac":"BC:DD:C2:0E:70:24","Webserver":2,"WifiConfig":5}}
00:00:16 RSL: stat/sonoff2_02/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.1.200","MqttPort":1883,"MqttClientMask":"sonoff2_02","MqttClient":"sonoff2_02","MqttUser":"sonoff","MqttType":1,"MAX_PACKET_SIZE":1000,"KEEPALIVE":15}}
00:00:16 RSL: stat/sonoff2_02/STATUS7 = {"StatusTIM":{"UTC":"Thu Jan 01 00:00:16 1970","Local":"Thu Jan 01 00:00:16 1970","StartDST":"Thu Jan 01 00:00:00 1970","EndDST":"Thu Jan 01 00:00:00 1970","Timezone":1,"Sunrise":"07:43","Sunset":"16:03"}}
00:00:16 RSL: stat/sonoff2_02/STATUS10 = {"StatusSNS":{"Time":"1970-01-01T00:00:16"}}
00:00:16 RSL: stat/sonoff2_02/STATUS11 = {"StatusSTS":{"Time":"1970-01-01T00:00:16","Uptime":"0T00:00:16","Vcc":3.100,"POWER":"OFF","Wifi":{"AP":1,"SSId":"INTERN-HORSTMANN","BSSId":"C8:0E:14:F7:68:6B","Channel":11,"RSSI":56}}}
00:00:28 WIF: Connect failed with AP timeout
00:00:29 WIF: Connecting to AP1 INTERN-HORSTMANN in mode 11N as sonoff2_02...
00:00:43 WIF: Connect failed with AP timeout
00:00:57 WIF: Connect failed with AP timeout
00:00:58 WIF: Connecting to AP1 INTERN-HORSTMANN in mode 11N as sonoff2_02...
00:01:12 WIF: Connect failed with AP timeout
```

ascillato2, 2018-11-17:
Hi,

Yes, it is a known issue, please flash those sonoff basic with the latest version from development branch at http://thehackbox.org/tasmota/

Thanks

closed 2018-11-17 by ascillato2

ascillato, 2018-11-17:
https://github.com/arendst/Sonoff-Tasmota/issues/4254